home-assistant/operating-system #2894: Can't update HAOS 10.5 -> 11.1 on Raspberry 4

Code-TT, 2023-11-01:
### Describe the issue you are experiencing

~ ha os update
Processing... Done.

Error: Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/11.1/haos_rpi4-64-11.1.raucb: Response payload is not completed`
`

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Try to update over gui or with terminal: both fail

### Anything in the Supervisor logs that might be useful for us?

```txt
23-11-01 13:32:39 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/11.1/haos_rpi4-64-11.1.raucb
23-11-01 13:32:39 ERROR (MainThread) [supervisor.os.manager] Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/11.1/haos_rpi4-64-11.1.raucb: Response payload is not completed
```

### Anything in the Host logs that might be useful for us?

```txt
Nov 01 12:29:22 homeassistant kernel: audit: type=1327 audit(1698841762.425:434): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D49004F5554505554002D64003132372E302E302E3131002D6A00444F434B45525F4F5554505554
Nov 01 12:29:22 homeassistant kernel: audit: type=1325 audit(1698841762.433:435): table=nat:4 family=2 entries=1 op=nft_register_chain pid=17141 subj=unconfined comm="iptables"
Nov 01 12:29:22 homeassistant kernel: audit: type=1300 audit(1698841762.433:435): arch=c00000b7 syscall=211 success=yes exit=100 a0=3 a1=7fd718a1c8 a2=0 a3=1 items=0 ppid=17128 pid=17141 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
Nov 01 12:29:22 homeassistant kernel: audit: type=1327 audit(1698841762.433:435): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4E00444F434B45525F504F5354524F5554494E47
Nov 01 12:29:22 homeassistant kernel: audit: type=1325 audit(1698841762.441:436): table=nat:5 family=2 entries=2 op=nft_register_chain pid=17142 subj=unconfined comm="iptables"
Nov 01 12:29:22 homeassistant kernel: eth0: renamed from veth3b1f37a
Nov 01 12:29:22 homeassistant kernel: IPv6: ADDRCONF(NETDEV_CHANGE): veth25a1fd5: link becomes ready
Nov 01 12:29:22 homeassistant kernel: hassio: port 6(veth25a1fd5) entered blocking state
Nov 01 12:29:22 homeassistant kernel: hassio: port 6(veth25a1fd5) entered forwarding state
Nov 01 12:29:22 homeassistant NetworkManager[441]: <info>  [1698841762.5441] device (veth25a1fd5): carrier: link connected
Nov 01 12:29:45 homeassistant dockerd[549]: time="2023-11-01T12:29:45.138309963Z" level=info msg="Download failed, retrying (1/5): local error: tls: bad record MAC"
Nov 01 12:29:45 homeassistant dockerd[549]: time="2023-11-01T12:29:45.677912666Z" level=info msg="Download failed, retrying (1/5): local error: tls: bad record MAC"
Nov 01 12:29:48 homeassistant dockerd[549]: time="2023-11-01T12:29:48.749914611Z" level=info msg="Download failed, retrying (1/5): local error: tls: bad record MAC"
Nov 01 12:29:54 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-63f45daee44ceccd51ced285c503c595c575ea681e755800ffb26dd84f66511b-runc.ONRkWM.mount: Deactivated successfully.
Nov 01 12:29:57 homeassistant dockerd[549]: time="2023-11-01T12:29:57.110619198Z" level=info msg="Download failed, retrying (2/5): local error: tls: bad record MAC"
Nov 01 12:30:02 homeassistant systemd[1]: var-lib-docker-overlay2-cda41cf9ea9988c1085f6735c88924b74a1890e2193e634dcab2de7649f16e56-merged.mount: Deactivated successfully.
Nov 01 12:30:02 homeassistant systemd[1]: mnt-data-docker-overlay2-cda41cf9ea9988c1085f6735c88924b74a1890e2193e634dcab2de7649f16e56-merged.mount: Deactivated successfully.
Nov 01 12:30:04 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0e27ebdc4893a09a51ca8789a78c6b6f7b9e3955b45eaf18fc3e1e25664c7b37-runc.APOQUt.mount: Deactivated successfully.
Nov 01 12:30:09 homeassistant systemd[1]: var-lib-docker-overlay2-fe6bab493e69790632fa179b458e2b9343e737b689b3b37b3f8b99bde77a5550-merged.mount: Deactivated successfully.
Nov 01 12:30:09 homeassistant systemd[1]: mnt-data-docker-overlay2-fe6bab493e69790632fa179b458e2b9343e737b689b3b37b3f8b99bde77a5550-merged.mount: Deactivated successfully.
Nov 01 12:30:11 homeassistant systemd[1]: var-lib-docker-overlay2-eaab745dc8bd39a633dc6fa92d2747a639e258416412ea83a0c16b8accd1b72a-merged.mount: Deactivated successfully.
Nov 01 12:30:11 homeassistant systemd[1]: mnt-data-docker-overlay2-eaab745dc8bd39a633dc6fa92d2747a639e258416412ea83a0c16b8accd1b72a-merged.mount: Deactivated successfully.
Nov 01 12:30:13 homeassistant systemd[1]: var-lib-docker-overlay2-b856159943887938550bcb47aadeffa60c4dfa1c37a3b3747f84eee7ea8a004f-merged.mount: Deactivated successfully.
Nov 01 12:30:13 homeassistant systemd[1]: mnt-data-docker-overlay2-b856159943887938550bcb47aadeffa60c4dfa1c37a3b3747f84eee7ea8a004f-merged.mount: Deactivated successfully.
Nov 01 12:30:14 homeassistant systemd[1]: var-lib-docker-overlay2-5d027662601e41331c149cafadc2e861dfbb4af94d33a4ab7ec9e15f8537c613-merged.mount: Deactivated successfully.
Nov 01 12:30:14 homeassistant systemd[1]: mnt-data-docker-overlay2-5d027662601e41331c149cafadc2e861dfbb4af94d33a4ab7ec9e15f8537c613-merged.mount: Deactivated successfully.
Nov 01 12:30:14 homeassistant dockerd[549]: time="2023-11-01T12:30:14.910817687Z" level=info msg="Download failed, retrying (3/5): local error: tls: bad record MAC"
Nov 01 12:30:30 homeassistant dockerd[549]: time="2023-11-01T12:30:30.275767794Z" level=info msg="Download failed, retrying (4/5): local error: tls: bad record MAC"
Nov 01 12:30:51 homeassistant dockerd[549]: time="2023-11-01T12:30:51.023557173Z" level=info msg="Download failed, retrying (5/5): local error: tls: bad record MAC"
Nov 01 12:30:55 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-63f45daee44ceccd51ced285c503c595c575ea681e755800ffb26dd84f66511b-runc.l5zlI3.mount: Deactivated successfully.
Nov 01 12:31:16 homeassistant dockerd[549]: time="2023-11-01T12:31:16.486824817Z" level=error msg="Download failed after 6 attempts: local error: tls: bad record MAC"
Nov 01 12:31:16 homeassistant dockerd[549]: time="2023-11-01T12:31:16.587262730Z" level=info msg="Attempting next endpoint for pull after error: local error: tls: bad record MAC"
Nov 01 12:31:16 homeassistant dockerd[549]: time="2023-11-01T12:31:16.594106417Z" level=info msg="Layer sha256:db6eff751cad7126fa616acf69a2fa5104d2514305ff9daa2527f72ec9e0e26a cleaned up"
Nov 01 12:31:16 homeassistant dockerd[549]: time="2023-11-01T12:31:16.599146948Z" level=info msg="Layer sha256:fd65f03b87355d4dc8326a3df0ae6c3ac5cdcedc1ebd7315db0839b37eb19f6b cleaned up"
Nov 01 12:31:16 homeassistant dockerd[549]: time="2023-11-01T12:31:16.615746664Z" level=info msg="Layer sha256:58f106dd8b5308d7c08b3040be4981c63fa6b19bbcd87eed4d90926709c6ecf5 cleaned up"
Nov 01 12:31:16 homeassistant dockerd[549]: time="2023-11-01T12:31:16.616434042Z" level=info msg="Layer sha256:f474dec804666c88825e34a160e0fcdcf7cc99e2ebcb95cf0c7fde299428eebd cleaned up"
[... 124 lines cut ...]
version_api | ok
installed_addons | Advanced SSH & Web Terminal (15.1.0), File editor (5.6.0), Mosquitto broker (6.3.1), Studio Code Server (5.10.1), Dnsmasq (1.6.0), TasmoAdmin (0.25.1), MariaDB (2.6.1), Home Assistant Google Drive Backup (0.111.1), Valetudo Mapper (1.11.0), Samba share (10.0.2), NGINX Home Assistant SSL proxy - Port 80 Version (3.0.2), Cloudflared (4.2.9)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 25
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 28. Oktober 2023 um 10:52
-- | --
current_recorder_run | 1. November 2023 um 13:32
estimated_db_size | 387.53 MiB
database_engine | mysql
database_version | 10.6.12

</details>

agners, 2023-11-02:
This looks some type of Internet connectivity issue, could be intermittent. Do you still experience that problem?

Code-TT, 2023-11-05:
It's working well after updating to Home Assistant 2023.11.1. Thx

closed 2023-11-05 by Code-TT