home-assistant/operating-system #2896: Homeassistant as VM does not start after shutdown

chuckfy, 2023-11-01:
### Describe the issue you are experiencing

Homeassistant is installed as an appliance / VM under truenas. Original installation followed [this guide](https://community.home-assistant.io/t/installing-haos-in-a-vm-on-truenas-scale/511393)

After creation of the VM it does start and work normally. When it is shutdown it usually will not start afterwards. One time it took three shutdowns until the VM did not start anymore for good.

The error message under truenas reads:

[EFAULT] Unable to define domain for homeassistant: cannot serialize None (type NoneType)

There are [reports from different](https://www.truenas.com/community/threads/sudden-issue-to-start-vm.113827/#post-788032) users which are now facing the same issue. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Create VM
2. Stop VM / Reboot Server
3. VM will not start

[EFAULT] Unable to define domain for homeassistant: cannot serialize None (type NoneType)

### Anything in the Supervisor logs that might be useful for us?

```txt
no because HA does not start
```

### Anything in the Host logs that might be useful for us?

```txt
no because HA does not start
```

### System information

## System Information

version | core-2023.10.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.59
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4944
Installed Version | 1.33.0
Stage | running
Available Repositories | 1326
Downloaded Repositories | 8

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 37

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.1
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 48.5 GB
disk_used | 8.5 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), Mosquitto broker (6.3.1), RaspberryMatic CCU (3.71.12.20231020), Samba Backup (5.2.0), ESPHome (2023.10.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 5
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 26. Oktober 2023 um 19:37
-- | --
current_recorder_run | 1. November 2023 um 21:19
estimated_db_size | 343.78 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Additional information

Logfile from truenas:

```
2023-10-19 21:27:36.109+0000: starting up libvirt version: 7.0.0, package: 3 (Andrea Bolognani <eof@kiyuko.org> Fri, 26 Feb 2021 16:46:34 +0100), qemu version: 5.2.0Debian 1:5.2+dfsg-11+deb11u2, kernel: 5.15.131+truenas, hostname: truenas.local
LC_ALL=C \
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin \
HOME=/var/lib/libvirt/qemu/domain-3-13_HomeAssistantApp \
XDG_DATA_HOME=/var/lib/libvirt/qemu/domain-3-13_HomeAssistantApp/.local/share \
XDG_CACHE_HOME=/var/lib/libvirt/qemu/domain-3-13_HomeAssistantApp/.cache \
XDG_CONFIG_HOME=/var/lib/libvirt/qemu/domain-3-13_HomeAssistantApp/.config \
QEMU_AUDIO_DRV=spice \
/usr/bin/qemu-system-x86_64 \
-name guest=13_HomeAssistantApp,debug-threads=on \
-S \
-object secret,id=masterKey0,format=raw,file=/var/lib/libvirt/qemu/domain-3-13_HomeAssistantApp/master-key.aes \
-blockdev '{"driver":"file","filename":"/usr/share/OVMF/OVMF_CODE.fd","node-name":"libvirt-pflash0-storage","auto-read-only":true,"discard":"unmap"}' \
-blockdev '{"node-name":"libvirt-pflash0-format","read-only":true,"driver":"raw","file":"libvirt-pflash0-storage"}' \
-blockdev '{"driver":"file","filename":"/var/lib/libvirt/qemu/nvram/13_HomeAssistantApp_VARS.fd","node-name":"libvirt-pflash1-storage","auto-read-only":true,"discard":"unmap"}' \
-blockdev '{"node-name":"libvirt-pflash1-format","read-only":false,"driver":"raw","file":"libvirt-pflash1-storage"}' \
-machine pc-i440fx-5.2,accel=kvm,usb=off,dump-guest-core=off,pflash0=libvirt-pflash0-format,pflash1=libvirt-pflash1-format,memory-backend=pc.ram \
-cpu Broadwell-IBRS,vme=on,ss=on,vmx=on,pdcm=on,f16c=on,rdrand=on,hypervisor=on,arat=on,tsc-adjust=on,umip=on,md-clear=on,stibp=on,arch-capabilities=on,ssbd=on,xsaveopt=on,pdpe1gb=on,abm=on,ibpb=on,amd-stibp=on,amd-ssbd=on,skip-l1dfl-vmentry=on,pschange-mc-no=on \
-m 4096 \
-object memory-backend-ram,id=pc.ram,size=4294967296 \
-overcommit mem-lock=off \
-smp 4,sockets=1,dies=1,cores=2,threads=2 \
-uuid df6c288c-9edb-4afc-9c73-542bc1de9e3d \
-no-user-config \
-nodefaults \
-chardev socket,id=charmonitor,fd=42,server,nowait \
-mon chardev=charmonitor,id=monitor,mode=control \
-rtc base=utc \
-no-shutdown \
-boot strict=on \
-device nec-usb-xhci,id=usb,bus=pci.0,addr=0x4 \
-device virtio-serial-pci,id=virtio-serial0,bus=pci.0,addr=0x5 \
-blockdev '{"driver":"host_device","filename":"/dev/zvol/pool/homeassistant","aio":"threads","node-name":"libvirt-1-storage","cache":{"direct":true,"no-flush":false},"auto-read-only":true,"discard":"unmap"}' \
-blockdev '{"node-name":"libvirt-1-format","read-only":false,"cache":{"direct":true,"no-flush":false},"driver":"raw","file":"libvirt-1-storage"}' \
-device virtio-blk-pci,bus=pci.0,addr=0x6,drive=libvirt-1-format,id=virtio-disk0,bootindex=1,write-cache=on \
-netdev tap,fd=47,id=hostnet0,vhost=on,vhostfd=48 \
-device virtio-net-pci,netdev=hostnet0,id=net0,mac=REDACTED,bus=pci.0,addr=0x3 \
-chardev pty,id=charserial0 \
-device isa-serial,chardev=charserial0,id=serial0 \
-chardev spicevmc,id=charchannel0,name=vdagent \
-device virtserialport,bus=virtio-serial0.0,nr=1,chardev=charchannel0,id=channel0,name=com.redhat.spice.0 \
-chardev socket,id=charchannel1,fd=49,server,nowait \
-device virtserialport,bus=virtio-serial0.0,nr=2,chardev=charchannel1,id=channel1,name=org.qemu.guest_agent.0 \
-device usb-tablet,id=input0,bus=usb.0,port=1 \
-spice port=5902,addr=0.0.0.0,seamless-migration=on \
-device qxl-vga,id=video0,ram_size=67108864,vram_size=67108864,vram64_size_mb=0,vgamem_mb=16,max_outputs=1,xres=1024,yres=768,bus=pci.0,addr=0x2 \
-device virtio-balloon-pci,id=balloon0,bus=pci.0,addr=0x7 \
-sandbox on,obsolete=deny,elevateprivileges=deny,spawn=deny,resourcecontrol=deny \
-msg timestamp=on
char device redirected to /dev/pts/2 (label charserial0)
```

bradarnold, 2023-11-02:
I am experiencing the same issue after allowing HAOS to update last night.

chuckfy, 2023-11-02:
> I am experiencing the same issue after allowing HAOS to update last night.

Can you add from which version you updated? Since I didn't shutdown the VM for month I can say for sure which was the last working version.

agners, 2023-11-02:
> generic-x86-64 (Generic UEFI capable x86-64 systems)

Are you truly using this image? We have an image optimized for virtualization named `ova`. This should be used on virtual machines.

I am not familiar with TrueNAS, what type of Hypervisor is underneath?

> [EFAULT] Unable to define domain for homeassistant: cannot serialize None (type NoneType)

To me this looks like an error message from TrueNAS itself, I don't think that this is caused by Home Assistant OS.

chuckfy, 2023-11-02:
> > generic-x86-64 (Generic UEFI capable x86-64 systems)
> 
> Are you truly using this image? We have an image optimized for virtualization named `ova`. This should be used on virtual machines.
I wondered that myself but didn't figure out that there was a drop down menu during creation of the report. And afterwards it wasn't clear whether the information came from a copied log or somewhere else.

See the link I posted, it uses the ova image and I checked with the system info:
`Board | ova`

> I am not familiar with TrueNAS, what type of Hypervisor is underneath?
KVM
> > [EFAULT] Unable to define domain for homeassistant: cannot serialize None (type NoneType)
> 
> To me this looks like an error message from TrueNAS itself, I don't think that this is caused by Home Assistant OS.
As I [linked](https://ixsystems.atlassian.net/browse/NAS-124992), I also created a ticket over there and cross linked this report there.
However, as others have stated in my linked thread: 
1) the issue seemed to have appeared after updating home assistant
2) home assistant is the only VM affected

Interestingly enough I was not able to reproduce the error on another machine using the exact same home assistant zvol. On the other hand I couldn't replicate my system 1:1 although I imported my config.

matthewfurr, 2023-11-15:
I ran into this same problem today. The issue was actually caused by USB device that was added to the VM. Somehow the USB that was previously selected was no longer selected, but still an option. The VM started after adding it back.

[image]

bradarnold, 2023-11-16:
This was also the problem that I found.

md-bo, 2023-11-21:
I haven't rebooted after upgrading to cobia yet, but I couldn't reproduce the issue on another machine with no USB devices attached, so I would assume this seems to have been the culprit.

marcinwasilewski, 2023-12-04:
I have the same problem, not after an update to HAOS but after I plgged out a USB zigbee dongle that was causing some I/O issues on the usb controller. So after  removing the missing USB device  from the settings the VM is coming back it seems

closed 2024-03-12 by github-actions[bot]