home-assistant/operating-system #2909: `CONFIG_VIRTIO_FS` not set in `generic-aarch64` kernel

Skirmisher, 2023-11-06:
### Describe the issue you are experiencing

The kernel configuration for the `generic-aarch64` HAOS image does not include `CONFIG_VIRTIO_FS`, which precludes use of virtiofs when running the image in a VM, e.g. in order to use a host filesystem directory as a data disk.

For reference, the `ova` (x86_64) kernel config [includes the requisite virtiofs options](https://github.com/home-assistant/operating-system/blob/fccb11cad19b21f45f8fc57dad78ca5157b4099e/buildroot-external/board/pc/ova/kernel.config#L46-L52), so it would make sense to enable it in the `generic-aarch64` board as well, given that it is targeted at VMs to some extent.

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Set up a VM and boot `haos_generic-aarch64-11.1.qcow2` with your qemu frontend of choice
2. Obtain `/proc/config.gz`  from the running VM
3. Observe that `zgrep CONFIG_VIRTIO_FS config.gz` produces the output `# CONFIG_VIRTIO_FS is not set`

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

closed 2023-11-06 by github-actions[bot]

agners, 2023-11-06:
Forgot that this label will post that message and close the issue  :sweat_smile: 

I created a PR to address this, so no need to open a feature request.

agners, 2023-11-07:
@Skirmisher the latest nightly builds have the drivers enabled: https://os-artifacts.home-assistant.io/index.html.

Skirmisher, 2023-11-07:
Great, thanks for the quick turnaround :)