home-assistant/operating-system #2916: HA OS causes Cloud Director to partially crash

patschi, 2023-11-09:
### Describe the issue you are experiencing

This is a very weird issue, I've been facing in my environment since deployment of HomeAssistant. I initially thought having some IP conflict, as VMware Cloud Director (essentially for managing virtual datacenters) was constantly loosing connectivity to VMware vCenter, but turned out it is related to something else.

When HA OVA is deployed and started once (VMware vCenter is caching VMware Tools data), it causes Cloud Director services to crash. Having checked the logs, you can see:
```
==> vcloud-container-info.log <==
2023-11-09 19:05:47,698 | WARN     | pc-activity-pool-727      | HttpProtocolBindingBase        | Asynchronous execution requested but no Executor configured. The request will be executed as synchronous one. | activity=(com.vmware.vcloud.vimproxy.internal.impl.PropertyCollectorReader,urn:uuid:6708a8ae-8a64-3f7e-8d06-f060964aea0a)
2023-11-09 19:05:48,202 | ERROR    | pc-activity-pool-727      | JDBCExceptionReporter          | Batch entry 0 /* Method: unknown */ /* Method: unknown */ /* insert com.vmware.vcloud.common.model.inventory.VirtualMachineInvModel */ insert into vm_inv (deleted_date, age, change_version, cloud_uuid, configured_guest_id, connection_state, cpu_limit, cpu_reservation, is_deleted, is_encrypted, firmware, hw_version, has_question, host_moref, instance_uuid, is_vrouter, managed_by, mem, mem_limit, mem_overhead, mem_reservation, moref, name, parent_moref, resource_pool_moref, role, state, is_template, tools_guest_full_name, tools_guest_id, tools_version_status, is_tpm_present, vcpu_count, vc_id, vm_path_name, vm_tools_version, vm_inv_id) values (NULL, 0, '2023-11-09T16:15:01.692394Z', NULL, 'otherGuest64', 2, -1, 0, 'FALSE', 'FALSE', 'efi', 10, 'FALSE', 'host-129618', '502fb350-4b8c-e7fe-6682-c243b865dcf0', 'FALSE', NULL, 2048, -1, 62074880, 0, 'vm-204616', 'kbhomeassistant', 'group-v3', 'resgroup-129624', NULL, 1, 'FALSE', 'Linux 6.1.59 Home Assistant OS 11.1 (Open Virtual Appliance) 11.1 Home Assistant OS 11.1 cpe:2.3:o:home-assistant:haos:11.1:*:production:*:*:*:ova:*', '', 'guestToolsUnmanaged', 'FALSE', 2, '64eb082b-5210-49bf-bda9-74cc50b8e189'::uuid, '[vsanDatastore] 66a94a65-fbf2-7391-07b5-1402ec8c2ddc/kbhomeassistant.vmx', '12384', '04cc5caf-507d-4c27-9458-3a41478bafee'::uuid) was aborted: ERROR: value too long for type character varying(128)  Call getNextException to see other errors in the batch. |
2023-11-09 19:05:48,202 | ERROR    | pc-activity-pool-727      | JDBCExceptionReporter          | ERROR: value too long for type character varying(128) |
```

In this case the extremely unusual (and personally never-seen) long Operating System value of `'Linux 6.1.59 Home Assistant OS 11.1 (Open Virtual Appliance) 11.1 Home Assistant OS 11.1 cpe:2.3:o:home-assistant:haos:11.1:*:production:*:*:*:ova:*'` with 148 characters is exceeding the maximum database column and causing the database transaction to fail.

As this is the longest "Guest Operating System Name" I've ever seen in dozens of thousands of VMs and many environments, I think it might be best to align this reported Guest OS name to something "normal" and more seen in real-life scenarios.

I'd propose to simply report `Home Assistant OS 11.1` back.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Deploy and start HA-OVA on ESXi at least once

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

### System information

n/a

sairon, 2023-11-16:
I'm not familiar with VMware Cloud Director, where does it take this value from? It obviously contains almost all the content of the `os-release` file plus the Linux version as `uname -sr`, which is something I can't find we're combining together anywhere. If we do, we could absolutely reduce it to something less cluttered (e.g. without the `CPE_NAME` and redundant information) but by all means if it is crashing the VMware tooling, it's a bug that should be addressed there.

patschi, 2023-11-16:
> I'm not familiar with VMware Cloud Director

Simply said - it is a complex software solution for cloud providers to manage virtual datacenters. It is also indexing all VMs in the cluster, containing Home-Assistant VM where it is exceeding the DB row.

> It obviously contains almost all the content of the `os-release` file plus the Linux version as `uname -sr`, which is something I can't find we're combining together anywhere.

I was also not able to find where this value comes from. It must be reported through open-vm-tools (most likely) inside the VM from somewhere. I'd assume the naming is retrieved somewhere along here: https://github.com/vmware/open-vm-tools/blob/80904f67eff118fbc1338d1db764994bbb7bad02/open-vm-tools/services/plugins/guestInfo/guestInfoServer.c#L517

But I guess it doesn't help much...

> but by all means if it is crashing the VMware tooling, it's a bug that should be addressed there.

I do agree. Especially, as per open-vm-tools (and ESXi itself) the maximum character limit for the Guest OS-reported name is 512 - whereas Cloud Director's database is limited to 128:
https://github.com/vmware/open-vm-tools/blob/80904f67eff118fbc1338d1db764994bbb7bad02/open-vm-tools/lib/include/hostinfo.h#L43

But beyond that, I think having more concise, user-friendly operating system-reported name and having a more quicker fix, it would be good to adjust the reported value.

patschi, 2023-11-16:
I think I found the line - it seems indeed using the `os-release` file:
https://github.com/vmware/open-vm-tools/blob/80904f67eff118fbc1338d1db764994bbb7bad02/open-vm-tools/lib/misc/hostinfoPosix.c#L724

Additionally, I found a way manually specifying/overriding the reported Guest OS name:
```
$ cat /etc/vmware-tools/tools.conf
[guestosinfo]
short-name=Home Assistant OS 11.1
long-name=Home Assistant OS 11.1
```

This will result in:
[image]

So during build process the `tools.conf` could be generated like above and report a user-friendly operating system name to the hypervisor.

patschi, 2023-11-20:
As mentioned above in https://github.com/home-assistant/operating-system/issues/2916#issuecomment-1814312589, it is possible to overwrite the values what is being reported by open-vm-tools to the GuestOS. So the `tools.conf` can be written during build process.

agners, 2023-11-20:
I'd prefer if we don't have to maintain additional configuration file generation in HAOS. This seems something other operating systems could have issues with too. If it is possible to teach open-vm-tools to come up with more sensible defaults, I think that would be the better approach.

closed 2024-02-26 by github-actions[bot]