home-assistant/operating-system #2918: Hostname homeassistant.local not resolved + errors in avahi browser

prog8, 2023-11-10:
### Describe the issue you are experiencing

I have home assistant running in my local network. OS is running on virtual machine on Qemu. After changing network settings through UI (switching IPv4 settings from static to dhcp) there is homeassistant.local visible through `avahi-browse -alrt`. I run this on another machine in the local network.

```
=   eno1 IPv4 homeassistant [5bc0a1a4ee0b490123123108cc6f55] Workstation          local
   hostname = [homeassistant.local]
   address = [192.168.0.59]
```

When this hostname is properly resolved the same time `avahi-browse` reports a set of errors as below

```
Failed to resolve service '_homeconnect' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_home-assistant' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_workstation' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_workstation' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_homeconnect' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_smb' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_wd' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_privet' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_uscans' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_uscan' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_http-alt' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_scanner' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_printer' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_pdl-datastream' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_ipp' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_http' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_home-assistant' of type '_tcp' in domain 'local': Invalid service type
Failed to resolve service '_hap' of type '_tcp' in domain 'local': Invalid service type
```

But after a couple minutes `homeassistant.local` disappears for IPv4 and so the errors in `avahi-browse`. `homeassistant.local` is still visible for IPv6

```
=   eno1 IPv6 homeassistant [5bc0a1a4ee0b490123123108cc6f55] Workstation          local
   hostname = [homeassistant.local]
   address = [fe80::ce52:7b89:2ca5:ebe
```

Is it some misconfiguration of Homeassistant OS  or a bug?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start Operating system in Qemu installed on Debian
2. Start `avahi-browse` on another machine and see `homeassistant.local` being properly resolved
3. Wait N minutes
4. Start `avahi-browse` on another machine again and see no more `homeassistant.local` resolved on IPv4

### Anything in the Supervisor logs that might be useful for us?

```txt
ov 10 07:31:39 homeassistant NetworkManager[361]: <info>  [1699601499.7164] audit: op="connection-update" uuid="29791942-35e8-36bb-a530-67dfd3601095" name="Supervisor enp1s0" args="connection.timestamp,ipv4.addresses,ipv4.method,ipv4.dns,ipv4.gateway" pid=129946 uid=0 result="success"
Nov 10 07:31:39 homeassistant NetworkManager[361]: <info>  [1699601499.7202] device (enp1s0): state change: activated -> deactivating (reason 'new-activation', sys-iface-state: 'managed')
Nov 10 07:31:39 homeassistant NetworkManager[361]: <info>  [1699601499.7209] manager: NetworkManager state is now DISCONNECTING
Nov 10 07:31:39 homeassistant NetworkManager[361]: <info>  [1699601499.7239] device (enp1s0): disconnecting for new activation request.
Nov 10 07:31:39 homeassistant NetworkManager[361]: <info>  [1699601499.7248] audit: op="connection-activate" uuid="29791942-35e8-36bb-a530-67dfd3601095" name="Supervisor enp1s0" pid=129946 uid=0 result="success"
[... 147 lines cut ...]
version_api | ok
installed_addons | Terminal & SSH (9.7.1), NGINX Home Assistant SSL proxy (3.5.0), Duck DNS (1.15.0), File editor (5.6.0), Google Assistant SDK (2.5.0), Samba Backup (5.2.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 31 października 2023 09:19
-- | --
current_recorder_run | 8 listopada 2023 18:42
estimated_db_size | 10.70 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

agners, 2023-12-05:
> OS is running on virtual machine on Qemu.

> generic-x86-64 (Generic UEFI capable x86-64 systems)

We have a specific image for virtualization, named "ova". I recommend using this over generic-x86-64.

I am not very familiar with `avahi-browse`, but from what I understand it is mostly useful to browse services. What you attempt is to resolve host name. Both, services and host name resolving use the mDNS mechanism to communicate, but these things are not the same. Maybe `avahi-browse` output just hides certain information/messages? :thinking: 

To test if Home Assistant properly announces IPv4 and IPv6 addressing via mDNS try using:
```
avahi-resolve -4 --name homeassistant.local
```

```
avahi-resolve -6 --name homeassistant.local
```

FWIW, HAOS uses systemd as mDNS responder.

prog8, 2023-12-05:
That's what I get this

```
$ avahi-resolve -4 --name homeassistant.local                  
Failed to resolve host name 'homeassistant.local': Timeout reached
$ avahi-resolve -6 --name homeassistant.local                  
homeassistant.local	fe80::ce52:7b89:2ca5:ebe2
```

But then after I reboot machine I see IP address for IPv4 for a moment

```
$ avahi-resolve -4 --name homeassistant.local                  
homeassistant.local	192.168.0.59
```

And then it disappears again.

prog8, 2023-12-05:
I don't have any more instances but logs you requested look very suspicious

```
Dec 05 11:25:51 homeassistant systemd-resolved[345]: enp1s0: Bus client set default route setting: yes
Dec 05 11:25:51 homeassistant systemd-resolved[345]: enp1s0: Bus client set MulticastDNS setting: yes
Dec 05 11:25:51 homeassistant systemd-resolved[345]: enp1s0: Bus client set DNS server list to: 8.8.8.8
Dec 05 11:25:59 homeassistant systemd-resolved[345]: enp1s0: Bus client set default route setting: no
Dec 05 11:25:59 homeassistant systemd-resolved[345]: enp1s0: Bus client set MulticastDNS setting: no
Dec 05 11:25:59 homeassistant systemd-resolved[345]: enp1s0: Bus client reset DNS server list.
Dec 05 11:25:59 homeassistant systemd-resolved[345]: enp1s0: Bus client set default route setting: yes
Dec 05 11:25:59 homeassistant systemd-resolved[345]: enp1s0: Bus client set MulticastDNS setting: yes
Dec 05 11:25:59 homeassistant systemd-resolved[345]: enp1s0: Bus client set DNS server list to: 8.8.8.8
```

So it seems `MulticastDNS setting: yes` is then switched after a couple seconds to  `MulticastDNS setting: no`. Could it be our problem? If yes how the setting could have been changed automatically?

agners, 2023-12-06:
I do have similar logs, but it always ends with a `yes`, so I don't think that is the problem. :thinking: 

```
Nov 28 17:54:24 ha-shelf1-rpi3plus systemd-resolved[391]: eth0: Bus client set MulticastDNS setting: no
Nov 28 17:54:24 ha-shelf1-rpi3plus systemd-resolved[391]: eth0: Bus client reset DNS server list.
Nov 28 17:54:24 ha-shelf1-rpi3plus systemd-resolved[391]: eth0: Bus client set MulticastDNS setting: yes
```

agners, 2023-12-06:
Can you change the hostname and see if that changes the behavior for you (under Settings > System > Network)?

prog8, 2023-12-06:
Hi @agners thank you for new suggestions. I changed the hostname to `ha` but still doesn't see `ha.local` in my network. The funny thing is that I see this `hostname = [35f8df4618044e8ab08ae504f933aaac.local]   address = [192.168.0.59]` which points to home assistant. Not sure what this identifier is about but for sure is related to HA.

closed 2024-03-14 by github-actions[bot]