home-assistant/operating-system #2920: Supervisor 2023.11.0 update problem

jensjakobandersen, 2023-11-12:
### Describe the issue you are experiencing

When I try to update to Supervisor 23.11.0 I get an error.

Error: Update of Supervisor failed: Can't install ghcr.io/jhome-assistant/aarch64-hassio-supervisor:2023.11.0: 404 Client Error for http+docker://localhost/v1.43/image/ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.11.0/json: Not found ("No such image: ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.11.0")

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

11.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Either run the update from the webpage - or open terminal and run ha supervisor update

### Anything in the Supervisor logs that might be useful for us?

```txt
23-11-12 13:35:53 INFO (MainThread) [__main__] Running Supervisor
23-11-12 13:35:53 INFO (MainThread) [supervisor.os.manager] Rauc: B - marked slot kernel.1 as good
23-11-12 13:35:53 INFO (MainThread) [supervisor.supervisor] Fetching AppArmor profile https://version.home-assistant.io/apparmor_stable.txt
23-11-12 13:35:53 INFO (MainThread) [supervisor.host.apparmor] Adding/updating AppArmor profile: hassio-supervisor
23-11-12 13:35:54 INFO (MainThread) [supervisor.supervisor] Update Supervisor to version 2023.11.0
23-11-12 13:35:54 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/aarch64-hassio-supervisor with tag 2023.11.0.
23-11-12 13:35:58 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state startup
23-11-12 13:35:58 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-11-12 13:35:58 WARNING (MainThread) [supervisor.jobs] 'StoreManager.reload' blocked from execution, supervisor needs to be updated first
23-11-12 13:36:03 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-11-12 13:36:13 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
23-11-12 13:36:16 ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.11.0: 404 Client Error for http+docker://localhost/v1.43/images/ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.11.0/json: Not Found ("No such image: ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.11.0")
23-11-12 13:36:16 INFO (MainThread) [supervisor.resolution.module] Create new issue update_failed - supervisor / None
23-11-12 13:36:16 ERROR (MainThread) [supervisor.supervisor] Update of Supervisor failed: Can't install ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.11.0: 404 Client Error for http+docker://localhost/v1.43/images/ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.11.0/json: Not Found ("No such image: ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.11.0")
23-11-12 13:36:16 CRITICAL (MainThread) [supervisor.core] Can't update Supervisor! This will break some Add-ons or affect future versions of Home Assistant!
23-11-12 13:36:16 INFO (MainThread) [supervisor.addons] Phase 'initialize' starting 0 add-ons
23-11-12 13:36:16 INFO (MainThread) [supervisor.addons] Phase 'system' starting 0 add-ons
23-11-12 13:36:16 INFO (MainThread) [supervisor.addons] Phase 'services' starting 1 add-ons
23-11-12 13:36:16 INFO (SyncWorker_3) [supervisor.docker.manager] Cleaning addon_core_deconz application
23-11-12 13:36:18 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-deconz with version 6.20.0
```

### Anything in the Host logs that might be useful for us?

```txt
Nov 12 12:35:58 Plantagen kernel: Bluetooth: SCO socket layer initialized
Nov 12 12:36:16 Plantagen dockerd[505]: time="2023-11-12T12:36:16.580042663Z" level=info msg="Attempting next endpoint for pull after error: failed to register layer: error creating overlay mount to /mnt/data/docker/overlay2/e97c1b5082447c6300adb0ddbfba5a617ab9a6fdd0eabf35577711e5f92690e8/merged: too many levels of symbolic links"
Nov 12 12:36:17 Plantagen systemd[1]: var-lib-docker-overlay2-3a947bccecfebf4a723ee993134044855c8a269073c54bba73c77a0a3ea66790\x2dinit-merged.mount: Deactivated successfully.
Nov 12 12:36:17 Plantagen systemd[1]: mnt-data-docker-overlay2-3a947bccecfebf4a723ee993134044855c8a269073c54bba73c77a0a3ea66790\x2dinit-merged.mount: Deactivated successfully.
Nov 12 12:36:17 Plantagen NetworkManager[434]: <info>  [1699792577.8184] manager: (vethdd84af4): new Veth device (/org/freedesktop/NetworkManager/Devices/18)
Nov 12 12:36:17 Plantagen kernel: hassio: port 6(vethe90f92e) entered blocking state
Nov 12 12:36:17 Plantagen kernel: hassio: port 6(vethe90f92e) entered disabled state
Nov 12 12:36:17 Plantagen kernel: device vethe90f92e entered promiscuous mode
Nov 12 12:36:17 Plantagen kernel: kauditd_printk_skb: 147 callbacks suppressed
Nov 12 12:36:17 Plantagen kernel: audit: type=1700 audit(1699792577.815:183): dev=vethe90f92e prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
Nov 12 12:36:17 Plantagen kernel: audit: type=1300 audit(1699792577.815:183): arch=c00000b7 syscall=206 success=yes exit=40 a0=c a1=4001427aa0 a2=28 a3=0 items=0 ppid=1 pid=505 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj=unconfined key=(null)
Nov 12 12:36:17 Plantagen kernel: audit: type=1327 audit(1699792577.815:183): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D636F6E7461696E6572643D2F72756E2F636F6E7461696E6572642F636F6E7461696E6572642E736F636B
Nov 12 12:36:17 Plantagen NetworkManager[434]: <info>  [1699792577.8206] manager: (vethe90f92e): new Veth device (/org/freedesktop/NetworkManager/Devices/19)
Nov 12 12:36:17 Plantagen dockerd[505]: time="2023-11-12T12:36:17.834909639Z" level=warning msg="Failed to delete conntrack state for 172.30.33.0: invalid argument"
Nov 12 12:36:17 Plantagen systemd[1]: Started libcontainer container 439b234b662f2464192f2d2ec7858319b7442f84e5f11b2afa29b00ae35d6fb8.
Nov 12 12:36:17 Plantagen kernel: audit: type=1334 audit(1699792577.975:184): prog-id=45 op=LOAD
Nov 12 12:36:17 Plantagen kernel: audit: type=1334 audit(1699792577.979:185): prog-id=46 op=LOAD
```

### Additional information

## System Information

version | core-2023.10.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.56
arch | aarch64
timezone | Europe/Copenhagen
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1394
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | December 12, 2023 at 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | eu-central-1-6.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.1
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 28.0 GB
disk_used | 6.1 GB
healthy | failed to load: Unhealthy
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | deCONZ (6.20.0), Terminal & SSH (9.7.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 16
-- | --
resources | 2
views | 16
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | November 2, 2023 at 10:09
-- | --
current_recorder_run | November 12, 2023 at 13:35
estimated_db_size | 151.58 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.3 (a8c6d45)
-- | --
cloud_online | 7 / 9
local_online | 0 / 0
debug | failed to load: 

</details>

erxaz, 2023-11-13:
I got the exakt same issue.

GJSchroeder, 2023-11-13:
I have the same issue as well.  Running on Home Assistant Yellow.  It happened right after I migrated from running 2023.10.5 on a Pi to the Yellow.  Not sure whether that is related or just a coincidence.

jensjakobandersen, 2023-11-13:
I am sorry - I realised this should probably have been created over at the Supervisor - so I have created it there as well:
https://github.com/home-assistant/supervisor/issues/4703

closed 2023-11-13 by github-actions[bot]