home-assistant/operating-system #2926: Issue with Netgear A8000 / 6E AXE3000

tessharp, 2023-11-14:
### Describe the issue you are experiencing

I am trying to use the Netgear A8000 USB adapter with my Home Assistant OS, but unfortunately it does not work. Issue is that it uses the mt7921au, which I understand works with the mt7921u driver, but needs its VID/PID to be added to the Kernel in versions older than 6.4.

I chose my adapter based on its feature-set and thinking it would not be an issue to add the necessary configuration to the HAOS... but I was wrong.

Links I've been reading to choose the adapter:
* https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md
* https://github.com/morrownr/USB-WiFi/issues/87

Note: I'm still running 9.5 because I'm looking to migrate from an old adapter that no longer works with newer versions of Home Assistant based off the Realtek 8188 chipset

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Updated Home Assistant OS from 9.5 to 11.1
2. Connected the adapter
3. Checked on the local console with "nmcli device" to see if a "wifi" device would show up
4. No "wifi" devices were available
...

### Anything in the Supervisor logs that might be useful for us?

```txt
None of assistance.
```

### Anything in the Host logs that might be useful for us?

```txt
None of assistance.
```

### System information

Hyper-V image with PCI passthrough for USB devices.

sairon, 2023-11-15:
You can try running this in the root console:
```
modprobe mt7921u
echo 0846 9060 > /sys/bus/usb/drivers/mt7921u/new_id
```
That way you should be able to check it on the running system at least. Newer kernels contain bunch of patches for `mt7921u` and I'm not how your card will perform with 6.1, yet adding the support for your device to the driver is just about adding its record to the device table (https://github.com/torvalds/linux/commit/03eb52dd78cab08f13925aeec8315fbdbcba3253), which we can backport.

tessharp, 2023-11-15:
Do I do this through the Ash shell at the "login" prompt?

sairon, 2023-11-16:
Yes, correct, you must run this command as the root user in HAOS directly.

tessharp, 2023-11-21:
I got an error message after updating to HA OS 10.5
`modprobe: FATAL: Module mt7921u not found in directory /lib/modules/6.1.45`

sairon, 2023-11-22:
D'oh, sorry for that, I was looking at a wrong place - that driver is not enabled yet in any of the releases. But a release with bunch of extra WiFi drivers enabled is around the corner, I'll make sure to add a patch with adding A8000 to the device table, so you could try it then.

tessharp, 2023-11-22:
no worries, will be looking forward to that release!

closed 2023-12-04 by sairon

tessharp, 2024-01-11:
Just tested this again after the holiday break, and unfortunately is still not working for me. Should I expect to use ```modprobe mt7921u``` to load the module/driver? Or should I expect for it to be plug-and-play?

Updated to the latest 11.4 HA OS release, where modprobe does work, but nothing happens with ```nmcli```

PR #2962 (2023-12-04): Backport support for Netgear AXE3000/A8000 and Comfast CF-952AX
Current mainline contains support for two more WiFi cards in the mt7921u driver that only use a proprietary VID/PID but are compatible with the standard driver. Backport support for those via a simple driver patch.

Fixes #2926
commit 934273e3 (2024-01-04): Backport support for Netgear AXE3000/A8000 and Comfast CF-952AX (#2962)

Current mainline contains support for two more WiFi cards in the mt7921u
driver that only use a proprietary VID/PID but are compatible with the
standard driver. Backport support for those via a simple driver patch.

Fixes #2926