home-assistant/operating-system #2932: HA OS 11.1 Fresh Install Raspberry Pi 4 USB SSD Disk boot loop

mlinko6, 2023-11-16:
### Describe the issue you are experiencing

The installation gets stuck in a boot loop and there is only black screen, I've tried it three times, but I didn't have any success.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.
2.
3.
...
Fresh install of the Image "haos_rpi4-64-11.1.img.xz" it works on the SD Card, but it doesn't work on USB SSD Disk. I've also copied the SD Card to the SSD Disk with the Image copier but it didn't work either - similar issue: black screen at the end.

Errors from starting the SSD Disk with the SD Card image:

[image]
[image]

### Anything in the Supervisor logs that might be useful for us?

```txt
no logs to get
```

### Anything in the Host logs that might be useful for us?

```txt
no logs to get
```

### System information

Raspberry Pi 4

sairon, 2023-11-22:
What model of the USB disk are you using? Can you try plugging it to a different PC and getting its USB VID/PID?

mlinko6, 2023-11-22:
Hi Jan,

Thx for the response. I’m using a Samsung Pro 128GB SSD - as soon as I connected the SSD Drive to USB 2.0 it was working without as Problem… I assume that the disk drains more “power” over 3.0 USB and the Pi is not willing to “give” it… I’m using the original Pi usb C charger… I didn’t investigate any further…

KR
Rok

Coremac, 2023-12-09:
I have the same Issue, boot from SSD only works on the USB 2.0 ports since the update. Same issue with a differen adapter.

closed 2024-03-16 by github-actions[bot]