home-assistant/operating-system #2941: Zigbee completely stopped working, seem related to access to /dev/ttyAMA0

marteno, 2023-11-21:
### Describe the issue you are experiencing

Hi

Running home assistant with home assistant os on Raspberry pi 3 and a zigbee shield using uart.
 
Updated to new 11.2rc1 earlier today and during the evening noticed that lights connected to zigbee network didn't turn on.
Got an exclamation mark in the ZHA integration right next to the /dev/ttyAMA0 serial port.

SSH-ing into home assistant and doing ls -l /dev/ttyAMA0.
Unfortunately I don't have the output saved but think it was only write permissions for the group and group set to dialout.

Downgrading to 11.1 made it worked again and looking at the device file the permissions differ.
crw-rw----    1 root     audio     204,  64 Nov 21 19:38 /dev/ttyAMA0

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.2rc

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. update to home assistant os 11.2rc
2. see that all zigbee devices are unavailable
3. ssh into box at check permissions on /dev/ttyAMA0 file
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No, doesn't seem to be any logs available from before the downgrade
```

### Anything in the Host logs that might be useful for us?

```txt
No, doesn't seem to be any logs available from before the downgrade
```

### System information

## System Information

version | core-2023.11.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4875
Installed Version | 1.33.0
Stage | running
Available Repositories | 1338
Downloaded Repositories | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | beta
supervisor_version | supervisor-2023.11.4
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 57.8 GB
disk_used | 9.3 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.8.1), File editor (5.7.0), Home Panel (2.7.0), Advanced SSH & Web Terminal (16.0.1), Let's Encrypt (4.12.9), WireGuard (0.9.0), Dynu DNS (1.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 10
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 10 november 2023 kl. 19:02
-- | --
current_recorder_run | 21 november 2023 kl. 19:04
estimated_db_size | 110.28 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Svensk Postutdelning</summary>

Version | 1.0.7
-- | --
Idle | false
Instances | 1

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.13
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 0
total_devices | 0

</details>

sairon, 2023-11-22:
It's unlikely the different permissions would be the problem, since all the related components run as root. Currently I can only guess what was the reason, could you please kindly try reproducing the error and sending the logs, especially from the Host and Core?

agners, 2023-11-23:
Can you share the `config.txt` of your board? Did you enable the `dtoverlay=miniuart-bt` line?

marteno, 2023-11-26:
[host_2023-11-26T22-20-41.141Z.log](https://github.com/home-assistant/operating-system/files/13468345/host_2023-11-26T22-20-41.141Z.log)
[home-assistant_2023-11-26T22-19-41.873Z.log](https://github.com/home-assistant/operating-system/files/13468346/home-assistant_2023-11-26T22-19-41.873Z.log)
[haos-error-dmesg.txt](https://github.com/home-assistant/operating-system/files/13468347/haos-error-dmesg.txt)
[config.txt](https://github.com/home-assistant/operating-system/files/13468348/config.txt)
[cmdline.txt](https://github.com/home-assistant/operating-system/files/13468349/cmdline.txt)

Sorry for the delay, but I finally got a chance to sit down with this.
The miniuart-bt looks enabled on line 66 in config.txt

Also in home-assistant_2023-11-26T22-19-41.873Z.log the lines 12-13 and 18-37

marteno, 2023-11-26:
[host_2023-11-26T23-00-58.421Z.log](https://github.com/home-assistant/operating-system/files/13468366/host_2023-11-26T23-00-58.421Z.log)
[home-assistant_2023-11-26T23-00-33.821Z.log](https://github.com/home-assistant/operating-system/files/13468367/home-assistant_2023-11-26T23-00-33.821Z.log)

Host and core logs when running os version 11,1 as reference.
BR,
Mårten

marteno, 2023-11-28:
Hi

Adding the console-config as suggested helped.
Not sure why it was missing, obviously I made a mistake at some point and it got removed.
Sorry for the noise

BR,
Mårten

sairon, 2023-11-29:
Perfect, thanks for the update!

closed 2023-11-29 by sairon

adrianrdzv, 2023-12-05:
In my case I didn't have the line, I added it but still no luck.
```
dwc_otg.lpm_enable=0
console=tty1
```
and
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# HassOS - don't change it!
disable_splash=1
kernel=u-boot.bin

# uncomment for aarch64 bit support
#arm_64bit=1

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[all]
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

enable_uart=1
dtoverlay=pi3-miniuart-bt
```

DarrenBishop, 2024-03-02:
I've been struggling with this for a while; can't even say for how long as I've been travelling, but noticed something was up when LEDs weren't coming on at night.

I believe this was triggered after some Core updates.

Tried  `lsof | grep /dev/ttyAMA0` and found that agetty was using th serial port.

Addedd `console=tty1` to the end of `cmdline.txt` and after a reboot, everything started working again.

Thank you @sairon 

Going to reapply updates now and see how it goes.