home-assistant/operating-system #2945: Some automations unavailable after 11.x, must use 10.5

MrRudeDog, 2023-11-26:
### Describe the issue you are experiencing

Whenever I try to update to 11.x, I lose access to 3 or 4 automatons related to a 4 button zigbee device. The automatons are listed as unavailable. Automatons assigned to some of the buttons work, some don't. From the log files:
Logger: homeassistant.components.zha.core.cluster_handlers
Source: components/zha/core/cluster_handlers/__init__.py:537
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 10:09:50 AM (12 occurrences)
Last logged: 10:10:30 AM

[0x4357:1:0x0008]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>')]
[0x0222:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.MAC_NO_ACK: 233>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_NO_ACK: 233>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_NO_ACK: 233>')]
[0x0222:1:0x0b04]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.MAC_NO_ACK: 233>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_CHANNEL_ACCESS_FAILURE: 225>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_NO_ACK: 233>'), DeliveryError('Failed to deliver packet: <TXStatus.MAC_NO_ACK: 233>')]
[0x8E8C:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>')]
[0x4357:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>')]

And this:
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:268
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 10:11:44 AM (1 occurrences)
Last logged: 10:11:44 AM

[139791751407680] Error handling message: Unknown device 'f1ac07dbaa98b7f9de3bb094abfbecdf' (unknown_error) mrrudedog from 127.0.0.1 (Mozilla/5.0 (Linux; Android 13; SM-T970 Build/TP1A.220624.014; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/119.0.6045.67 Safari/537.36 Home Assistant/2023.10.2-11484 (Android 13; SM-T970))

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

	Home Assistant OS 11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Any attempt to update to 11.x
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
23-11-26 09:42:14 INFO (MainThread) [supervisor.store] Loading add-ons from store: 73 all - 0 new - 0 remove
23-11-26 09:42:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-11-26 10:02:37 INFO (MainThread) [supervisor.backups.manager] Found 197 backup files
23-11-26 10:02:38 INFO (MainThread) [supervisor.backups.manager] Found 197 backup files
23-11-26 10:02:48 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
23-11-26 10:02:48 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
23-11-26 10:03:17 INFO (MainThread) [supervisor.backups.manager] Removed backup file d0bb70cd
23-11-26 10:03:17 INFO (MainThread) [supervisor.backups.manager] Removed backup file 2caef49c
23-11-26 10:03:17 INFO (MainThread) [supervisor.backups.manager] Removed backup file f4ec7062
23-11-26 10:03:18 INFO (MainThread) [supervisor.backups.manager] Found 194 backup files
23-11-26 10:03:28 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
23-11-26 10:03:28 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
23-11-26 10:03:28 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
23-11-26 10:03:28 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
23-11-26 10:03:44 INFO (MainThread) [supervisor.backups.manager] Backup 8c883148 starting stage addon_repositories
23-11-26 10:03:44 INFO (MainThread) [supervisor.backups.manager] Backup 8c883148 starting stage docker_config
23-11-26 10:03:44 INFO (MainThread) [supervisor.backups.manager] Creating new full backup with slug 8c883148
23-11-26 10:03:44 INFO (MainThread) [supervisor.backups.manager] Backup 8c883148 starting stage addons
23-11-26 10:03:44 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on core_mosquitto
23-11-26 10:03:45 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon core_mosquitto
23-11-26 10:03:45 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on a0d7b954_vscode
23-11-26 10:03:45 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon a0d7b954_vscode
23-11-26 10:03:45 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on core_samba
23-11-26 10:03:45 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon core_samba
[... 62 lines cut ...]
23-11-26 10:09:36 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/brenner-tobias/ha-addons repository
23-11-26 10:09:36 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
23-11-26 10:09:36 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
23-11-26 10:09:36 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
23-11-26 10:09:36 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
23-11-26 10:09:37 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-11-26 10:09:37 INFO (MainThread) [supervisor.store] Loading add-ons from store: 73 all - 0 new - 0 remove
23-11-26 10:09:37 INFO (MainThread) [supervisor.store] Loading add-ons from store: 73 all - 0 new - 0 remove
23-11-26 10:09:40 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant Core state changed to NOT_RUNNING
23-11-26 10:10:15 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant Core state changed to RUNNING
23-11-26 10:10:15 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
23-11-26 10:10:15 INFO (MainThread) [supervisor.homeassistant.core] Successfully started Home Assistant 2023.11.3
23-11-26 10:10:15 INFO (SyncWorker_5) [supervisor.docker.manager] Cleanup images: ['ghcr.io/home-assistant/generic-x86-64-homeassistant:2023.10.5']
23-11-26 10:12:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
```

### Anything in the Host logs that might be useful for us?

```txt
Posted above, sorry. Noob
```

### Additional information

I am more of a "fight my way through " user than an expert. I am not remotely a programmer.  I have been reverting back to 10.5 OS because i can't figure this out. Please be patient with me if additional information is needed.  Thanks

sairon, 2023-11-27:
Are we really talking about the difference between OS versions? In the Supervisor logs I see you upgraded from version 2023.10.5 to 2023.11.3 of the `homeassistant` image, which is what we call "Core" and is unrelated to the OS version - although it's coincidentally similar to current OS versions (latest 11.1 and previous 10.5).

MrRudeDog, 2023-11-28:
Okay, I missed that difference between core and OS, thank you for pointing it out. Do I need to start a new thread in the core category or can this one be moved? Thanks again

sairon, 2023-11-29:
Yes, please create a new issue in the [core](https://github.com/home-assistant/core) repository, as the template is a bit different and it will guide you to get the required information and apply correct labels.