home-assistant/operating-system #2952: Zigbee2MQTT Add-on - Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

maulwurf01, 2023-11-29:
### Describe the issue you are experiencing

Hello together,
I am experiencing a weird issue with my Zigbee2MQTT Add-on. I am running this add-on within Home Assistant and have set it up to the guidelines specified in the documentation.
I am using the SONOFF Zigbee 3.0 USB Dongle Plus-P with the lasted Z-stack coordinator firmware (CC1352P2_CC2652P_launchpad_coordinator_20230507).

Whenever I try to start the add-on the following error appears (and thus the add-on does not start):
`Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)`

I have already tried the following:

- Made sure that the correct USB device is selected
- Only have the ZigBee stick plugged in
- Made sure that ZHA is not running and disabled
- Tried to use the /dev/serial/by-id path of the stick
- Re-installed the add-on
- Re-flashed the stick (multiple times) with the above firmware

I have also installed a zigbee2mqtt docker container on my QNAP (different from the host running HAOS). There, the stick is recognized and I can run zigbee2mqtt without problem.

My config for the add-on looks like the following:

<details><summary>Add-on config</summary>
<p>

```typescript
homeassistant: true
advanced:
  network_key:
    - 187
    - 79
    - 204
    - 129
    - 36
    - 29
    - 218
    - 28
    - 247
    - 110
    - 84
    - 13
    - 243
    - 81
    - 50
    - 83
  pan_id: 24677
  ext_pan_id:
    - 174
    - 156
    - 24
    - 139
    - 7
    - 219
    - 131
    - 237
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: xxx
serial:
  port: /dev/ttyUSB1
frontend:
  port: 8099
``` 

</p>
</details> 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Plug-in Zigbee stick
2. Try to start the add-on
3. Look at log files
4. Observe error

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2023.11.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.59
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4999
Installed Version | 1.33.0
Stage | running
Available Repositories | 1347
Downloaded Repositories | 29

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 27. Dezember 2023 um 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-11.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.11.6
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 234.0 GB
disk_used | 12.6 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.14.2), ESPHome (2023.11.4), Node-RED (16.0.2), Terminal & SSH (9.8.1), RaspberryMatic CCU (3.71.12.20231020), InfluxDB (4.8.0), Mosquitto broker (6.4.0), Grafana (9.1.1), Zigbee2MQTT (1.33.2-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 21
views | 19
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 20. November 2023 um 10:52
-- | --
current_recorder_run | 27. November 2023 um 20:11
estimated_db_size | 659.36 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

maulwurf01, 2023-12-04:
> Könnten Sie bitte die Host-Protokolle posten, die kurz nach dem Einstecken des USB-Sticks erhalten wurden? Und sind Sie **wirklich** sicher, dass das richtige Gerät verwendet wird? Ich sehe, du hast `/dev/ttyUSB1`in der Konfiguration, was `/dev/ttyUSB0`?

[image]
[image]
[image]

B0n3s21, 2023-12-05:
Try port: /dev/ttyUSB0

maulwurf01, 2023-12-05:
> Try port: /dev/ttyUSB0 and delete adapter line.

Firstly, thank you for your reply.
I have entered port: /dev/ttyUSB0 but unfortunately without success
I'm in despair

maulwurf01, 2023-12-06:
> LOG erzeugt dieselben Fehlermeldungen?
Yes

[image]

[image]

eronisko, 2023-12-06:
Could you try enabling herdsman debug and posting the logs here?
https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

maulwurf01, 2023-12-08:
Hi, here is the herdmans debug log:
<details><summary>Herdsman Log</summary>
<p>

```shell
[12:59:14] INFO: Preparing to start...
[12:59:14] INFO: Socat not enabled
[12:59:14] INFO: Zigbee Herdsman debug logging enabled
[12:59:15] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-12-08 12:59:16: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-12-08 12:59:16: Logging to console and directory: '/config/zigbee2mqtt/log/2023-12-08.12-59-16' filename: log.txt
Zigbee2MQTT:debug 2023-12-08 12:59:16: Removing old log directory '/config/zigbee2mqtt/log/2023-12-06.12-26-28'
Zigbee2MQTT:info  2023-12-08 12:59:16: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
Zigbee2MQTT:info  2023-12-08 12:59:16: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:debug 2023-12-08 12:59:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[174,156,24,139,7,219,131,237],"networkKey":"HIDDEN","panID":24677},"serialPort":{"path":"/dev/ttyUSB1"}}'
2023-12-08T11:59:16.916Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-12-08T11:59:16.917Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[187,79,204,129,36,29,218,28,247,110,84,13,243,81,50,83],"panID":24677,"extendedPanID":[174,156,24,139,7,219,131,237],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB1"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-12-08T11:59:16.917Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/ttyUSB1","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-12-08T11:59:16.921Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2023-12-08T11:59:16.922Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-12-08T11:59:16.923Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-12-08T11:59:17.175Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2023-12-08T11:59:17.176Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2023-12-08T11:59:18.177Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-12-08T11:59:18.178Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-12-08T11:59:18.430Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352
2023-12-08T11:59:18.887Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-12-08T11:59:18.888Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-12-08T11:59:24.890Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-12-08T11:59:24.890Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-12-08T11:59:30.896Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-12-08T11:59:30.897Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2023-12-08 12:59:36: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-12-08 12:59:36: Failed to start zigbee
Zigbee2MQTT:error 2023-12-08 12:59:36: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-12-08 12:59:36: Exiting...
Zigbee2MQTT:error 2023-12-08 12:59:36: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
``` 

</p>
</details>

eronisko, 2023-12-08:
Have you tried changing the `adapter` line in your `serial` configuration? I'd try setting it to `ezsp` (as per [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/11311#issuecomment-1483721479))  or removing the whole line.

maulwurf01, 2023-12-09:
Hi,
<details><summary>With ezsp</summary>
<p>

```powershell
[10:55:31] INFO: Preparing to start...
[10:55:31] INFO: Socat not enabled
[10:55:31] INFO: Zigbee Herdsman debug logging enabled
[10:55:32] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-12-09 10:55:33: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-12-09 10:55:33: Logging to console and directory: '/config/zigbee2mqtt/log/2023-12-09.10-55-33' filename: log.txt
Zigbee2MQTT:debug 2023-12-09 10:55:33: Removing old log directory '/config/zigbee2mqtt/log/2023-12-06.13-00-39'
Zigbee2MQTT:info  2023-12-09 10:55:33: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
Zigbee2MQTT:info  2023-12-09 10:55:33: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:debug 2023-12-09 10:55:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[174,156,24,139,7,219,131,237],"networkKey":"HIDDEN","panID":24677},"serialPort":{"adapter":"ezsp","path":"/dev/ttyUSB1"}}'
2023-12-09T09:55:33.999Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-12-09T09:55:34.000Z zigbee-herdsman:adapter:ezsp:debg Adapter concurrent: 8
2023-12-09T09:55:34.000Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[187,79,204,129,36,29,218,28,247,110,84,13,243,81,50,83],"panID":24677,"extendedPanID":[174,156,24,139,7,219,131,237],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB1","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-12-09T09:55:34.001Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyUSB1","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-12-09T09:55:34.005Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2023-12-09T09:55:34.005Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2023-12-09T09:55:34.005Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2023-12-09T09:55:34.006Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2023-12-09T09:55:34.006Z zigbee-herdsman:adapter:ezsp:uart -?- waiting reset
2023-12-09T09:55:44.010Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":-1} after 10000ms
2023-12-09T09:55:44.011Z zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 0
2023-12-09T09:55:44.012Z zigbee-herdsman:adapter:ezsp:driv Stop driver
2023-12-09T09:55:44.012Z zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp
2023-12-09T09:55:44.013Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-12-09T09:55:44.013Z zigbee-herdsman:adapter:ezsp:driv Close driver
Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:299:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

``` 

</p>
</details> 

<details><summary>Without the line:</summary>
<p>

```powershell
[10:53:55] INFO: Preparing to start...
[10:53:55] INFO: Socat not enabled
[10:53:55] INFO: Zigbee Herdsman debug logging enabled
[10:53:55] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-12-09 10:53:57: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-12-09 10:53:57: Logging to console and directory: '/config/zigbee2mqtt/log/2023-12-09.10-53-57' filename: log.txt
Zigbee2MQTT:debug 2023-12-09 10:53:57: Removing old log directory '/config/zigbee2mqtt/log/2023-12-06.12-30-08'
Zigbee2MQTT:info  2023-12-09 10:53:57: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
Zigbee2MQTT:info  2023-12-09 10:53:57: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:debug 2023-12-09 10:53:57: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[174,156,24,139,7,219,131,237],"networkKey":"HIDDEN","panID":24677},"serialPort":{"path":"/dev/ttyUSB1"}}'
2023-12-09T09:53:57.344Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-12-09T09:53:57.344Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[187,79,204,129,36,29,218,28,247,110,84,13,243,81,50,83],"panID":24677,"extendedPanID":[174,156,24,139,7,219,131,237],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB1"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-12-09T09:53:57.345Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/ttyUSB1","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-12-09T09:53:57.348Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2023-12-09T09:53:57.349Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
[... cut ...]
Zigbee2MQTT:error 2023-12-09 10:54:17: Exiting...
Zigbee2MQTT:error 2023-12-09 10:54:17: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
``` 

</p>
</details>

eronisko, 2023-12-09:
@maulwurf01 Thanks. I can see that the error is different with the ezsp adapter (but unfortunately, there doesn't seem to be any response?)

I see that you're using /dev/ttyUSB1, but [from your screenshot](https://private-user-images.githubusercontent.com/147982980/287787704-8c81604c-ea0c-4a77-b1bc-e41b3175a13e.png), that doesn't look like the right device. 

Can you show the debug logs again please while using the ezsp adapter and `/dev/serial/by-id/...` port like you did earlier?

maulwurf01, 2023-12-09:
Setup now looks like this:

```shell
port: /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0
adapter: ezsp

``` 

<details><summary>Logs</summary>
<p>

```powershell
[11:39:36] INFO: Preparing to start...
[11:39:37] INFO: Socat not enabled
[11:39:37] INFO: Zigbee Herdsman debug logging enabled
[11:39:37] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-12-09 11:39:39: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-12-09 11:39:39: Logging to console and directory: '/config/zigbee2mqtt/log/2023-12-09.11-39-39' filename: log.txt
Zigbee2MQTT:debug 2023-12-09 11:39:39: Removing old log directory '/config/zigbee2mqtt/log/2023-12-07.09-27-21'
Zigbee2MQTT:info  2023-12-09 11:39:39: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
Zigbee2MQTT:info  2023-12-09 11:39:39: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:debug 2023-12-09 11:39:39: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[174,156,24,139,7,219,131,237],"networkKey":"HIDDEN","panID":24677},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0"}}'
2023-12-09T10:39:39.210Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-12-09T10:39:39.211Z zigbee-herdsman:adapter:ezsp:debg Adapter concurrent: 8
2023-12-09T10:39:39.211Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[187,79,204,129,36,29,218,28,247,110,84,13,243,81,50,83],"panID":24677,"extendedPanID":[174,156,24,139,7,219,131,237],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-12-09T10:39:39.212Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-12-09T10:39:39.216Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2023-12-09T10:39:39.216Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2023-12-09T10:39:39.216Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2023-12-09T10:39:39.217Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2023-12-09T10:39:39.217Z zigbee-herdsman:adapter:ezsp:uart -?- waiting reset
2023-12-09T10:39:49.220Z zigbee-herdsman:adapter:ezsp:uart --> Error: Error: {"sequence":-1} after 10000ms
2023-12-09T10:39:49.222Z zigbee-herdsman:adapter:ezsp:driv Reset connection. Try 0
2023-12-09T10:39:49.222Z zigbee-herdsman:adapter:ezsp:driv Stop driver
2023-12-09T10:39:49.222Z zigbee-herdsman:adapter:ezsp:ezsp Stop ezsp
2023-12-09T10:39:49.223Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-12-09T10:39:49.223Z zigbee-herdsman:adapter:ezsp:driv Close driver
Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:299:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
``` 

</p>
</details> 

Note that there are two version of the Zigbee Stick:
[image]

The Dongle-E is the one with the EZSP protocol. I use the Dongle-P.

eronisko, 2023-12-09:
@maulwurf01 
> The Dongle-E is the one with the EZSP protocol. I use the Dongle-P.

eronisko, 2023-12-09:
@maulwurf01 

> The Dongle-E is the one with the EZSP protocol. I use the Dongle-P.

You're right, you were using the right adapter -- sorry.

I'm afraid I'm at a loss here. Since your device is working fine on another hardware, I'm thinking this could be something on hardware level or OS level... One last thing I could suggest is updating your OS to version 11.2, but I've had to revert from that update (my zigbee coordinator stopped working properly -- but that was a different device and a different error from yours).

I'm sorry I can't offer any better ideas right now.

maulwurf01, 2023-12-09:
Hi Ernest,
since I have the same hardware here twice (Dell OptiPlex 3040), I have installed another, fresh Home Assistant instance on the second OptiPlex. There, the stick gets instantly recognized and Zigbee2MQTT starts properly! Then, I have swapped the SSDs of the two OptiPlex PCs and the stick still gets recognized by the fresh install (this was to make sure there wasn't any BIOS setting messing with HAOS).

I have now decided to completely re-do my HA setup to make it work.

One guess what the issue could be is that I initially ran my HA as a virtual machine on my QNAP (OVA template) and then moved to the OptiPlex. It could be that the backup I had pulled (twice actually) somehow isn't compatible for a move from a OVA virtual machine to a generic-x86 device...but that's just a guess.

Thank you for your help and merry christmas!

closed 2023-12-09 by maulwurf01

maulwurf01, 2023-12-09:
Whilst setting up the new HA, the issue reappeared. But only after installing the Add-on "RaspberryMatic"! I got curious and researched these two add-ons in combination and managed to find the fix I've been looking for for 3 weeks...
https://github.com/jens-maus/RaspberryMatic/discussions/2226

The issue was that the CUxD part of RaspberryMatic somehow blocked the USB port and prevented from Zigbee2MQTT getting access to the port.
After putting in the TTYPARAM=NONE and using a backup to restore my old system, it works now everytime!

Thanks again for your help!