home-assistant/operating-system #2957: HA OS does not detect USB2Serial Connector from Prolific on Yellow

TimWeyand, 2023-12-03:
### Describe the issue you are experiencing

I downgraded from HA OS 11.1 to HA OS 10.5, to make sure it is not an issue with the Hardware.

I use an USB2Serial Connector from Prolific for controlling my Centralise House Vending (Zender Comfoair 350). The functionality is in place sind 8 Years and worked like a charm even as I changed from x86-64 to arm and from openhab to home assistant. With HA OS 11.x it stopped working.

That's the device information I see in 10.5
`ttyUSB0
/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0

Subsystem:
tty
Gerätepfad:
/dev/ttyUSB0
ID:
/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
Attribute:
DEVLINKS: >-
  /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0
  /dev/serial/by-path/platform-fe980000.usb-usb-0:1.2.3:1.0-port0
DEVNAME: /dev/ttyUSB0
DEVPATH: >-
  /devices/platform/soc/fe980000.usb/usb1/1-1/1-1.2/1-1.2.3/1-1.2.3:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS: usb
ID_MODEL: USB-Serial_Controller
ID_MODEL_ENC: USB-Serial\x20Controller
ID_MODEL_ID: '2303'
ID_PATH: platform-fe980000.usb-usb-0:1.2.3:1.0
ID_PATH_TAG: platform-fe980000_usb-usb-0_1_2_3_1_0
ID_REVISION: '0300'
ID_SERIAL: Prolific_Technology_Inc._USB-Serial_Controller
ID_TYPE: generic
ID_USB_DRIVER: pl2303
ID_USB_INTERFACES: ':ff0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: USB-Serial_Controller
ID_USB_MODEL_ENC: USB-Serial\x20Controller
ID_USB_MODEL_ID: '2303'
ID_USB_REVISION: '0300'
ID_USB_SERIAL: Prolific_Technology_Inc._USB-Serial_Controller
ID_USB_TYPE: generic
ID_USB_VENDOR: Prolific_Technology_Inc.
ID_USB_VENDOR_ENC: Prolific\x20Technology\x20Inc.
ID_USB_VENDOR_ID: 067b
ID_VENDOR: Prolific_Technology_Inc.
ID_VENDOR_ENC: Prolific\x20Technology\x20Inc.
ID_VENDOR_ID: 067b
MAJOR: '188'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '9290738'` 

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Use a Prolific USB2Serial Adapter
2. Make it work in 10.5
3. Upgrade to 11.x
[... 280 lines cut ...]
version_api | ok
installed_addons | Node-RED (16.0.2), Samba share (12.2.0), Studio Code Server (5.14.2), Zigbee2MQTT (1.34.0-1), Comfoair 2 MQTT (0.1.3), Terminal & SSH (9.8.1), Mosquitto broker (6.3.1), ESPHome (2023.11.6), owserver (1-wire) (0.3.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 9
views | 27
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 19. November 2023 um 12:11
-- | --
current_recorder_run | 3. Dezember 2023 um 14:13
estimated_db_size | 1620.88 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

sairon, 2023-12-05:
That's strange, I can see it on my Yellow on 11.1 too:
```
# dmesg -w
(...)
[  480.710067] usb 1-1.2: new full-speed USB device number 3 using dwc2
[  480.915554] usb 1-1.2: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 3.00
[  480.915591] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  480.915608] usb 1-1.2: Product: USB-Serial Controller
[  480.915622] usb 1-1.2: Manufacturer: Prolific Technology Inc.
[  480.963234] usbcore: registered new interface driver pl2303
[  480.963302] usbserial: USB Serial support registered for pl2303
[  480.963378] pl2303 1-1.2:1.0: pl2303 converter detected
[  480.965168] usb 1-1.2: pl2303 converter now attached to ttyUSB0

# ls -la /dev/serial/by-id/
total 0
drwxr-xr-x    2 root     root            60 Dec  5 15:34 .
drwxr-xr-x    4 root     root            80 Dec  5 15:34 ..
lrwxrwxrwx    1 root     root            13 Dec  5 15:34 usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0 -> ../../ttyUSB0

# ha os info
board: yellow
boot: A
data_disk: AJTD4R-0x54affa52
update_available: true
version: "11.1"
version_latest: "11.2"
```

Can you check the `dmesg` output right after you unplug and re-plug the adapter? Alternatively just check the host logs after doing so. The report with the default length of the host logs buffer unfortunately doesn't contain anything useful.

agners, 2023-12-05:
Maybe a problem with the USB port? Potential over-current event? There is a over-current protection, not sure what happens when it triggers. A complete power off and power back on might help in that case.

closed 2023-12-07 by TimWeyand