home-assistant/operating-system #2967: Missing WiFi support on HP T620

ericswpark, 2023-12-05:
### Describe the issue you are experiencing

I installed HAOS on a HP T620 and discovered that the onboard Broadcom Wi-Fi (BCM4352) is not getting detected. Can the necessary driver be added? Thanks.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Fresh install HAOS
2. See there is no Wi-Fi hardware support for BCM4352

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

ericswpark, 2023-12-05:
@sairon thanks! I don't really need BT, so this is perfect for me. If I grab the development builds, can I later switch to a stable channel once the commit lands in a stable release? Or do I have to reflash?

ericswpark, 2023-12-05:
Also noticed I got the model name wrong, changed title and description to reflect correct model name. Sorry!

ericswpark, 2023-12-05:
@sairon I've installed the latest development build, but `WIFI-HW` still shows as missing and attempting to scan for Wi-Fi devices says there are no Wi-Fi devices. Can this issue be re-opened?

sairon, 2023-12-11:
@ericswpark Could you please send the output of `dmesg`?

ericswpark, 2023-12-11:
@sairon sure: https://gist.github.com/ericswpark/565b3f3b59ae9db4b083b8721964a6e4

TLaren, 2024-03-02:
Over three months later, pretty sure there has been a new release:
  OS Version: Home Assistant: OS 12.0
  Home Assistant Core:           2024.2.5
I have a HP T620, AMD GX-415GA 1.6Ghz, 128GB SSD, 8Gb RAM, and a WiFi that is not seen.
I would like to use it as an AP for several WiFi Smart Devices.
The unit is and will be hardwired to my router.
I can give logs or test if needed, this is not a production unit.

arripeo, 2024-04-13:
I know this type of bug, i got in the past the same problem than solved by an new OS release, so when I've seen that in 12.1 release they changed the driver of Wifi I've deiced to NOT UPDATE waiting for a new suitable release. So if you want, please vote my new feature request that cloud be, on my opinion, a strong solution for all these type of problem that are very common

https://community.home-assistant.io/t/interface-by-ui-or-cli-for-installing-wifi-propitary-driver-or-linux-compatible/716527

closed 2024-07-21 by github-actions[bot]

ericswpark, 2024-07-22:
Please re-open this bug, I believe it's still not fixed yet

ericswpark, 2024-08-08:
I can confirm Wi-Fi is still missing on the T620 on Home Assistant OS 12.4. @sairon can this issue please be re-opened?

nze-123, 2024-08-30:
I confirm the issue persists ; HAOS 13.1 on HP T620  - still no WIFI support