home-assistant/operating-system #2968: Some dashboards no longer visible after HaOs update 11.1 -> 11.2

AllardKatan, 2023-12-05:
### Describe the issue you are experiencing

I just updated from OS 11.1 to 11.2, in a virtual machine install using Virtualbox under Linux Mint 21.2., with HA core 2023.11.3
After the update, I can no longer access the 'History', 'Energy' and 'Logbook' dashboards from the sidebar. I get a little red rectangle with 'error while loading page' 
I reverted back to 11.1, but the problem remains. Could it be something got corrupted during the OS upgrade? 

I have the feeling something went wrong during the upgrade process. There was also no automatic backup made. Isn't that supposed to happen unless you 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Nope, nothing that seems even vaguely related
```

### Anything in the Host logs that might be useful for us?

```txt
Dec 05 19:51:59 VeulenSmart systemd[1]: Reached target Multi-User System.
Dec 05 19:51:59 VeulenSmart systemd[1]: Startup finished in 1.989s (kernel) + 1min 30.558s (userspace) = 1min 32.548s.
Dec 05 19:52:01 VeulenSmart systemd[1]: var-lib-docker-overlay2-81e3ad46cf919c2834a3f8ab5f7964175381fb3177923c20d0fab230d5a9403b\x2dinit-merged.mount: Deactivated successfully.
Dec 05 19:52:01 VeulenSmart systemd[1]: mnt-data-docker-overlay2-81e3ad46cf919c2834a3f8ab5f7964175381fb3177923c20d0fab230d5a9403b\x2dinit-merged.mount: Deactivated successfully.
Dec 05 19:52:01 VeulenSmart kernel: hassio: port 8(veth7f329d8) entered blocking state
Dec 05 19:52:01 VeulenSmart kernel: hassio: port 8(veth7f329d8) entered disabled state
Dec 05 19:52:01 VeulenSmart kernel: device veth7f329d8 entered promiscuous mode
Dec 05 19:52:01 VeulenSmart kernel: kauditd_printk_skb: 1 callbacks suppressed
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1700 audit(1701805921.422:228): dev=veth7f329d8 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1300 audit(1701805921.422:228): arch=c000003e syscall=44 success=yes exit=40 a0=c a1=c001582c60 a2=28 a3=0 items=0 ppid=1 pid=380 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj=unconfined key=(null)
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1327 audit(1701805921.422:228): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D636F6E7461696E6572643D2F72756E2F636F6E7461696E6572642F636F6E7461696E6572642E736F636B
Dec 05 19:52:01 VeulenSmart kernel: hassio: port 8(veth7f329d8) entered blocking state
Dec 05 19:52:01 VeulenSmart kernel: hassio: port 8(veth7f329d8) entered forwarding state
Dec 05 19:52:01 VeulenSmart kernel: hassio: port 8(veth7f329d8) entered disabled state
Dec 05 19:52:01 VeulenSmart NetworkManager[293]: <info>  [1701805921.4357] manager: (veth4224e14): new Veth device (/org/freedesktop/NetworkManager/Devices/22)
Dec 05 19:52:01 VeulenSmart NetworkManager[293]: <info>  [1701805921.4463] manager: (veth7f329d8): new Veth device (/org/freedesktop/NetworkManager/Devices/23)
Dec 05 19:52:01 VeulenSmart systemd[1]: Started libcontainer container 8f26d60a226feb45f5d831105b879035057daebf148dd7fd7e267598c7375ee4.
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1334 audit(1701805921.653:229): prog-id=57 op=LOAD
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1334 audit(1701805921.653:230): prog-id=58 op=LOAD
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1300 audit(1701805921.653:230): arch=c000003e syscall=321 success=yes exit=15 a0=5 a1=c000131988 a2=78 a3=0 items=0 ppid=3071 pid=3082 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1327 audit(1701805921.653:230): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F38663236643630613232366665623435663564383331313035
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1334 audit(1701805921.653:231): prog-id=59 op=LOAD
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1300 audit(1701805921.653:231): arch=c000003e syscall=321 success=yes exit=17 a0=5 a1=c000131720 a2=78 a3=0 items=0 ppid=3071 pid=3082 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
Dec 05 19:52:01 VeulenSmart kernel: audit: type=1327 audit(1701805921.653:231): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F38663236643630613232366665623435663564383331313035
Dec 05 19:52:01 VeulenSmart kernel: eth0: renamed from veth4224e14
Dec 05 19:52:01 VeulenSmart NetworkManager[293]: <info>  [1701805921.9416] device (veth7f329d8): carrier: link connected
Dec 05 19:52:01 VeulenSmart kernel: IPv6: ADDRCONF(NETDEV_CHANGE): veth7f329d8: link becomes ready
Dec 05 19:52:01 VeulenSmart kernel: hassio: port 8(veth7f329d8) entered blocking state
Dec 05 19:52:01 VeulenSmart kernel: hassio: port 8(veth7f329d8) entered forwarding state
Dec 05 19:52:02 VeulenSmart systemd[1]: Started libcontainer container b073da3ff12a8ff7c01e45d533b358f0d7eda379e14d8af77fd8a8f64a45ee3b.
Dec 05 19:52:02 VeulenSmart systemd[1]: run-docker-runtime\x2drunc-moby-b073da3ff12a8ff7c01e45d533b358f0d7eda379e14d8af77fd8a8f64a45ee3b-runc.RzxYnu.mount: Deactivated successfully.
Dec 05 19:52:27 VeulenSmart systemd[1]: run-docker-runtime\x2drunc-moby-8ea7c9fd5de540d11f002aae56a8e7e2ff2687bfe3e2d5ee6806fb2e0bece5ef-runc.7KILep.mount: Deactivated successfully.
Dec 05 19:52:32 VeulenSmart systemd[1]: run-docker-runtime\x2drunc-moby-8f26d60a226feb45f5d831105b879035057daebf148dd7fd7e267598c7375ee4-runc.YsKVqn.mount: Deactivated successfully.
Dec 05 19:52:32 VeulenSmart kernel: kauditd_printk_skb: 41 callbacks suppressed
[... 135 lines cut ...]
dashboards | 2
-- | --
resources | 3
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | November 25, 2023 at 22:49
-- | --
current_recorder_run | December 5, 2023 at 20:51
estimated_db_size | 46.59 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

AllardKatan, 2023-12-05:
OK, nevermind. It was a browser issue, not HA. Clearing the cache solved it.

closed 2023-12-05 by AllardKatan

gfn256, 2023-12-05:
If you have a VirtualBox VM backup from before the HAOS update, try restoring that, and see if the problem is gone. Then try updating HAOS again.