home-assistant/operating-system #2969: USB Boot broken in 11.2

fencer, 2023-12-05:
### Describe the issue you are experiencing

Since upgrading to 11.2 USB boot on RPI Compute 4 no longer works. In the UBOOT screen I get an erorr 
```
... XHCI timeout on event 33... cannot recover. 
Bug at driver/usb/host/hci-ring.c:481/xhci_wiat_for_evet()!
Bug!
resetting ...
```
[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update to version 11.2
2. Reboot

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

RaspberryPI 4 Compute Module with USB Boot

TravisWilder, 2023-12-06:
USB SSD on RPi4 (not CM)  works for me

agners, 2023-12-06:
@fencer what type of USB disk/adapter are you using?

fencer, 2023-12-06:
It's a [Manhattan 130103](https://www.amazon.com/Manhattan-Hi-Speed-2-5-Inch-Enclosure-130103/dp/B002RMB5C6?th=1). This is the same enclosure I had been using on the RPI3 and then CM4 (after #1577)

agners, 2023-12-06:
It seems that the boot loader is not able to talk to the SSD, in such cases the A/B fallback doesn't even work :cry:  Since USB is notoriously unreliable at early boot, we recommend using a SD card as boot media for HAOS and move only the data to the USB attached SSD disk.

You might be able to get it to boot by changing things around: Use a different USB port, use a powered hub in-between, or plug it in later (start the Pi and plug in the USB disk a bit later). In case you can make it boot, you can downgrade using `ha os update --version 11.1`).

TravisWilder, 2023-12-06:
I can confirm that my SSD USB is booting only on the upper USB Port not on the lower :)

fencer, 2023-12-06:
> I can confirm that my SSD USB is booting only on the upper USB Port not on the lower :)

Keep in mind that the peripherals on the RPI4 are different from CM4.

TekFan, 2023-12-06:
> It seems that the boot loader is not able to talk to the SSD, in such cases the A/B fallback doesn't even work 😢 Since USB is notoriously unreliable at early boot, we recommend using a SD card as boot media for HAOS and move only the data to the USB attached SSD disk.
>
Reading this, I didn't dare upgrading to 11.2 yet...
So, what would be the best procedure to migrate from USB boot to SD boot ?

teppic01, 2023-12-07:
Hi,
sorry im confused. Is this a problem on raspberry pi 400 with ssd too?
i do not like to change back to sd, because of there life time limit...
T.

imerr, 2023-12-11:
Same issue here, using a different usb enclosure made it work again, working on a different system though. The problematic enclosure reports as:
>usb 4-1: new SuperSpeed USB device number 4 using xhci_hcd
usb 4-1: New USB device found, idVendor=174c, idProduct=5106, bcdDevice= 0.01
usb 4-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
usb 4-1: Product: AS2105
usb 4-1: Manufacturer: ASMedia
usb 4-1: SerialNumber: S0VZNYAC302876

odis68, 2023-12-14:
I have RaspberryPI 3B+ . Preliminary load with SD , The whole system is on SSD. After the update at 11.2, he stopped loading. Solution - I copied the preliminary download files from SSD to SD.
​

ohessel, 2023-12-18:
I have RaspberryPI 3B+
Took me some time, but keeping 2 USB RF sticks unplugged until the SSD started booting did the trick.

..USB seems to be little broken in this version - I experience this: https://github.com/home-assistant/operating-system/issues/2995

GHGiampy, 2023-12-19:
Also related #2977

criptych, 2023-12-30:
Just experienced this after upgrading on a CM4IO and I'm not even using any USB devices; it boots and runs entirely from an SD card.

PavelKuzub, 2023-12-30:
WORKAROUND TO RECOVER AND BOOT

I have got the same error having the same setup as OP - Rpi Compute Module 4 with HASS OS on USB bootable flash disk (Samsung).
None of the fiddling with config.txt and cmdline.txt helped to boot off USB.
I had a spare hardware and USB stick to play around with and I eventually recovered with this method:
- Downloaded previously stable 2023.11.1 HAOS image for my board [haos_rpi4-64-11.1.img.xz](https://github.com/home-assistant/operating-system/releases/download/11.1/haos_rpi4-64-11.1.img.xz) from [Releases](https://github.com/home-assistant/operating-system/releases) page.
- Flashed .img into a SPARE Test USB stick using "Raspberry Pi Imager" (it has erased the USB, so do NOT flash to your USB with data)
- Reconnected Test USB flash - a drive "hassos-boot" has been mounted. It is FAT filesystem, so easy to work with in Windows.
- Copied all the files from the partition with the label "hassos-boot" (u-boot.bin, config.txt, overlays, etc) to my PC.
- Tested that I could boot from this Test USB flash on Rpi CM4 board.
- Connected my Prod USB flash, took a backup of all files from its "hassos-boot" disk (that was running 2023.11.2)
- Replaced all files on "hassos-boot" disk with the ones from Test USB flash that was running 2023.11.1
- Connected Prod USB flash to Raspberry Pi CM4 and was able to boot from USB stick. No data was lost.
- [Optionally] Formally Downgrade using command: "ha os update --version 11.1"

Uploading a ZIP with 2023.11.1 boot files that recovered by setup, if you don't have a spare USB to flash to get the files:
[haos_rpi4-64-11.1_hassos-boot.zip](https://github.com/home-assistant/operating-system/files/13799344/haos_rpi4-64-11.1_hassos-boot.zip)

teppic01, 2023-12-31:
Thanks a lot for the work around! It maybe helps in other situations in future too!
I can report, that i have no problems with PI400 an 11.2

fencer, 2024-04-01:
did anyone try any of the newer versions on USB boot with CM4? I'm was scared to upgrade the os since then

closed 2024-07-09 by github-actions[bot]