home-assistant/operating-system #2975: HA OS Raspberry Pi 4 BlueTooth issue on version 11.2

nanard69, 2023-12-09:
### Describe the issue you are experiencing

I'have updated HA OS on my Raspberry Pi 4 from version 11.1 to 11.2 and i've lost bluetooth functionality. If i downgrade to 11.1, everthing works fine. I've discovered that happened because i have disactivated my wifi in the /mnt/boot/config.txt file. 
In version 11.2 of HA OS, if i want the bluetooth, i need to activate wifi in the /mnt/boot/config.txt file. Is it possible to give the choice of activating or not the wifi, because i don't need it ?

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. update HA OS to 11.2
2. desactivate wifi in the /mnt/boot/config.txt
3. reboot

the bluetooth will work no longer

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

vtb77, 2023-12-12:
Hello:

The same thing happens to me.  I would appreciate it if it will be solved because I do not use WiFi because I have the Raspberry connected by a network cable.

Thank you. 

Greetings.

virtualj, 2023-12-13:
Similar issue for me. I use HAOS on intel minipc. Waiting for the fix. In the mean time I'm not able to downgrade the OS. What is the procedure?

mikenhu, 2023-12-15:
Same issue for me. I'm on Home Assistant Yellow

BTCanada, 2023-12-20:
Same issue for me.  It looks like #2948 broke bluetooth for those of us with wifi disabled in config.txt.  The condition that patch uses to disable bluetooth is `ConditionPathExists=!/sys/bus/sdio/devices/mmc1:0001:1`
However, it looks like if you disable wifi, that device is never created, which causes that patch to remove the hci_uart module though `/usr/sbin/modprobe -r hci_uart`.
If you manually reenable hci_uart by using `/usr/sbin/modprobe hci_uart`, then bluetooth immediately starts working again.  @agners, #2948 breaks bluetooth on devices where wifi is disabled; can #2944 get a different fix?  Thanks!

virtualj, 2023-12-20:
I confirm: I have also wi-fi disabled in config

kimtoiva, 2023-12-23:
My bluetooth also stopped working after updating. I have wifi active.

markonsmith, 2023-12-29:
Bluetooth on Raspberry Pi 4 simply does not work regardless of the state of wifi. The problem has started after I upgraded OS from version 11.0 to 11.2 and Core from 2023.11.2 to 2023.12.3. 

Restoring full 11.2 core backup did not work as well as downgrading OS to 11.0. I am getting the "failed setup" error message for bluetooth device.

I can delete device, restart RP4 and configure bluetooth again to make it work for 2-3 days after which the same problem reocurrs.

maisen20, 2023-12-29:
I have the same issues and switched yesterday from openhab to HA. 

Unfortunately, it seems to pick out bluetooth somehow when I restart and it then suddenly recognizes a second bluetooth adapter, although there is only one.

I use Home Assistant on my raspberry pi 3b.

bcutter, 2023-12-30:
> Bluetooth on Raspberry Pi 4 simply does not work regardless of the state of wifi. The problem has started after I upgraded OS from version 11.0 to 11.2 and Core from 2023.11.2 to 2023.12.3. [...]

Oh dear, that's bad news. I was hoping to finally update from HA OS 10.5 to latest "stable" 11.2 on my Pi 4. WiFi is not disabled (so enabled) so I initially thought "well I don't care about #2975, let's finally update!". But that sounds like a real **update blocker** to me. I need bluetooth e. g. for the Switchbot integration.

agners, 2023-12-30:
The reported issue **here is that deactivating WiFi using config.txt also affects Bluetooth**. I am guessing this is related to the new kernel Bluetooth initialization which came in with the upgrade to the latest Raspberry Pi Linux kernel (#2904).

If you experience Bluetooth issues even with WiFi enabled (no config.txt change, e.g. as @markonsmith reports): Please **do not** hijack this thread, and open a new issue along with details of your system's configuration. Thanks!

@bcutter note that we have thousands of user using Raspberry Pi 4 with HAOS 11.2 and Bluetooth without problems. Don't get discouraged by one report :sweat_smile: You can downgrade the OS also relatively safely in case the update does not work out for you (using `ha os update --version 10.5`). Just in case, a full backup is always good advice.

bcutter, 2023-12-30:
> note that we have thousands of user using Raspberry Pi 4 with HAOS 11.2 and Bluetooth without problems

Thanks for pointing this out. Last general question (warning: off-topic) before this issue can re-focus on the core issue itself:

How do you know? Because of the analytics/statistics from https://analytics.home-assistant.io ? From that data I can
- neither map a specific board to a specific OS version
- nor see if those "are running without issues"

Therefore **really** interested in this. Would indeed encourage future update decisions, as my current (focusing on stability instead of cutting-edge features) update strategy is "wait at least 4 to 8 weeks, then check open GitHub issues with OS version and board type".

maisen20, 2024-01-01:
As far as I know, this problem occurs when using a Raspberry Pi 3b. When I switched from Openhab to Home Assistant, I first tested this on a 3B model and I always had the problem with Bluetooth. Now that I have switched to Raspberry Pi 4B, I no longer have any problems.

I set up both devices in exactly the same way, but when I switched to the 4B model, I imported everything from the backup of the 3B model.

So I no longer have this problem thanks to the change.

EugeneTM, 2024-01-12:
Got the same problem: if I disable WiFi, BT no longer works.

If I use generic Pi OS, BT still works after disabling WiFi.

rehakf, 2024-01-21:
I had to downgrade HAOS to ver. 11.0 with terminal (or SSH):
ha os update --version 11.0

With HAOS ver. 11.2 the integrated Bluetooth adapter in RPi4 was not working when WiFi disabled.
I didnt try HAOS ver. 11.1

Now, with HAOS ver. 11.0 the integrated Bluetooth adapter in RPi4 is working fine for me from 2023-12-17

Does anyone know if  this problem is solved in current release of HAOS?

Alxspb, 2024-02-06:
HassOS 11.1, 11.2, 11.3, 11.4, 11.5 still affected
Rpi3b+ bluetooth doesn't work on that versions

arripeo, 2024-04-13:
I know this type of bug, i got in the past the same problem than solved by an new OS release, so when I've seen that in 12.1 release they changed the driver of Wifi I've deiced to NOT UPDATE waiting for a new suitable release. So if you want, please vote my new feature request that cloud be, on my opinion, a strong solution for all these type of problem that are very common

https://community.home-assistant.io/t/interface-by-ui-or-cli-for-installing-wifi-propitary-driver-or-linux-compatible/716527

nanard69, 2024-04-14:
> I know this type of bug, i got in the past the same problem than solved by an new OS release, so when I've seen that in 12.1 release they changed the driver of Wifi I've deiced to NOT UPDATE waiting [...]

It's not a drivers issue but a misconfiguration of the OS. So your request won't help in my case.

TheHolyRoger, 2024-07-17:
Is there any news on this? 12.x still affected?

vtb77, 2024-07-18:
I have version 12.4 and it still does not work, waiting for it to be resolved in a future version.

olivermay, 2024-09-11:
I have the same issue, 13.1, and wifi disabled in config.txt

nanard69, 2024-12-11:
Same issue on 14.0

rehakf, 2024-12-18:
Can this be fixed? It is really annoying that disabling wifi on Raspberry
disables bluetooth too.

vtb77, 2025-03-19:
With the HassOS v15 update, it still doesn't work. Thanks.

PR #4132 (2025-07-01): Do not disable RPi Bluetooth if Wi-Fi is not present
This reverts commit eab18076ad7717ff4bce848b95fdb4e6c9f88f29.

This change was added in #2948 as a workaround for all-zero adapter appearing in the HA frontend (#2944). With changes implemented in [1], this is no longer needed, the only minor issue is that the ghost adapter still appears in hciconfig (and other utilities') output as reported in [2]. However, this should be less problematic than the Bluetooth being unavailable if WiFi is disabled through disable-wifi DT overlay, so let's start with removing the workaround.

Fixes #2975

[1] h [... cut]
commit 17ae2d47 (2025-08-04): Do not disable RPi Bluetooth if Wi-Fi is not present (#4132)

This reverts commit eab18076ad7717ff4bce848b95fdb4e6c9f88f29.

This change was added in #2948 as a workaround for all-zero adapter appearing
in the HA frontend (#2944). With changes implemented in [1], this is no longer
needed, the only minor issue is that the ghost adapter still appears in
hciconfig (and other utilities') output as reported in [2]. However, this
should be less problematic than the Bluetooth being unavailable if WiFi is
disabled through disable-wifi DT overlay, so let's start with removing the
workaround.

Fixes #29 [... cut]