home-assistant/operating-system #2976: Every boot has a 91 second "Wait until kernel time synchronized" timer

FusselTV, 2023-12-09:
### Describe the issue you are experiencing

I am unsure if it is correct to post this issue here. My setup is the haos_ova-10.2.vmdk on Unraid, and every time I reboot, I get this for 91 seconds. There is an easy fix ([link](https://github.com/systemd/systemd/issues/15031)). Would it be possible to apply this?
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.63-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

just reboot

### Anything in the Supervisor logs that might be useful for us?

```txt
nope
```

### Anything in the Host logs that might be useful for us?

```txt
nope
```

sairon, 2023-12-13:
Once the system boots, can you enter `login` to enter the OS console and check what `systemctl status systemd-timesyncd.service` reports? There is likely some issue with NTP sync in your environment, disabling the service mentioned in the ticket would not help, as it could lead to another problems in general.

FusselTV, 2023-12-13:
This is my output 192.168.178.20 is the ip of my homeassistant any idea how to fix the problem?
[image]

PieterHattinghZA, 2024-01-11:
I have the same issue

Xornop, 2024-02-24:
I have the same issue, however my log does say "timed out waiting for reply from ..... (time.cloudflare.com)"

fawzy46, 2024-03-03:
same issue here

bigruid, 2024-06-11:
Change the ntp server in this file
/etc/systemd/timesyncd.conf

closed 2024-12-18 by github-actions[bot]