home-assistant/operating-system #2977: RPi 3 USB ports stopped working after HAOS 11.2

JonatanHammarstedt, 2023-12-10:
### Describe the issue you are experiencing

I have RPi3 with HAOS and a UZB stick. I upgraded HAOS, HA and the Z-wave JS addon today. I updated HAOS from 11.1 to 11.2.

After the updates, I noticed my Z-Wave devices didn't work and the Integration couldn't start at all after the upgrades.

**From HA Core log:**
`Failed to set the Z-Wave JS add-on options: not a valid value for dictionary value @ data['options']. Got {'device': '/dev/serial/by-id/usb-0658_0200-if00', 's0_legacy_key':...`

Initially I thought it was a problem with the Z-wave JS addon so I tried to change a setting there but it couldn't find the device.
**Error message pop-up:**
`Device '/dev/serial/by-id/usb-0658_0200-if00' does not exist in Z-Wave JS (core_zwave_js)...`

I checked the /dev/ and I could no longer find the UZB device. In fact /dev/serial, /dev/ttyACM0 and /dev/ttyAMA0 was all gone. I also noticed that the LED on the UZB was off. When reinserting it, it goes on for a few sec before off again.

After reverting to HAOS 11.1, the /dev/ devices are all back and the UZB and Z-Wave integration works again.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Have RPi3 and use USB device / UZB dongle
2. Upgrade to 11.2
3. USB device no longer works

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System information

## System Information

version | core-2023.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 7 januari 2024 kl. 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-14.ui.nabu.casa
certificate_status | ready
instance_id | -
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.11.6
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 13.9 GB
disk_used | 5.4 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Z-Wave JS (0.4.2), Studio Code Server (5.14.2), File editor (5.7.0), eufy-security-ws (1.7.1), Terminal & SSH (9.8.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 29 november 2023 kl. 15:30
-- | --
current_recorder_run | 10 december 2023 kl. 00:15
estimated_db_size | 142.39 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

frankkobz, 2023-12-11:
Hello, it seems, that i've got the same problem. Here a screenshot.
[image]

redgtux, 2023-12-13:
Hi. Same issue with rpi3b and UZB z-wave device.
Device is not listed in /dev/serial/

GHGiampy, 2023-12-19:
Another one, rpi3b and UZB device...

[image]

Tried also to swap usb port

sairon, 2023-12-19:
See my comment in https://github.com/home-assistant/operating-system/issues/2995#issuecomment-1862701767. Although the newer RPi kernel is a few versions behind, any feedback after the update will be appreciated as well, maybe we could identify the cause more easily after that.

thedalrok, 2023-12-20:
+1 on this. I cannot boot the system at all since upgrading to 11.2. I have a RPi3B and booting the OS from attached usb ssd disc.

thedalrok, 2023-12-23:
Reverting to 10.5 resolved the issue.

sairon, 2024-01-02:
Since OS 11.3.rc1 is out for a while already, has anyone been able to check if the problem persists there?

davidnajar, 2024-01-02:
I've just tested and for me is not fixed.

thedalrok, 2024-01-03:
Tried rc2 today, still didn't get it to boot unfortunately.

agners, 2024-01-04:
@thedalrok so your case is probably better covered with https://github.com/home-assistant/operating-system/issues/2969. Can you check if 11.1 work for you? `ha os update --version 11.1`?

GHGiampy, 2024-01-04:
In my case the 11.1 was able to boot and start the UZB (usb) device, but has some stuttering problems with the usb device so i reverted to 11.0

thedalrok, 2024-01-05:
> @thedalrok so your case is probably better covered with #2969. Can you check if 11.1 work for you? `ha os update --version 11.1`?

Well I updated to 11.1 and after the first power cycle we t wouldn’t boot, same error as reported in the beginning of this thread. Tried another power cycle and it booted ok. Seems a bit glitchy or some sort of timing issue when powering up.

Taraman17, 2024-01-21:
Downgrading to 11.1 worked for me to regain ZWaveMe USB stick.

Taraman17, 2024-02-13:
Is there any change in this with HAOS 11.5 maybe?

feedspeed, 2024-02-27:
I'm still on 11.1 due to this issue. HAOS 12.0 is now released. Anybody already had time to test this version with the RPi 3 and its USB ports?

Taraman17, 2024-02-28:
I just checked - does not work :-(

sairon, 2024-03-05:
Since there is most information gathered in #2995 and the issue already contains mixed reports from RPi and other platforms, I'm closing this issue in favor of the other one. Please refer to that one for information about the current development.

closed 2024-03-05 by sairon