home-assistant/operating-system #2983: Update From 11.1 to 11.2 does not boot from correct slot

jscongdon, 2023-12-12:
### Describe the issue you are experiencing

I am running Home Assistant on a Dell Laptop. No issues; it runs well.
I attempted to upgrade from 11.1 to 11.2 5 times, and it never appeared to work... then I had a thought to switch the slot manually. On a fresh boot, the machine will boot to SlotA = 11.1, so I forced it to boot to SlotB, and after full boot I see 11.2 is installed on it. It works great, with no issues. Then we lost power and the battery died... and on next boot, it defaults to SlotA again and I'm back to 11.1
The mechanism is in place to change the Slot it boots from is not working correctly.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade the OS
2. Watch it not boot the alternate slot

### Anything in the Supervisor logs that might be useful for us?

```txt
Not that I know of.
```

### Anything in the Host logs that might be useful for us?

```txt
Not that I know of.
```

Yarr0tten, 2023-12-24:
Hi, i have exactly the same issue. My HA is also running on Dell laptop.

jscongdon, 2023-12-24:
I haven't put much effort in, but a good "band-aid" would be if I could manually edit the boot process that selects the slot to boot from.

FrankGiesecke, 2024-01-02:
I have the same issue running HAOS on a raspberry pi 4.

```shell
[core-ssh ~]$ ha os update --version "11.2"
Processing... Done.

Command completed successfully.
```

But after rebooting, the system ist still running version 11.1

```shell
[core-ssh ~]$ ha os info
board: rpi4-64
boot: A
data_disk: INTENSO-AA000000000000008831
update_available: true
version: "11.1"
version_latest: "11.2"
```

How can i force using slot 2?

FrankGiesecke, 2024-01-02:
Some additional information:

I'm running two instances of HA. Both on Raspberry Pi 4 (rpi4-64) with SONOFF Zigbee Dongle and an USB SSD Drive.

Instance KG:
- SONOFF Zigbee 3.0 USB Dongle Plus V2
  ```yaml
  DEVLINKS: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810183616-if00
    /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0
  DEVNAME: /dev/ttyACM0
  DEVPATH: >-
    /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/tty/ttyACM0
  ID_BUS: usb
  ID_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
  ID_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
  ID_MODEL_ID: 55d4
  ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0
  ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0-usb-0_1_3_1_0
  ID_REVISION: '0442'
  ID_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810183616
  ID_SERIAL_SHORT: '20220810183616'
  ID_TYPE: generic
  ID_USB_DRIVER: cdc_acm
  ID_USB_INTERFACES: ':020201:0a0000:'
  ID_USB_INTERFACE_NUM: '00'
  ID_USB_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
  ID_USB_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
  ID_USB_MODEL_ID: 55d4
  ID_USB_REVISION: '0442'
  ID_USB_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220810183616
  ID_USB_SERIAL_SHORT: '20220810183616'
  ID_USB_TYPE: generic
  ID_USB_VENDOR: ITEAD
  ID_USB_VENDOR_ENC: ITEAD
  ID_USB_VENDOR_ID: 1a86
  ID_VENDOR: ITEAD
  ID_VENDOR_ENC: ITEAD
  ID_VENDOR_ID: 1a86
  MAJOR: '166'
  MINOR: '0'
  SUBSYSTEM: tty
  TAGS: ':systemd:'
  USEC_INITIALIZED: '5309089'
  ```
- Intenso Portable SSD
  ```yaml
  DEVLINKS: >-
    /dev/disk/by-diskseq/28
    /dev/disk/by-id/usb-Intenso_Portable_SSD_20220812006D2-0:0
    /dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:2:1.0-scsi-0:0:0:0
  DEVNAME: /dev/sda
  DEVPATH: >-
    /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2- 2:1.0/host0/target0:0:0/0:0:0:0/block/sda
  DEVTYPE: disk
  DISKSEQ: '28'
  ID_BUS: usb
  ID_INSTANCE: '0:0'
  ID_MODEL: Portable_SSD
  ID_MODEL_ENC: Portable\x20SSD\x20\x20\x20\x20
  ID_MODEL_ID: '0579'
  ID_PART_TABLE_TYPE: gpt
  ID_PART_TABLE_UUID: 63cb89d4-5a0a-47bd-8e0f-c07cb289d1f1
  ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0-usb-0:2:1.0-scsi-0:0:0:0
  ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0-usb-0_2_1_0-scsi-0_0_0_0
  ID_REVISION: '0506'
  ID_SERIAL: Intenso_Portable_SSD_20220812006D2-0:0
  ID_SERIAL_SHORT: 20220812006D2
  ID_TYPE: disk
  ID_USB_DRIVER: usb-storage
[... 109 lines cut ...]
  ID_USB_INSTANCE: '0:0'
  ID_USB_INTERFACES: ':080650:080662:'
  ID_USB_INTERFACE_NUM: '00'
  ID_USB_MODEL: Portable_SSD
  ID_USB_MODEL_ENC: Portable\x20SSD\x20\x20\x20\x20
  ID_USB_MODEL_ID: '0578'
  ID_USB_REVISION: '0214'
  ID_USB_SERIAL: Intenso_Portable_SSD_DD564198851F0-0:0
  ID_USB_SERIAL_SHORT: DD564198851F0
  ID_USB_TYPE: disk
  ID_USB_VENDOR: Intenso
  ID_USB_VENDOR_ENC: Intenso\x20
  ID_USB_VENDOR_ID: 152d
  MAJOR: '8'
  MINOR: '0'
  SUBSYSTEM: block
  TAGS: ':systemd:'
  USEC_INITIALIZED: '4128996'
  ```

Instance KG can be updated without any issues and is now running version 11.2

Instance OG can be updated but is still running version 11.1

Please, can some one explain why?

jscongdon, 2024-01-02:
It is odd that you are not seeing the same issue on two environments with the same hardware.
I believe the PC-based installations issue deals with systems with a BIOS boot instead of UEFI. If you are able to connect a display and keyboard to the Pi that is not booting correctly, you can try to interrupt the boot process at the time it displays the "boot menu". You will be able to manually select the alternate slot. After booting that slot, check the version.

FrankGiesecke, 2024-01-02:
@jscongdon Unfortunately ist it not possible to attach a monitor to the Pi :(

Both Pi's are setup simultaneous. I have made no changes to the startup process.

FrankGiesecke, 2024-01-02:
After posting my device information I realized, that there is a difference between the two systems and their USB SSD recognation.

On instance KG, the USB SSD is listed as:
```yaml
DEVTYPE: disk
DISKSEQ: '28'
ID_BUS: usb
ID_INSTANCE: '0:0'
ID_MODEL: Portable_SSD
ID_MODEL_ENC: ...
ID_PART_TABLE_TYPE: gpt
``` 

On instance OG, the USB SSD is listed as:
```yaml
DEVTYPE: disk
DISKSEQ: '29'
ID_ATA: '1'
ID_ATA_* ...
ID_BUS: ata
ID_MODEL: INTENSO
ID_MODEL_ENC: ...
ID_PART_TABLE_TYPE: gpt
```

jscongdon, 2024-01-02:
@FrankGiesecke also, looking at `DEVNAME: /dev/sda` it appears one instance sees the external as a USB attached disk and the other as an internal

jscongdon, 2024-01-02:
I looked into my issue a bit more today.
In running on a non-EUFI bootable machine. In order for the machine to boot correctly I had to install GRUB and update its config to point to the EFI boot config.
`boot/grub/grub.cfg`
```
set root=(hd0,gpt1)
configfile (hd0,gpt1)/EFI/BOOT/grub.cfg
```
`EFI/BOOT/grub.cfg`
```
set default=99
set timeout=1

set ORDER="A B"
set A_OK=0
set B_OK=0
set A_TRY=0
set B_TRY=0
set MACHINE_ID=""
load_env

# select bootable slot
for SLOT in $ORDER; do
    if [ "$SLOT" == "A" ]; then
        INDEX=0
        OK=$A_OK
        TRY=$A_TRY
    fi
    if [ "$SLOT" == "B" ]; then
        INDEX=1
        OK=$B_OK
        TRY=$B_TRY
    fi
    if [ "$OK" -eq 1 -a "$TRY" -lt 3 ]; then
        default=$INDEX

        # increment attempts and save back to slot
        if [ "$TRY" -eq 1 ]; then
            TRY=2
        elif [ "$TRY" -eq 2 ]; then
            TRY=3
	else
            TRY=1
        fi

	if [ "$SLOT" == "A" ]; then
            A_TRY=$TRY
        fi
	if [ "$SLOT" == "B" ]; then
            B_TRY=$TRY
        fi

        break
    fi
done

# No bootable slot found, choose rescue...
if [ "$default" -eq 99 ]; then
    if [ "$A_OK" -eq 1 ]; then
        default=2
    fi
    if [ "$B_OK" -eq 1 ]; then
        default=3
    fi
fi

if [ "$MACHINE_ID" == "" ]; then
    boot_condition="systemd.condition-first-boot=true"
fi

save_env A_TRY A_OK B_TRY B_OK ORDER MACHINE_ID

default_cmdline="rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id=$MACHINE_ID fsck.repair=yes $boot_condition"
file_env -f ($root)/cmdline.txt cmdline

# root is a full HDD/partition definition in GRUB format like hd0,gpt1
# We extract the part before the comma to then append our own partition index
# at the end. This is hacky but the best way I found
regexp --set 1:boothd (.+),.+ ${root}

menuentry "Slot A (OK=$A_OK TRY=$A_TRY)" {
    linux (${boothd},gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd $default_cmdline $cmdline rauc.slot=A
}

menuentry "Slot B (OK=$B_OK TRY=$B_TRY)" {
    linux (${boothd},gpt4)/bzImage root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 $default_cmdline $cmdline rauc.slot=B
}

menuentry "Slot A (rescue shell)" {
    linux (${boothd},gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd $default_cmdline $cmdline rauc.slot=A systemd.unit=recovery.target
}

menuentry "Slot B (rescue shell)" {
    linux (${boothd},gpt4)/bzImage root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 $default_cmdline $cmdline rauc.slot=B systemd.unit=recovery.target
}
```
`EFI/BOOT/grubenv`
```
# GRUB Environment Block
# WARNING: Do not edit this file by tools other than grub-editenv!!!
MACHINE_ID=bfc125814f4a4c78920621c75f47cc15
A_OK=1
A_TRY=0
B_OK=1
B_TRY=0
ORDER=B A
###...
```

Currently, the installation for the latest OS version, 11.2, is in SLOT B. If I manually select that slot at boot, it boots correctly to that OS version and runs fine.
It appears that the EFI `grubenv` has been updated correctly so I am not sure why it isn't selecting the correct slot

One question I do have... GRUB has a grubenv as well
`boot/grub/grubenv`
```
# GRUB Environment Block
# WARNING: Do not edit this file by tools other than grub-editenv!!!
A_TRY=0
A_OK=0
B_TRY=0
B_OK=0
ORDER=A B
MACHINE_ID=
###...
```
but there is no `load_env` in the `boot/grub/grub.cfg` that is loading it
I'm not too familiar with GRUB... could this cfg be loaded via any other mechanism?

jscongdon, 2024-01-02:
I tested changing `boot/grub/grubenv`, updating ORDER to be `B A` >> did not work
I tested changing  `EFI/BOOT/grub.cfg`, updating `set ORDER` to be `B A` >> did not work

So, my assumption is that the EFI grubenv is being loaded but the ORDER is not "taking" ?

I have another remote HA instance running on a Pi with SD card... and it hasn't had issues through OS upgrades.

jscongdon, 2024-01-02:
Ok... I'm not sure why the manual updates did not work... but I found an answer via https://community.home-assistant.io/t/install-ha-on-old-laptop-without-uefi/407443/66

pointing the load and save env commands to the EFI/BOOT/grubenv did it.
Also... FWIW my paths are `/EFI/BOOT` all caps, not changed by me. I would check and use the exact path as it is on your system

I'm not sure if this fix would be put in place for backwards compatibility but is a fix none-the-less

closed 2024-01-02 by jscongdon

jscongdon, 2024-01-05:
With the release of OS 11.3, this fix has been verified.

FrankGiesecke, 2024-01-08:
Please, can some one explain how to upgrade from 11.1 to 11.3?
I still run into this problem, because my system is stuck on 11.1.

Installing 11.3 works fine, but the system still boot up from slot 1 where 11.1 is installed.

jscongdon, 2024-01-08:
@FrankGiesecke You will need to connect the storage medium for your HA hardware to another machine. If it were me, I would mount the first partition and look for `/EFI/BOOT/grub.cfg`.
Then apply the updates to the config as stated in https://community.home-assistant.io/t/install-ha-on-old-laptop-without-uefi/407443/72 where it mentions editing the `grub.cfg`.

jscongdon, 2024-01-10:
*Update*
When I applied this fix, the latest OS was in SlotB. After making the update to the grub.cfg, it started booting to Slot B as the default. I applied an OS update, and then it started booting to Slot A - as expected. I thought as was well. But when the next OS update was installed, it continued to boot to Slot A. I inspected the grub.cfg and the updates I had applied were gone. My "fix" was a false positive in thinking that the Slot boor order was being updated correctly. It appears the OS update re-writes the grub.cfg
I'm not sure how to fix this, at this point.

xazqe, 2024-12-06:
The Home Assistant OS still seems to overwrite the boot order every time it applies an OS update.

I boot from a USB SSD, but updating resets boot order to SD Card, and the Pi gets stuck and won't boot with fans at 100%.

My solution to this is to always have an SD Card with the Raspberry Pi official Boot Order Loader/Flasher installed on the SD Card, which I've preconfigured to change the boot order to USB. 

Powering on the Pi with that SD Card inserted turns the connected display/screen green as confirmation and the power button flashes green. After that I power the Pi down, remove the SD Card and reconnect the USB boot drive.. and it boots correctly again.

But this shouldn't be necessary, the boot order overwrite has to be removed from the OS updates!