home-assistant/operating-system #2988: DSN resolution for LAN is broken

nazar-pc, 2023-12-13:
### Describe the issue you are experiencing

I found in https://github.com/home-assistant/core/issues/103302 that multiple integrations are unable to connect to LAN devices via hostname, while IP address works.

This appears to be caused by DHCP server implementation (likely related to Docker) used in Home Assistant OS.

Specifically, this is what I see in `homeassistant` container:
```
homeassistant:/config# cat /etc/resolve.conf
search local.hass.io
nameserver 172.30.32.3
```

`search` there is clearly wrong because on my LAN DHCP server returns `localdomain` there. As the result `device.localdomain` hostname works, but `device` doesn't, because I believe it gets translated into `device.local.hass.io`, which, obviously, doesn't exist on my network.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Use DHCP for network configuration
2. Try to reach any LAN device by hostname
3. Observe DSN resolution failure due to incorrect `/etc/resolve.conf` contents

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

codyc1515, 2023-12-28:
Not sure why HA sets this. Perhaps it does not automatically get it from the network DHCP, so needs to specify it manually.

nazar-pc, 2024-01-05:
Every single other device on the network gets it over DHCP just fine and in HA it doesn't and even if I wanted to configure it manually, there is no option for it in the UI.

codyc1515, 2024-01-10:
I think it's an artifact of docker (via HAOS) but could be wrong. What happens if you try run it in a docker container directly?

nazar-pc, 2024-01-10:
It is definitely related to Docker, but since HAOS is an "appliance" kind of OS, it can customize `resolv.conf` of the container or do other necessary trickery to make it work out of the box. Being only able to use IP addresses is inconvenient.

proxict, 2024-04-02:
I've been banging my head against the wall over this and just came across this thread. Has anyone been able to come up with a workaround?
If I manually edit the `/etc/resolv.conf` to:
```
search localdomain
nameserver 192.168.1.1
```
... everything works as expected. But only temporarily, inside the ssh'ed docker container.

nazar-pc, 2024-04-02:
As mentioned originally: either use IP address of fully qualified domain name.

This is a bug and needs to be fixed in the way containers are set up, I don't think there could be a persistent workaround for this.

proxict, 2024-04-02:
I resorted to using FQDN for the time being, but was rather inconvenient as I had to reconfigure a few of my services because of `Host` header matching (I'm also utilizing subdomains in my LAN).

I was just curious if there was perhaps a way to somehow "hack" it in home-assistant. I'm new to HA, so I'm still exploring its capabilities and configuration options.

agners, 2024-04-03:
DNS resolution is largely taken care of by the DNS plug-in, which is essentially a container running CoreDNS: https://github.com/home-assistant/plugin-dns. The configuration is dynamically generated from this template https://github.com/home-assistant/plugin-dns/blob/master/rootfs/usr/share/tempio/corefile. The `resolve.conf` in containers points to that plug-in.

From what I understand, the configuration currently only handles FQDN properly. Or in other words: DHCP provided DNS suffixes are ignored :cry: .

nazar-pc, 2024-07-04:
I don't think this one was fixed yet

proxict, 2025-04-04:
This issue hasn't been addressed yet in this HA version:
  OS Version:               Home Assistant OS 15.1
  Home Assistant Core:      2025.4.0

nazar-pc, 2025-07-04:
I don't think this has been fixed yet

proxict, 2025-07-04:
I can confirm it hasn't yet been fixed in this HA version:
OS Version: Home Assistant OS 15.2
Home Assistant Core: 2025.7.0

nazar-pc, 2025-10-04:
Still an issue: Home Assistant OS 16.2, Home Assistant 2025.10.1

nazar-pc, 2026-08-16:
Still an issue: Home Assistant OS 18.2, Home Assistant 2026.8.2.

The host has correct `/etc/resolv.conf`:
```
# Generated by NetworkManager
search localdomain
nameserver 192.168.1.1
```

While `homeassistnat` container still has incorrect `local.hass.io` search option.

nazar-pc, 2026-11-15:
Nothing has changed