home-assistant/operating-system #2989: "MCE Hardware Error" appears on Intel NUC6i3SYH during OS startup and while its running

VanyaMuz, 2023-12-13:
### Describe the issue you are experiencing

During OS startup and while its running the MCE Hardware Errors appears every second. The OS is installed on Intel NUC6i3SYH with  i3-6100U 8GB, SSD 120GB. All the log file filled with these errors. Now the system work correctly, but it is unknown how it will behave in the future. Seems to be add noacpi and noapic boot options.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.63-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. 
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No errors
```

### Anything in the Host logs that might be useful for us?

```txt
Dec 13 13:46:04 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 143200200 MISC 3020004086 
Dec 13 13:46:04 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475164 SOCKET 0 APIC 2 microcode d6
Dec 13 13:46:04 homeassistant kernel: mce: [Hardware Error]: CPU 0: Machine Check: 0 Bank 6: cc50b54000041136
Dec 13 13:46:04 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 142227d40 MISC 13003004086 
Dec 13 13:46:04 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475164 SOCKET 0 APIC 0 microcode d6
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: CPU 3: Machine Check: 0 Bank 6: cc5bf38000041136
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 276d80140 MISC 17043004086 
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475165 SOCKET 0 APIC 3 microcode d6
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: CPU 0: Machine Check: 0 Bank 6: cc5bf38000041136
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 276d80140 MISC 17043004086 
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475165 SOCKET 0 APIC 0 microcode d6
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: CPU 2: Machine Check: 0 Bank 6: cc5bf38000041136
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 276d80140 MISC 17043004086 
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475165 SOCKET 0 APIC 1 microcode d6
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: CPU 1: Machine Check: 0 Bank 6: cc5bf48000041136
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 14432ae80 MISC b043004086 
Dec 13 13:46:05 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475165 SOCKET 0 APIC 2 microcode d6
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: CPU 2: Machine Check: 0 Bank 6: cc58f54000041136
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 10006c200 MISC 3042004086 
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475166 SOCKET 0 APIC 1 microcode d6
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: CPU 3: Machine Check: 0 Bank 6: cc58f54000041136
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 10006c200 MISC 3042004086 
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475166 SOCKET 0 APIC 3 microcode d6
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: CPU 1: Machine Check: 0 Bank 6: cc58f68000041136
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 14432ae80 MISC 3041004086 
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475166 SOCKET 0 APIC 2 microcode d6
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: CPU 0: Machine Check: 0 Bank 6: cc58f68000041136
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 14432ae80 MISC 3041004086 
Dec 13 13:46:06 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475166 SOCKET 0 APIC 0 microcode d6
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: CPU 3: Machine Check: 0 Bank 6: cc5d938000041136
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 276d1bc00 MISC 7041004086 
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475167 SOCKET 0 APIC 3 microcode d6
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: CPU 1: Machine Check: 0 Bank 6: cc5d938000041136
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 276d1bc00 MISC 7041004086 
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475167 SOCKET 0 APIC 2 microcode d6
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: CPU 2: Machine Check: 0 Bank 6: cc5d938000041136
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: TSC 0 ADDR 276d1bc00 MISC 7041004086 
Dec 13 13:46:07 homeassistant kernel: mce: [Hardware Error]: PROCESSOR 0:406e3 TIME 1702475167 SOCKET 0 APIC 1 microcode d6
[... 123 lines cut ...]
version_api | ok
installed_addons | Advanced SSH & Web Terminal (17.0.0), File editor (5.7.0), Studio Code Server (5.14.2), Mosquitto broker (6.4.0), MQTT Explorer (browser-1.0.1), Zigbee2MQTT (1.34.0-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 5
views | 5
mode | yaml

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 3, 2023 at 12:20 PM
-- | --
current_recorder_run | December 11, 2023 at 3:51 PM
estimated_db_size | 22.35 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

sairon, 2023-12-19:
You say it appears to be caused by disabling ACPI or APIC, what was the reason for that, and does it disappear without those options? In general MCE errors suggest some hardware problems, like failing RAM or CPU, or PSU issues, but it could be the case it happens only with ACPI/APIC disabled. Other than that, there's little to be done on the OS side. It would be interesting to check if you're able to reproduce it on another production OS.

VanyaMuz, 2023-12-21:
I tried installing Ubuntu on this machine and disabling ACPI and APIC, but unfortunately the errors persisted. I didn't notice any changes in the computer's operation. Windows was previously installed and there were no errors, or they were simply hidden). I have no more thoughts about this other than changing the hardware, but so far I see no reason for this.

sairon, 2023-12-21:
That really points to some problem with the hardware (or operating in conditions it doesn't like) rather than HAOS itself. You haven't said what is the reason for disabling ACPI/APIC but I have no other suggestion than not doing so if you want to avoid those error messages :) There's not much we can do about it in HAOS.

closed 2023-12-21 by sairon