home-assistant/operating-system #2992: rcu_preempt detected stalls on cpus/tasks

dankocrama, 2023-12-14:
### Describe the issue you are experiencing

Hello, 

We have home Assistant OS running in a VM on Truenass, and we are getting sometimes this error and this means that we have to reboot the VM every 1 or 2 days.

hope some one has a solution because its not really practical because all our home automations are not working 24h/7 

Thanks in advance 

[image]

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

6.1.63

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

We don't know where the problem is coming from we change the number of VM core, but no change.

### Anything in the Supervisor logs that might be useful for us?

```txt
No problems In log
```

### Anything in the Host logs that might be useful for us?

```txt
No problems In log
```

### System information

A Truenas with HA running in a VM

### Additional information

asd

baturinivan, 2024-01-22:
I have similar problem. Home Assistant OS is installed on Proxmox virtual environment on mini-pc.

siw1973, 2024-02-04:
I'm having similar issues on Home Assistant OS on VMWare Workstation Pro

dankocrama, 2024-02-20:
Any update on this ?

mwb9aa, 2024-02-21:
Maybe this link will [help](https://community.home-assistant.io/t/rcu-sched-self-detected-stall-on-cpu/317432/5)? I'm trying it now. Basically, in the virtual machine, I changed the chipset type to ICH9. On my old machine, I used the other chipset, but on my new machine, I may need the other chipset.

dankocrama, 2024-02-21:
I will have a check thanks for the feedback 🚀🚀

baturinivan, 2024-04-08:
I have fixed by updating proxmox VM to latest kernel version.

dankocrama, 2024-05-01:
No I still have the problem

siw1973, 2024-05-01:
Still got the issue on VMWare Workstation 17. 0.0

dankocrama, 2024-05-23:
Did anybody find a solution?

CristianGonzalezFernandez, 2024-07-14:
Same issue here :(

dankocrama, 2024-07-24:
Moved to Truenas scale, and already running for 30days without any bug

MomosX, 2024-07-25:
I was thinking of doing the upgrade to Scale also, but too scared that something will break and it will take me a long time to figure it out.

siw1973, 2024-10-24:
Staying alive.....

Still broken on VMWare Workstation and need to reboot everything every 4-5 days.

mischief, 2024-12-21:
seen after upgrade to HA OS 14.1. VM was unreachable over the network. previous version was OK. after a vm reset via virt-manager, it recovered. dumping what was on the console and my libvirt vm config here..

```
[277488.020311] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[277488.021572] rcu: 	0-...0: (0 ticks this GP) idle=955c/1/0x4000000000000000 softirq=1241785/1241785 fqs=1284672
[277488.023179] rcu: 	(detected by 3, t=6069646 jiffies, g=3192761, q=15615 ncpus=4)
[277551.026498] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[277551.027695] rcu: 	0-...0: (0 ticks this GP) idle=955c/1/0x4000000000000000 softirq=1241785/1241785 fqs=1297795
[277551.029152] rcu: 	(detected by 1, t=6132653 jiffies, g=3192761, q=15798 ncpus=4)
```

[ha-libvirt.xml](https://github.com/user-attachments/files/18216854/ha.txt)

johnwalk61, 2025-01-27:
I solved this issue on VMWare Workstation by turning off the automatic snapshots. I saw someone post that and didn't expect it to work. No CPU errors in 2 weeks.

siw1973, 2025-01-29:
> I solved this issue on VMWare Workstation by turning off the automatic snapshots. I saw someone post that and didn't expect it to work. No CPU errors in 2 weeks.

Thank you for this, however this is a good thing and a bad thing !

Good - VMWare Workstation HA instance over 24 hours no CPU errors 
Bad - My lazy backup strategy is now trashed and I actually have to put some effort into a proper backup plan....

johnwalk61, 2025-01-29:
Too funny. Hope it worked for you.

Mine has been stable for days now, not a single console error.

sairon, 2025-01-29:
Nice find, but I'd say it's really just a workaround. @johnwalk61 do you remember where you saw the post suggesting it?

johnwalk61, 2025-01-29:
Found it in my browser history...

https://community.home-assistant.io/t/solved-rcu-info-rcu-preempt-detected-stalls-on-cpus-tasks/795436

John