home-assistant/operating-system #2993: ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.21.3

Peeter887, 2023-12-15:
### Describe the issue you are experiencing

I installed the beta software on a RPI5.

When starting it up, everything seems fine - It gets a IP and loads the regular Home Assistant CLI page.

It's when i start to access: http://homeassistant.local:8123/ i get a Log file 

[logs.txt](https://github.com/home-assistant/operating-system/files/13682797/logs.txt)

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.3 dev

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install 11.3 dev
2.start up the RPI5
3. Access http://homeassistant.local:8123/
4. Log files received

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[07:58:54] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[07:58:54] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
23-12-15 07:58:56 INFO (MainThread) [__main__] Initializing Supervisor setup
23-12-15 07:58:57 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi5-64
23-12-15 07:58:57 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/aarch64-hassio-supervisor with version 2023.11.6
23-12-15 07:58:57 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
23-12-15 07:58:57 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-12-15 07:58:57 INFO (MainThread) [__main__] Setting up Supervisor
23-12-15 07:58:57 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
23-12-15 07:58:57 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
23-12-15 07:58:57 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
23-12-15 07:58:57 INFO (MainThread) [supervisor.host.services] Updating service information
23-12-15 07:58:57 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
23-12-15 07:58:57 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
23-12-15 07:58:57 INFO (MainThread) [supervisor.host.network] Updating local network information
23-12-15 07:58:57 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
23-12-15 08:58:57 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
23-12-15 08:58:57 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
[... 102 lines cut ...]
23-12-15 09:02:42 ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3: 500 Server Error for http+docker://localhost/v1.43/images/create?tag=2023.12.3&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi5-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
23-12-15 09:02:42 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
23-12-15 09:03:12 INFO (MainThread) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/raspberrypi5-64-homeassistant:landingpage to ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3
23-12-15 09:03:12 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/raspberrypi5-64-homeassistant with tag 2023.12.3.
23-12-15 09:03:12 ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3: 500 Server Error for http+docker://localhost/v1.43/images/create?tag=2023.12.3&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi5-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
23-12-15 09:03:12 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
23-12-15 09:03:42 INFO (MainThread) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/raspberrypi5-64-homeassistant:landingpage to ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3
23-12-15 09:03:42 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/raspberrypi5-64-homeassistant with tag 2023.12.3.
23-12-15 09:03:43 ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3: 500 Server Error for http+docker://localhost/v1.43/images/create?tag=2023.12.3&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi5-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
23-12-15 09:03:43 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
23-12-15 09:04:13 INFO (MainThread) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/raspberrypi5-64-homeassistant:landingpage to ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3
23-12-15 09:04:13 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/raspberrypi5-64-homeassistant with tag 2023.12.3.
23-12-15 09:04:14 ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3: 500 Server Error for http+docker://localhost/v1.43/images/create?tag=2023.12.3&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi5-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
23-12-15 09:04:14 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
23-12-15 09:04:44 INFO (MainThread) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/raspberrypi5-64-homeassistant:landingpage to ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3
23-12-15 09:04:44 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/raspberrypi5-64-homeassistant with tag 2023.12.3.
23-12-15 09:04:45 ERROR (MainThread) [supervisor.docker.interface] Can't install ghcr.io/home-assistant/raspberrypi5-64-homeassistant:2023.12.3: 500 Server Error for http+docker://localhost/v1.43/images/create?tag=2023.12.3&fromImage=ghcr.io%2Fhome-assistant%2Fraspberrypi5-64-homeassistant&platform=linux%2Farm64: Internal Server Error ("manifest unknown")
23-12-15 09:04:45 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
```

### Anything in the Host logs that might be useful for us?

```txt
Only getting 1 set of logs
```

aliocza, 2023-12-16:
The add-rpi5 branch was working, but since it was merged into dev, I have the same error.

markcra, 2023-12-17:
I had the same issue and followed the instructions in [this discussion](https://github.com/home-assistant/operating-system/discussions/2844?sort=old#discussioncomment-7844392), namely entering the following at command prompt with a keyboard and monitor attached to the RPi5;
```
ha supervisor options --channel=dev
ha supervisor reload
ha supervisor update`
```
When the update finished I reboot my device.

sairon, 2023-12-19:
Apart from some changes in version manifests that had to be done on our side, the above step is currently needed for RPi 5 dev build to work properly. The upcoming beta release will be more friendly in this regards and shouldn't raise errors without that (although it'd be still recommended to stay on dev to get bleeding edge changes for RPi 5).

closed 2023-12-19 by sairon

duongnm-dev, 2024-05-19:
> supervisor options --channel=dev

Thanks. I'm solved my issue