home-assistant/operating-system #2994: Sonoff Zigbee 3.0 Dongle Plus prevents Network Start During Bootup

GearDownGetDown, 2023-12-15:
### Describe the issue you are experiencing

When the Sonoff dongle is plugged into a USB port and the machine is rebooted, the network does not start. The HA CLI "network info" command shows the wired network is connected but no IP address, gateway of DNS a re configured. If dongle is removed and the machine is rebooted, HA OS starts up the network all OK. If the dongle is plugged into a USB port after the machine has been rebooted and  HA OS starts, then the network continues to operate fine as does the dongle. The Sonoff dongle appears to interfere with the initialization of the wired network connection during HA OS startup.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install fresh version of HA OS using balenaEtcher to unused hard drive
2. Install Sonoff dongle into USB port
3. Boot HA OS
4. Observe that network does not start and home assistant is not visible on the network
5. Use "host shutdown" to power down machine and remove dongle
6. Reboot machine into HA OS
7. Observe that home assistant is visible on the network
8. Install dongle
9. Observe that home assistant continues to run
...

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A since no network access when fault occurs.
```

### Anything in the Host logs that might be useful for us?

```txt
N/A since no network access when fault occurs.
```

### System information

N/A since no network access when fault occurs.

### Additional information

Machine is a Dell M6800 laptop with Windows 10 running on DRIVE-1 and home assistant operating system installed on DRIVE-0 in a dual boot configuration. Laptop is connected to router with Cat6a cable and the wireless interface is disabled. The Sonoff Zigbee 3.0 ZBDongle-E device is plugged into a USB 3.0 port. Windows 10 boots/runs without error when the dongle is installed but HA OS fails to start network when it is booted with dongle installed.

andy-81, 2023-12-18:
I have a similar issue, everything has been running fine prior to the update but now the wired Ethernet is refusing to use the DHCP address I have assigned on my router. 
I have noticed when I use the command line it shows the physical Ethernet connection has the right up address however is showing with a /32 cidr which is incorrect for my network, it should be the standard /24, not sure if that is what is causing the issue. In the web interface it does not show an ipaddress as being connected. 

I am running on an Intel NUC so with the same generic x86 64 build. It has only seemed to start occurring since the latest update to the OS. 

I have now enabled wireless which is working so not sure if this just impacts wired connections but it seems to on my setup. 

Happy to provide anything else which may help solve the issue.

closed 2024-04-14 by github-actions[bot]

GearDownGetDown, 2024-06-30:
Problem resolved. This Dell M6800 high performance laptop was abandoned because we couldn't make HAOS run on it. Finally, we came back to it because the hardware is so good. When we disabled the wireless adapter in the BIOS, HAOS started normally. We knew that HAOS required a hardwired network connection but did not realize that the wireless adapter needed to be disabled in the BIOS.