home-assistant/operating-system #2995: Z-Wave.me UZB dongle not fully recognized

jires, 2023-12-16:
### Describe the issue you are experiencing

After updating to 11.1, my usb dongle for Z-Wave became unresponsive and integration z-wave js ui hangs or won't start from time to time.
When I turn on the monitor connected to Intel NUC, I can se that dongle generates some errors when connected to usb port.

It's saying "device descriptor read/64,  error -32

 
It's like missing driver or something. Issue persist in 11.2 version.  Right now, I had to move my z-wave network to Aeotec Z-Stick 7.
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.63-haos - 	Home Assistant OS 11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. start the system up
2. Plug an USB dongle ZMEEUZB from Z-Wave.Me into usb port
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Not really
```

### Anything in the Host logs that might be useful for us?

```txt
Not Really
```

### System information

## System Information

version | core-2023.10.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.63-haos
arch | x86_64
timezone | Europe/Copenhagen
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1353
Downloaded Repositories | 22

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | September 7, 2024 at 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-11.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.2
-- | --
update_channel | stable
supervisor_version | supervisor-2023.11.6
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 916.2 GB
disk_used | 64.5 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (16.0.1), InfluxDB (4.8.0), Grafana (9.1.1), Node-RED (16.0.2), File editor (5.7.0), Studio Code Server (5.14.2), Mosquitto broker (6.4.0), Home Assistant Google Drive Backup (0.112.1), Network UPS Tools (0.12.2), ESPHome (2023.11.6), Samba share (12.2.0), Filebrowser (2.23.0_7), Portainer (2.19.3), Log Viewer (0.16.0), Glances (0.20.0), Frigate (Full Access) (0.12.1), Z-Wave JS UI (3.0.2), Zigbee2MQTT (1.34.0-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 5
views | 41
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 17, 2023 at 20:34
-- | --
current_recorder_run | December 7, 2023 at 19:12
estimated_db_size | 17118.81 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

mcolyer, 2023-12-17:
I'm having a similar issue, after upgrading to 11.2 my usb Zwave hub is no longer recognized.

sblogoshs, 2023-12-18:
In my case, it is the SkyConnect stick that stops working after a few hours and the problem can only be solved by unplugging it and plugging it in again. A downgrade to 11.1 solved the problem for me.

mcolyer, 2023-12-18:
Can confirm that downgrading to 11.1 and then unplugging and replugging in the dongle (Aeotec z-stick Gen 5) fixes the issue.

bnounours, 2023-12-18:
Same as all, I upgraded to OS 11.2 Zwave stop working as the stick was no more detected. I downgraded to OS 11.1, stick is back online.

GHGiampy, 2023-12-19:
Also related #2977

sairon, 2023-12-19:
We suspect it could be caused by some regression in the stable kernel, because at the same time similar issues started to appear on RPi and x86 platforms. There is bunch of changes in the USB subsystem in the newer stable releases which are currently in the development branch. If there's someone willing to switch to the `dev` channel and test it, it will help a lot (but make sure you have backups, since it switches all the components to `dev` and things can go south much more often).

```
ha supervisor options --channel=dev
ha supervisor reload
ha supervisor update
ha os update
```

Otherwise there should be another beta release soon, we will appreciate early feedback after the release too.

sairon, 2024-01-02:
Since OS 11.3.rc1 is out for a while already, has anyone been able to check if the problem persists there?

sblogoshs, 2024-01-02:
I also noticed that with OS 11.3.rc1 but for me it just took longer until the error reappeared, it just depends on how long there is no restart.

llroelj, 2024-01-02:
> We suspect it could be caused by some regression in the stable kernel, because at the same time similar issues started to appear on RPi and x86 platforms. There is bunch of changes in the USB [...]

I will try this tomorrow morning and will keep you posted if it works.

agners, 2024-01-03:
@mcolyer @sblogoshs @ohessel @bnounours @llroelj do you see the same device descriptor read errors as the original poster?

@bnounours which stick are you using?

llroelj, 2024-01-03:
> @mcolyer @sblogoshs @ohessel @bnounours @llroelj do you see the same device descriptor read errors as the original poster?
> 
> @bnounours which stick are you using?

After going through the above mentioned steps I am having the same error as mentioned above and the one I had before downgrading to 11.1 > usb 1-1.1.2: device descriptor read/64, error -32
 
Currently I am running the following versions:

Core: 2023.12.3
Supervisor: 2024.01.0.dev0201
Operating System: 11.4.dev20231226
Frontend: 20231208.2

agners, 2024-01-03:
@llroelj so you are saying you had the device descriptor read with 11.2 as well as the OS version from the dev channel?

What (exact) hardware are you running HAOS on?

llroelj, 2024-01-03:
> @llroelj so you are saying you had the device descriptor read with 11.2 as well as the OS version from the dev channel?
> 
> What (exact) hardware are you running HAOS on?

Indeed I had the same message in 11.2 and now the dev channel. With both the zwave stick cannot be found by the system itself (no serial folder). 
If needed I can share my dmesg log with all information. 

I am running HAOS on a RPi3 B+ 1GB memory installed on a SanDisk Ultra 32GB microSDHC card
The Zwave USB stick is a Z-Wave.me Stick (ZMEEUZB1)

mnorrsken, 2024-01-05:
My problems with USB is solved now after upgrading to HAOS 11.3. I had problems with ConbeeII zigbee dongle not being detected. This is possibly another issue but worth trying 11.3 as I see a couple of kernel USB fixes.

sblogoshs, 2024-01-06:
With the final version OS 11.3 the error seems to be fixed, but due to various updates I haven't had 24 hours without a restart yet, I hope that it will still work.

jires, 2024-01-07:
I can tell that upgrade to OS 11.3 DID NOT FIXED (? or did?) the issue with Z-Wave.me UZB. Descriptor error is still coming, however it's looks like that dongle is working as expected
Log output from host, after plugin dongle in the usb port, when system reboted with dongle plugged in, there is no errors in the log:

```
Jan 07 12:04:09 homeassistanttest kernel: usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
Jan 07 12:04:09 homeassistanttest kernel: usb 1-1.3: device descriptor read/64, error -32
Jan 07 12:04:09 homeassistanttest kernel: usb 1-1.3: device descriptor read/64, error -32
Jan 07 12:04:10 homeassistanttest kernel: usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
Jan 07 12:04:10 homeassistanttest kernel: usb 1-1.3: device descriptor read/64, error -32
Jan 07 12:04:10 homeassistanttest kernel: usb 1-1.3: device descriptor read/64, error -32
Jan 07 12:04:10 homeassistanttest kernel: usb 1-1-port3: attempt power cycle
Jan 07 12:04:11 homeassistanttest kernel: usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
Jan 07 12:04:11 homeassistanttest kernel: usb 1-1.3: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
Jan 07 12:04:11 homeassistanttest kernel: usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Jan 07 12:04:11 homeassistanttest kernel: cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
```

kvandt, 2024-01-08:
I have the same on my HA Yellow. When upgrading to 11.2 or 11.3 my UZB Zwave stick cannot be found anymore. No usb device available. When downgrading to 11.1, everything works again. I have tried the upgrade three times, but no luck. I hope this can be fixed in next release.

ohessel, 2024-01-08:
@agners I did not download the logs, and the journal of the error is rotated away already.
If it helps, I could update again and see if those exact logs appear.

afaucogney, 2024-01-11:
I'm in the same situation, with 2 different Zwave devices.
Now running 11.4OS

ohessel, 2024-01-11:
@afaucogney So you're saying 11.4 has the same issue?

afaucogney, 2024-01-12:
@ohessel Yes it seems the same issue.
I have reboot to be certain that everything is clear, but then I canot find my usb stick in the all_hardware list
And ZWaveJS is not able to talk to the controller. Should it work with that version ?

afaucogney, 2024-01-18:
Is there some activity on this topic, or do you need some information/log to help you ?

sairon, 2024-01-24:
I finally got my hands on the UZB stick to test myself. What should be noted is that the `device descriptor read` messages appear to be a red herring - I get them every time the device is hotplugged, and I get them on other kernels/distros too - I assume it's caused by a HW design fault that the device appears on the USB bus before it's properly initialized. The following lines (starting with `New USB device found ...`) indicate it's loaded properly in the end.

So far the device works for me on 11.2 through 11.4, so there must be something different on the problematic devices. For those who have the issues, can you share a bit more info?

1. What HW are you running HAOS on?
2. What version is the UZB firmware (shown on Z-Wave integration detail)?
3. Is the stick missing immediately when the host device is cold-booted (i.e. no USB hot-plugging)? If so, please post the `dmesg` output or host logs.
4. If the issue appears after some time of HA running, how long does it take, and does anything appear in the host logs/dmesg when it happens?

MUN0X, 2024-01-25:
I had this issue after upgrading to 11.2 so I stayed on 11.1 for now. Will try 11.4 and see what happens.  
Running Home assistant on a Yellow. UZB firmware 5.27.
If i remember correctly the device never showed up after cold booting

MUN0X, 2024-01-25:
Problem still persists in 11.4
Here is the output of dmesg | grep -i usb:
[    0.112239] usbcore: registered new interface driver usbfs
[    0.112302] usbcore: registered new interface driver hub
[    0.112374] usbcore: registered new device driver usb
[    0.112728] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    1.449922] usbcore: registered new interface driver uas
[    1.450005] usbcore: registered new interface driver usb-storage
[    1.450107] usbcore: registered new interface driver usbserial_generic
[    1.450150] usbserial: USB Serial support registered for generic
[    1.462440] usbcore: registered new interface driver usbhid
[    1.462450] usbhid: USB HID core driver
[    6.225448] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    6.298391] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
[    6.546981] dwc2 fe980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[    6.555240] dwc2 fe980000.usb: DWC OTG Controller
[    6.567977] dwc2 fe980000.usb: new USB bus registered, assigned bus number 1
[    6.575236] dwc2 fe980000.usb: irq 46, io mem 0xfe980000
[    6.580794] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    6.589347] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    6.596674] usb usb1: Product: DWC OTG Controller
[    6.601449] usb usb1: Manufacturer: Linux 6.1.63-haos-raspi dwc2_hsotg
[    6.616774] usb usb1: SerialNumber: fe980000.usb
[    6.636862] hub 1-0:1.0: USB hub found
[    7.034989] usb 1-1: new high-speed USB device number 2 using dwc2
[    7.247798] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[    7.261414] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    7.275315] usb 1-1: Product: USB 2.0 Hub [MTT]
[    7.304109] hub 1-1:1.0: USB hub found
[    7.775013] usb 1-1.1: new full-speed USB device number 3 using dwc2
[    7.974937] usb 1-1.1: device descriptor read/64, error -32
[    8.286902] usb 1-1.1: device descriptor read/64, error -32
[    8.602957] usb 1-1.1: new full-speed USB device number 4 using dwc2
[    8.802978] usb 1-1.1: device descriptor read/64, error -32
[    9.114943] usb 1-1.1: device descriptor read/64, error -32
[    9.227169] usb 1-1-port1: attempt power cycle
[    9.934939] usb 1-1.1: new full-speed USB device number 5 using dwc2
[    9.959294] usb 1-1.1: device descriptor read/8, error -32
[   10.095291] usb 1-1.1: device descriptor read/8, error -32
[   10.386948] usb 1-1.1: new full-speed USB device number 6 using dwc2
[   10.407313] usb 1-1.1: device descriptor read/8, error -32
[   10.539288] usb 1-1.1: device descriptor read/8, error -32
[   10.651088] usb 1-1-port1: unable to enumerate USB device
[   10.854957] usb 1-1.2: new full-speed USB device number 7 using dwc2
[   11.077618] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[   11.077645] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   11.077663] usb 1-1.2: Product: ConBee II
[   11.077678] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[   11.077692] usb 1-1.2: SerialNumber: DE2215099
[   11.232100] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[   11.232189] usbcore: registered new interface driver cdc_acm
[   11.232196] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   11.431673] usb 1-1.2: USB disconnect, device number 7
[   11.754923] usb 1-1.2: new full-speed USB device number 8 using dwc2
[   11.977619] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[   11.977648] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   11.977666] usb 1-1.2: Product: ConBee II
[   11.977681] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[   11.977695] usb 1-1.2: SerialNumber: DE2215099
[   11.979327] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device

afaucogney, 2024-02-02:
Hi,

  I use the HA yellow with firmware 5.5 for the Zwave stick + HA OS 11.4
  I have also tried with another purchased stick that I return back, because it changed nothing (no info about the firmware, but should be recent)
  I get some info in the Zwave integration, but the ZWaveJS failed to start. I do not see it in the "all-hardware" list. 
  I didn't tested recently with hot-plugging ? does it make sens to help you ?

Anthony

Tomfr06, 2024-02-14:
Just tried 11.5 jump from 11.1 and my Z-Wave.Me stick went missing... 
Reverted  to 11.1 and it's back

LeSpocky, 2024-02-14:
FWIW, after a full backup of my uSD card, I upgraded from 11.3 to 11.5 and everything still seems to work fine.  I'm running HaOS on RPi2 with this stick:

- ZMEEUZB1
- Barcode: 0019962006500
- Bus 001 Device 005: ID 0658:0200 Sigma Designs, Inc. Aeotec Z-Stick Gen5 (ZW090) - UZB
- Firmware: 5.5
- UZB von Z-Wave.Me 

Maybe it depends on the firmware version of the stick?

sairon, 2024-02-15:
@LeSpocky Have you had any problems before or were you just hesitating to upgrade? There is definitely some factor we are missing, given there are some - but not that many - reports of the issue, and no one from the team is able to reproduce it with their Z-Wave UZB or Aeotec Z-Stick :(

MUN0X, 2024-02-15:
@sairon, I have a spare uzb that I updated to the last fw but also doesn’t work. I would be happy to provide this one to someone of the team.

sairon, 2024-02-15:
@MUN0X Thanks, but before we go this route, could you (or anyone else with the issue) verify this happens reproducibly with the same stick even with a fresh HAOS install, or even better, on a different different machine at all? This can be any Raspberry Pi as well - since we track the same issue for RPi in #2977.

LeSpocky, 2024-02-15:
> @LeSpocky Have you had any problems before or were you just hesitating to upgrade? There is definitely some factor we are missing, given there are some - but not that many - reports of the issue, and no one from the team is able to reproduce it with their Z-Wave UZB or Aeotec Z-Stick :(

Actually I hesitated and waited with the upgrade until I had time for a full backup.  In general that v5 stick works okay, at least it is recognized by the OS reliably.  I have other problems with Z-Wave (devices like thermostats not responding or disappearing for a while then coming back), but I don't think they are related to this ticket.

MUN0X, 2024-02-15:
I believe i have a spare rpi3 laying around. Will test my spare stick on a fresh install this weekend.

FredrikFornstad, 2024-02-29:
Some more test results: I decided to flash a separate SD-card with plain Raspbian OS (nothing else) to see if the same problem exists upstream. In short: Yes, Raspian show the same behaviour as HAOS when booting or if the UZB is unplugged/re-inserted. I tried three different versions:

1. Latest Raspbian Legacy 64bit, Bullseye (released Dec 5, 2023)  - UZB works (show exactly the same behaviour as in HAOS 11.1)
2. Latest Raspbian 64bit, Bookworm (released Dec 11, 2023) - UZB does NOT work (exactly the same behaviour as in HAOS 11.2 - 12.0)
3. Latest Raspbian 32bit, Bookworm (released Dec 11, 2023) - UZB does NOT work (exactly the same behaviour as in HAOS 11.2 - 12.0)

The Legacy Raspian is running the same kernel as HAOS 11.1 and I guess also the same firmware, so I guess that is why it works. See first lines of dmesg here:

> fredrik@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2

The "normal" Raspbian has the following first lines in dmesg:
64bit:
> [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.0-rpi7-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2

32bit:
> [    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.0-rpi7-rpi-v7 (debian-kernel@lists.debian.org) (gcc-12 (Raspbian 12.2.0-14+rpi1) 12.2.0, GNU ld (GNU Binutils for Raspbian) 2.40) #1 SMP Raspbian 1:6.1.63-1+rpt1 (2023-11-24)
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2

The UZB "discovery" part of the logs looks as follows:
Raspbian 64bit Legacy (UZB works after the initial "hickup"):
 > [ 1125.402169] usb 1-1.4: USB disconnect, device number 10
[ 1134.073005] usb 1-1.4: new full-speed USB device number 12 using dwc_otg
[ 1134.153006] usb 1-1.4: device descriptor read/64, error -32
[ 1134.341003] usb 1-1.4: device descriptor read/64, error -32
[ 1134.529004] usb 1-1.4: new full-speed USB device number 13 using dwc_otg
[ 1134.609063] usb 1-1.4: device descriptor read/64, error -32
[ 1134.797005] usb 1-1.4: device descriptor read/64, error -32
[ 1134.905181] usb 1-1-port4: attempt power cycle
[ 1135.717049] usb 1-1.4: new full-speed USB device number 14 using dwc_otg
[ 1135.741234] usb 1-1.4: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[ 1135.741275] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1135.743959] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device

Raspbian 64bit "Bookworm" (UZB does not work):
 > [  261.469754] usb 1-1.4: new full-speed USB device number 12 using dwc_otg
[  261.549717] usb 1-1.4: device descriptor read/64, error -32
[  261.737706] usb 1-1.4: device descriptor read/64, error -32
[  261.925716] usb 1-1.4: new full-speed USB device number 13 using dwc_otg
[  262.005752] usb 1-1.4: device descriptor read/64, error -32
[  262.193729] usb 1-1.4: device descriptor read/64, error -32
[  262.301983] usb 1-1-port4: attempt power cycle
[  263.113723] usb 1-1.4: new full-speed USB device number 14 using dwc_otg
[  263.134652] usb 1-1.4: device descriptor read/8, error -32
[  263.262402] usb 1-1.4: device descriptor read/8, error -32
[  263.449734] usb 1-1.4: new full-speed USB device number 15 using dwc_otg
[  263.470533] usb 1-1.4: device descriptor read/8, error -32
[  263.598402] usb 1-1.4: device descriptor read/8, error -32
[  263.705865] usb 1-1-port4: unable to enumerate USB device

I also tried to attach the UZB to different USB-ports of the rpi3b and also tried via a powered USB-hub (connected to usb 1-1.3 in my logs in previous post). This made no difference. In HAOS 11.1 and Raspbian Legacy it works, in HAOS 11.1-12.0 and in "normal" Raspbian it does not.

I also specifically tried HAOS 11.3 as some users reported above that they had success with that release. However, in my case it did not make any difference.

@sairon and @agners,  I guess this should be reported upstream as I guess HAOS cannot do much about it. Any suggestions?

capjay1, 2024-03-01:
Same situation here. It was working well on OS 11.1, upgraded to OS 12.0, same issue. This is my details: 

Hardware: Raspberry Pi 3 (32-bit) 
Z-Wave dongle: Z-Wave.Me UZB 
Core: 2024.2.5 
Supervisor: 2024.02.1
Operating System: 12.0
Frontend: 20240207.1

Relevant dmesg lines: 

```
[    3.012565] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    3.252798] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
[    3.256536] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.585719] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    3.712762] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
[    3.716357] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.832546] usb 1-1.4: new full-speed USB device number 4 using dwc_otg
[    3.942556] usb 1-1.4: device descriptor read/64, error -32
[    4.162528] usb 1-1.4: device descriptor read/64, error -32
[    4.392558] usb 1-1.4: new full-speed USB device number 5 using dwc_otg
[    4.492666] usb 1-1.4: device descriptor read/64, error -32
[    4.712585] usb 1-1.4: device descriptor read/64, error -32
[    4.832682] usb 1-1-port4: attempt power cycle
[    5.492604] usb 1-1.4: new full-speed USB device number 6 using dwc_otg
[    5.523050] usb 1-1.4: device descriptor read/8, error -32
[    5.673199] usb 1-1.4: device descriptor read/8, error -32
[    5.892598] usb 1-1.4: new full-speed USB device number 7 using dwc_otg
[    5.938729] usb 1-1.4: device descriptor read/8, error -32
[    6.083099] usb 1-1.4: device descriptor read/8, error -32
[    6.212900] usb 1-1-port4: unable to enumerate USB device

```

Taraman17, 2024-03-01:
> Anyway, if there's someone who's got the stick that can be used to reproducibly trigger the issue (especially on x86), I'll be interested in getting it - that way I can try reproducing it in my environment and bisect the kernel to find the exact commit it causes i. I'll strongly prefer someone from the EU, to avoid high shipping costs and customs nuisances 😇 Of course I'll pay for the shipping and for a new locally-sourced device.
Hi, I would be willing to send you my device which has this problem on my RPI3 - I'm not sure if that guarantees the same behavior on X86. I would need to get a replacement before however and backup restore, since this is my production system and my home relies on it.

FredrikFornstad, 2024-03-01:
OK here goes. A lot of interesting info:
I managed to find an 10+ year old HP Pavilion laptop with an Intel i7-3610QM CPU. I downloaded a Debian "LiveCD" today (debian-live-12.5.0-amd64-standard.iso) and put the image on an USB and booted the live image.

Here you see the first part of dmesg so you see OS and Computer info:
> [    0.000000] microcode: microcode updated early to revision 0x21, date = 2019-02-13
[    0.000000] Linux version 6.1.0-18-amd64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01)
[    0.000000] Command line: BOOT_IMAGE=/live/vmlinuz-6.1.0-18-amd64 boot=live components quiet splash findiso=
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000087fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000088000-0x00000000000bffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000039dbefff] usable
[    0.000000] BIOS-e820: [mem 0x0000000039dbf000-0x000000003a1befff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003a1bf000-0x000000003aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003aebf000-0x000000003afbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003afbf000-0x000000003affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003afff000-0x000000003affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003b000000-0x000000003f9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb00000-0x00000000feb03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb80000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000004bf5fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.31 by INSYDE Corp.
[    0.000000] efi: ACPI=0x3affe000 ACPI 2.0=0x3affe014 SMBIOS=0x3aebdf18 MOKvar=0x3a264000 
[    0.000000] secureboot: Secure boot disabled
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Hewlett-Packard HP Pavilion dv6 Notebook PC/181B, BIOS F.22 11/02/2012
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 2294.742 MHz processor

This laptop has 4 USB ports. 3 marked "SS" which I guess is USB3 and 1 port which is USB2. I had the "live CD usb" in one of the "SS" ports. I then plugged the UZB into the USB2 port and got this:
 > [  545.624330] usb 1-1.2: new full-speed USB device number 5 using ehci-pci
[  545.704323] usb 1-1.2: device descriptor read/64, error -32
[  545.892340] usb 1-1.2: device descriptor read/64, error -32
[  546.080327] usb 1-1.2: new full-speed USB device number 6 using ehci-pci
[  546.160330] usb 1-1.2: device descriptor read/64, error -32
[  546.348330] usb 1-1.2: device descriptor read/64, error -32
[  546.456552] usb 1-1-port2: attempt power cycle
[  547.268330] usb 1-1.2: new full-speed USB device number 7 using ehci-pci
[  547.297571] usb 1-1.2: device descriptor read/8, error -32
[  547.425569] usb 1-1.2: device descriptor read/8, error -32
[  547.612330] usb 1-1.2: new full-speed USB device number 8 using ehci-pci
[  547.641567] usb 1-1.2: device descriptor read/8, error -32
[  547.769604] usb 1-1.2: device descriptor read/8, error -32
[  547.876542] usb 1-1-port2: unable to enumerate USB device

Exactly the same behaviour as trying the UZB in a rpi3b on HAOS 12.0 etc. BUT with one difference: Since this is another type of USB-controller it is using ehci-pci instead of dwc_otg which is used on rpi3b.

So now we know two things:
1. The changed behaviour was not introduced by Raspian, but in Debian
2. The changed behaviour is not limited to ARM. x86-64 is also affected.

I now rebooted and made another try, and connected the UZB (to the SAME laptop, running the same liveCD...) to one of the USB-ports market "SS" (and thereby to another internal USB-controller):
 > [   38.244292] usb 2-3: new full-speed USB device number 3 using xhci_hcd
[   38.372319] usb 2-3: device descriptor read/64, error -71
[   38.608317] usb 2-3: device descriptor read/64, error -71
[   38.844287] usb 2-3: new full-speed USB device number 4 using xhci_hcd
[   38.972317] usb 2-3: device descriptor read/64, error -71
[   39.208325] usb 2-3: device descriptor read/64, error -71
[   39.316405] usb usb2-port3: attempt power cycle
[   39.936295] usb 2-3: new full-speed USB device number 5 using xhci_hcd
[   39.957228] usb 2-3: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[   39.957241] usb 2-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   39.999591] cdc_acm 2-3:1.0: ttyACM0: USB ACM device
[... 24 lines cut ...]
[    3.358020] usb usb2: SerialNumber: 0000:00:14.0
[    3.358149] hub 2-0:1.0: USB hub found
[    3.358165] hub 2-0:1.0: 4 ports detected
[    3.358592] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[... cut ...]
So to my surprise it actually works (after the usual hickup in the startup, however now with another error) using the "SS" USB ports with the xhci-hcd driver.
So now we know a bit more:
3. Some types of USB-controllers/drivers (now this was the xhci_hcd driver) may work with UZB and the newer Debian releases.
4. Everything points to that (some batches? of) the UZB is not fully USB compliant in the startup-process, but depending on how the USB driver handles it can work (and has done so until recently)

FredrikFornstad, 2024-03-01:
Forgot to say: I also booted the same laptop as in previous post with Windows 10 Home  (22H2). Then the laptop could successfully access the UZB in all the USB slots without any problem and exposing it as a COM port using the build-in Microsoft "usbser" driver.

FredrikFornstad, 2024-03-02:
@sairon : I did some more testing with plain Debian Live CDs on my HP Pavillion Laptop (Intel i7-3610QM CPU): The USB3 port always work, and the USB2 port as follows:

Debian 12.0.0 Works  (Kernel 6.1.0-9-amd64), released June 10, 2023
Debian 12.1.0 Works  (Kernel 6.1.0-10-amd64), released July 22, 2023
Debian 12.2.0 Does NOT work  (Kernel 6.1.0-13-amd64), released Oct 7, 2023
Debian 12.4.0 Does NOT work  (Kernel 6.1.0-15-amd64), released Dec 10, 2023  (Debian 12.3.0 was never released)
Debian 12.5.0 Does NOT work

FredrikFornstad, 2024-03-03:
I have been looking at the changelog for Debian 12.2 for "suspects". Found these commits in the Linux kernel that was introduced in Debian 12.2:
    - USB: core: Unite old scheme and new scheme descriptor reads
    - USB: core: Change usb_get_device_descriptor() API
    - USB: core: Fix race by not overwriting udev->descriptor in hub_port_init()
    - USB: core: Fix oversight in SuperSpeed initialization

My guess, but I might be wrong, is that the "Unite old scheme and new scheme descriptor reads is the cause why the UZB does no longer work. There is a fix for USB3 (SuperSpeed) that was found not to work after the schemes update.

Here you can read a bit about these patches and how they were introduced: [https://lore.kernel.org/linux-usb](https://lore.kernel.org/linux-usb/6eadec91-990a-4fbd-8883-8366c4a4d8e4@rowland.harvard.edu/)

sairon, 2024-03-04:
@FredrikFornstad Very interesting findings again, thanks! I've created a build with those four patches reverted for Raspberry Pi 3 (32-bit and 64-bit builds), could you test it? You can find the build here: https://github.com/home-assistant/operating-system/actions/runs/8141367477

Just ignore the failure, it's because the image of the target for testing wasn't built. Also note you must be logged in to download and the files are ZIP compressed when downloaded (it's a limitation of GH artifacts).

sairon, 2024-03-04:
FWIW, these changes were introduced to Linux stable in 6.1.53 - which means that last version working should be HAOS 10.5 for x86 (and other non-RPi boards, with 11.0.rc1 being the breaking version, bumping 6.1.52 -> 6.1.53), and HAOS 11.1 for Raspberry Pi (breaking version 11.2.rc1, bumping 6.1.21 -> 6.1.58). This issue is a bit of a mixed bag of reports from both platforms without any further details, and while there's clear consensus about reverting to 11.1 fixing the issue, it's not clear if it's related to x86, or those all are about RPi (which should have been reported in #2977).

FredrikFornstad, 2024-03-04:
Tested your special bild (64bit). Result: It works with UZB on rpi3b.

Dmesg reports Linux version 6.1.73-haos-raspi with build date today afternoon.

It is exactly as in haos 11.1:  2x2 error messages, and then directly after the "attempt powercycle" it find the UZB and do the handshake and then cdc_acm assigns it to ttyACM0

Next step?

sairon, 2024-03-05:
@FredrikFornstad Perfect, I have started a [discussion](https://lore.kernel.org/linux-usb/1e954652-dfb3-4248-beea-b8a449128ff0@sairon.cz/) in the linux-usb mailing list and reported the issue to the regressions list as well. Also, putting all the pieces together, I realized the "USB 2" port on my mini PC I used for the testing before, is in fact only black, but it's a SuperSpeed port anyway :facepalm:  That's why I wasn't able to reproduce it before (just like you, I got -71 errors instead and the driver recovers). And using RPi's USB 2 ports, I am reproducibly getting the USB enumeration error too :tada:

kvandt, 2024-03-05:
The solution to put the UZB stick in a USB3 port is not a solution for the HA Yellow. This device only has USB2 ports (in USB-A format). I just tried again the update from 11.1 to 11.5 and 12.0, but the UZB stick is not found. I switched USB port (but are both USB2) and no effect as expected. I hope this can be solved upstreams and if not in a special build for HA Yellow devices.
@sairon can you also add the HA Yellow label to this topic?

kvandt, 2024-03-06:
Thnx @sairon !

FredrikFornstad, 2024-03-11:
> @FredrikFornstad Perfect, I have started a [discussion](https://lore.kernel.org/linux-usb/1e954652-dfb3-4248-beea-b8a449128ff0@sairon.cz/) in the linux-usb mailing list and reported the issue to the regressions list as well. Also, putting all the pieces together, I realized the "USB 2" port on my mini PC I used for the testing before, is in fact only black, but it's a SuperSpeed port anyway 🤦 That's why I wasn't able to reproduce it before (just like you, I got -71 errors instead and the driver recovers). And using RPi's USB 2 ports, I am reproducibly getting the USB enumeration error too 🎉

@sairon, I saw Alan Stern asked for Windows Wireshark logs from UZB attachment. This is from my HP Pavilion laptop, running Windows 10 Home 22H2. The log starts exactly when I plug in the UZB, before I plugged it in it was 30 seconds of complete silence on the usb-bus. The UZB is plugged into "Port 2" which after windows has managed to start talking to it becomes destination 1.3.0. One notable difference (I think) is that Windows do 5 port resets, before it manage to start communication with the UZB. I guess the UZB "problem" can be timing related, and that Linux gives up too soon. Also, I tend to remember to have read somewhere that Windows uses the "new" scheme, just as Linux do since the "troublesome" patches that caused the problems with UZB, so maybe if Linux did a few more tries before giving up, it maybe works also with the new scheme. Anyway, maybe the attached log-file is of some help. Windows assign the UZB to COM3 in this particular case.

[usbPcap1_USB2-port.txt](https://github.com/home-assistant/operating-system/files/14563813/usbPcap1_USB2-port.txt)

FredrikFornstad, 2024-03-13:
@sairon, I looked in detail at your usbmon traces and compared them side by side. I understand Alan already have looked at those in detail, and as an amateur I can probably not provide much more useful input/ideas. But I stick out my neck a little bit and offer my observations for what it may be worth:

1. With the "troublesome" patches reverted (the working example), the sequence executes faster. Assuming the timestamps (second column) is in usec, then the working example is 1871 usec faster up to the powercycle instruction than the non-working example after the start of the logs. Most of these come early in the logs at two places 800 usec + 1000 usec. I guess this could just be a coincidence caused by some CPU interrupt calls that has nothing to do with the usb bus. Repetive runs of usbmon would tell if this was just a coincidence or if the troublesome patches actually made the whole init procedure slower than before.
2. After approx 1.28 seconds(I guess) after the start of the logs comes an interupt input (I guess this is the power-cycle). In the working example this took approx 1750 usec longer than in the non-working example. Why does the "interupt input" come later (take longer time?) in the working example? This I guess could be significant for the troubleshooting.
Non-working example:
`ffff9f9a29b3f300 298538893 C Co:1:002:0 0 0
ffff9f9a012cae40 298581342 C Ii:1:002:1 0:2048 1 = 04`  (timestamp diff = 42499)
Working example:
`ffff8fc4ee367240 368298823 C Co:1:002:0 0 0
ffff8fc4c0c5ac00 368343025 C Ii:1:002:1 0:2048 1 = 04`  (timestamp diff = 44202)
3. After the interupt input then follows a control input submission and directly after that a control input callback where for the first time there is a difference in bit sequence. If this is where the init has been successful then the cause of the difference is before this point which indicates a potential timing issue, unless there are things going on that is not captured by usbmon. Below is the first difference I see in the sequence apart from timing:
Non-working example:  `ffff9f9a29b3f300 298742459 C Ci:1:002:0 0 4 = 03011000`
Working example:  `ffff8fc4ee367240 368502372 C Ci:1:002:0 0 4 = 01011100`

diablodale, 2024-03-14:
As a datapoint... my Z-Wave.Me UZB has none of these failures that I've seen. Looking now I also see nothing in various logs.
It has firmware 5.26. This is the newest firmware possible on my specific UZB stick.
There are many sub-types of UZB sticks and each has distinct firmware upgrade paths that DO NOT all reach the same numbers.

MUN0X, 2024-03-21:
I updated my HA yellow from 11.1 to 12.1. 
My uzb stick works unlike at 11.2-11.4.

Looking at dmesg output, I still get some errors while it attempts to read the descriptor, after a single power cycle it recognizes the stick.

FredrikFornstad, 2024-03-28:
@sairon , I saw that you are thinking about contacting Z-Wave.Me team on possible correction of the UZB firmware. For your information, in case you are not aware, relevant information for them might be the combination of bootloader version and firmware version. The bootloader version is easily obtained using their Z-Way software using the Expert UI (no license needed for viewing the info or to upgrade bootloader and/or firmware).

I do not recall exactly what versions my UZB sticks had when I bought them many years ago. But at least one of them was on 5.04 or 5.05. The other one must have been 5.06 or earlier.

Today, both my UZB sticks are running bootloader 40196  and firmware 5.39 (after several steps of upgrading both bootloader and firmware). As @diablodale says, the bootloader/firmware upgrade paths are not straight forward. Actually, I think it is possible  to upgrade his UZB to 5.39 also, but requires to filter on "all" upgrade paths and not only the "active" ones. However, I do not know why some paths (the non-active) are normally hidden, so maybe this is not recommended.

See here for the very complicated upgrade path map: [https://[service.z-wave.me/expertui/uzb-stats/versions-graph.html?hw=277&with_hidden](https://service.z-wave.me/expertui/uzb-stats/versions-graph.html?hw=277&with_hidden)]

Given the amount of time that has pasted since 5.39 firmware was released, I somewhat doubt if the Z-Wave.Me team still have a possibility to make any adjustments (if it is technically possible).

kvandt, 2024-03-31:
For the record: 12.1 fixed the issue for me.

Taraman17, 2024-04-01:
For me as well, the ZWAVE stick was recognized with 12.1

PoltoS, 2024-04-16:
Thank you @sairon for calling for me. I'm from Z-Wave.Me, the manufacturer of the UZB1.

Before we dive in deeper, just to test it, does an intermediate old-school USB hub helps? This might be an easy solution.

Based on the details analysis here and in the Linux Kernel thread I can deduce that UZB1 is setting up pull-ups too early. Even taking into account that this device is EOL and is not produced for more than 4 years, we still have the ability to make a new firmware. In theory ;) But I'll have to check with my colleagues if we have access to the USB part of the code as in old times Sigma Designs was not sharing sources of internal parts of the SDK, especially in the CDC-ACM driver implementation (in contrast, now we control everything except for the very low-level RF driver called RAIL).

I'll come back soon with more info from our R&D.

Please note that we have released new cool hardware to substitute UZB1. Modern and more professional.
[Z-Station](https://smartsd.ch/z-station-usb-adaptor-for-z-wave-zigbee-thread-ble) dual-protocol Z-Wave & Zigbee/Thread/BLE USB dongle
[mPCI-E](https://smartsd.ch/mpcie-card-with-z-wave-zigbee-support) dual-protocol Z-Wave & Zigbee/Thread/BLE mini PCI-Express extension card
[RaZberry 7 Pro](https://smartsd.ch/razberry-7-pro-z-wave-plug-on-module-for-raspberry-pi) Z-Wave extension card for Raspberry Pi and similar

All are using high-performant external antennas providing 3-4 times better range compared to UZB1. And with Long Range support (if you are in the US; soon in EU too)! Recently we published our test results and it went up to 1.9 km light-of-sight! And of course a simple backup & restore can help to migrate without any re-inclusion. So, if you thought about replacing the old 5th gen UZB1 with new hardware, it is the right time — to compensate the issue above we propose an offer of 20 EUR discount on any of the three hardware listed above with the coupon 2CPLD4DT.

closed 2024-07-24 by github-actions[bot]

MehdiTouns31, 2024-11-25:
Hello guys,

I have had the same problem as you for a few months on my Home Assistant installation on RPI 4B: 

my ZMEEUZB Z wave USB drive (firmware version 5.37) is not read correctly by the Z wave JS add-on. Here are some logs from the add-on:
[image]
Here is the output of the dmesg command:
[image]

I'm embarrassed to no longer have any information coming back from my Zwave network... Any idea of ​​the problem I'm encountering?

i tested to downgrade to HAOS 10.5, but still the same issue... I don't understand... 
[image]

MUN0X, 2025-03-22:
Hi,

I just updated to HA OS 15.0 and the problem appears to be back.

` OS Version:               Home Assistant OS 15.0
  Home Assistant Core:      2025.2.5

  Home Assistant URL:       http://homeassistant.local:8123
  Observer URL:             http://homeassistant.local:4357
➜  ~ dmesg | grep -i usb
[    0.036600] usbcore: registered new interface driver usbfs
[    0.036610] usbcore: registered new interface driver hub
[    0.036621] usbcore: registered new device driver usb
[    0.362051] usbcore: registered new interface driver uas
[    0.362061] usbcore: registered new interface driver usb-storage
[    0.362092] usbcore: registered new interface driver usbserial_generic
[    0.362099] usbserial: USB Serial support registered for generic
[    0.365430] usbcore: registered new interface driver usbhid
[    0.365434] usbhid: USB HID core driver
[    5.608027] dwc2 1000480000.usb: supply vusb_d not found, using dummy regulator
[    5.631610] dwc2 1000480000.usb: supply vusb_a not found, using dummy regulator
[    5.721152] dwc2 1000480000.usb: DWC OTG Controller
[    5.728280] dwc2 1000480000.usb: new USB bus registered, assigned bus number 1
[    5.747591] dwc2 1000480000.usb: irq 173, io mem 0x1000480000
[    5.753439] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.06
[    5.761709] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    5.769034] usb usb1: Product: DWC OTG Controller
[    5.773948] usb usb1: Manufacturer: Linux 6.6.74-haos-raspi dwc2_hsotg
[    5.780544] usb usb1: SerialNumber: 1000480000.usb
[    5.787608] hub 1-0:1.0: USB hub found
[    6.109419] usb 1-1: new high-speed USB device number 2 using dwc2
[    6.313554] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[    6.321746] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    6.328879] usb 1-1: Product: USB 2.0 Hub [MTT]
[    6.329035] hub 1-1:1.0: USB hub found
[    6.733429] usb 1-1.1: new full-speed USB device number 3 using dwc2
[    6.925410] usb 1-1.1: device descriptor read/64, error -32
[    7.217416] usb 1-1.1: device descriptor read/64, error -32
[    7.513432] usb 1-1.1: new full-speed USB device number 4 using dwc2
[    7.713589] usb 1-1.1: device descriptor read/64, error -32
[    8.029426] usb 1-1.1: device descriptor read/64, error -32
[    8.141559] usb 1-1-port1: attempt power cycle
[    8.849413] usb 1-1.1: new full-speed USB device number 5 using dwc2
[    8.869848] usb 1-1.1: device descriptor read/8, error -32
[    9.008084] usb 1-1.1: device descriptor read/8, error -32
[    9.301412] usb 1-1.1: new full-speed USB device number 6 using dwc2
[    9.321839] usb 1-1.1: device descriptor read/8, error -32
[    9.453838] usb 1-1.1: device descriptor read/8, error -32
[    9.565552] usb 1-1-port1: unable to enumerate USB device
[    9.913407] usb 1-1.2: new high-speed USB device number 7 using dwc2
[   10.134058] usb 1-1.2: New USB device found, idVendor=214b, idProduct=7250, bcdDevice= 1.00
[   10.134063] usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[   10.134066] usb 1-1.2: Product: USB2.0 HUB
[   10.134339] hub 1-1.2:1.0: USB hub found
[   10.421411] usb 1-1.2.2: new full-speed USB device number 8 using dwc2
[   10.527001] usb 1-1.2.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[   10.527006] usb 1-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   10.527009] usb 1-1.2.2: Product: USB JTAG/serial debug unit
[   10.527011] usb 1-1.2.2: Manufacturer: Espressif
[   10.527014] usb 1-1.2.2: SerialNumber: CC:8D:A2:EC:B2:DC
[   10.531444] cdc_acm 1-1.2.2:1.0: ttyACM0: USB ACM device
[   10.531480] usbcore: registered new interface driver cdc_acm
[   10.531483] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters`