home-assistant/operating-system #3021: Containers continuously deactivated, system is almost useless

MatteGary, 2023-12-26:
### Describe the issue you are experiencing

Hi,
since several days, my HA installation became really slow, almost useless. UI crashes every other minute, Alexa and Apple Home have problem in sending command to UI.
I’m running HASS OS on a x64 minipc, I have HA from 5 year, and more than 1 year on this HW.
I did some investigation, and I think the reason of all of this is related to the containers behind HA. I have a lot of identical logs in the Host registry

```
Dec 22 20:20:56 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-fc8db1ba81fa8e4fc1efe6e962c6a1b1c22db3153fe3fcdba2f0a23f467ffdcd-runc.ve78Ok.mount: Deactivated successfully.
Dec 22 20:20:58 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b4f04e67d1a73ad21ea784e13b561666fda2150913c93d8915863e6e2274da4c-runc.EEzPUc.mount: Deactivated successfully.
Dec 22 20:20:59 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-8546a3eb45afbfe7ba92e255af53b8e35c932e77c44a2ac9aff1b32dbf0e6b97-runc.3vgVgt.mount: Deactivated successfully.
Dec 22 20:21:26 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-fc8db1ba81fa8e4fc1efe6e962c6a1b1c22db3153fe3fcdba2f0a23f467ffdcd-runc.9QmutO.mount: Deactivated successfully.
Dec 22 20:21:28 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b4f04e67d1a73ad21ea784e13b561666fda2150913c93d8915863e6e2274da4c-runc.jzUyNk.mount: Deactivated successfully.
Dec 22 20:21:29 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-8546a3eb45afbfe7ba92e255af53b8e35c932e77c44a2ac9aff1b32dbf0e6b97-runc.F4kLGS.mount: Deactivated successfully.
Dec 22 20:21:56 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-fc8db1ba81fa8e4fc1efe6e962c6a1b1c22db3153fe3fcdba2f0a23f467ffdcd-runc.pZjFzJ.mount: Deactivated successfully.
Dec 22 20:22:56 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-fc8db1ba81fa8e4fc1efe6e962c6a1b1c22db3153fe3fcdba2f0a23f467ffdcd-runc.GekjlS.mount: Deactivated successfully.
Dec 22 20:22:59 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b4f04e67d1a73ad21ea784e13b561666fda2150913c93d8915863e6e2274da4c-runc.75gZY6.mount: Deactivated successfully.
Dec 22 20:23:26 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-fc8db1ba81fa8e4fc1efe6e962c6a1b1c22db3153fe3fcdba2f0a23f467ffdcd-runc.FdJirn.mount: Deactivated successfully.
Dec 22 20:23:30 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-8546a3eb45afbfe7ba92e255af53b8e35c932e77c44a2ac9aff1b32dbf0e6b97-runc.sLDQHZ.mount: Deactivated successfully.
Dec 22 20:23:59 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-b4f04e67d1a73ad21ea784e13b561666fda2150913c93d8915863e6e2274da4c-runc.62TxE4.mount: Deactivated successfully.
Dec 22 20:24:30 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-8546a3eb45afbfe7ba92e255af53b8e35c932e77c44a2ac9aff1b32dbf0e6b97-runc.idiXKt.mount: Deactivated successfully.
```

and these

```
23-12-26 17:37:19 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-26 17:37:43 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-26 17:37:43 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-26 17:37:48 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-26 17:37:48 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-12-26 17:37:48 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
```

These logs, thanks to another post in the community, led me to believe that some (or all) containers are constantly failing and being reactivated by the Watchdog.
I managed to understand which are the docker that keep getting deactivate while I was performing some troubleshooting:

b4f04e homeassistant/amd64-addon-configurator:5.7.0
8546a3 ghcr.io/hassio-addons/node-red/amd64:16.0.2
fc8db1 ghcr.io/hassio-addons/adguard/amd64:5.0.1

at this moment these are the only ones that are getting deactivated and activate all over again, but during the last 24 hours I noticed also others having the same behavior. How can I investigate more this issue? Where I can find the root cause of this activation/deactivation? Logs currently doesn't give any information on why everything seems to be failing.
There must be something, maybe on docker side, that can points me in the right direction.

Thanks
Matteo

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

I wish I knew, it just keep happening.

### Anything in the Supervisor logs that might be useful for us?

```txt
23-12-26 17:37:19 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-26 17:37:43 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-26 17:37:43 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-26 17:37:48 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
[... 165 lines cut ...]
version_api | ok
installed_addons | Samba share (12.2.0), File editor (5.7.0), Let's Encrypt (5.0.8), Mosquitto broker (6.4.0), Terminal & SSH (9.8.1), AppDaemon (0.16.0), Uptime Kuma (0.10.1), Home Assistant Google Drive Backup (0.112.1), Node-RED (16.0.2), ESPHome (2023.12.5), MariaDB (2.6.1), phpMyAdmin (0.9.0), AdGuard Home (5.0.1), Advanced SSH & Web Terminal (17.0.0), Portainer (2.19.4-3)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 29
views | 32
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 18 dicembre 2023 alle ore 07:01
-- | --
current_recorder_run | 26 dicembre 2023 alle ore 14:46
estimated_db_size | 38.69 MiB
database_engine | mysql
database_version | 10.6.12

</details>

agners, 2023-12-27:
Is there maybe more in the logs of Supervisor when the add-on shuts down? E.g. is it the Supervisor shutting down the add-on or is it the add-on itself which shuts itself down?

I'd guess an out-of-memory situation being the issue here. You can monitor memory usage on your system using https://www.home-assistant.io/integrations/systemmonitor?

Impact123, 2023-12-27:
These `*.mount` messages related to docker's health checks and very common: <https://github.com/docker/for-linux/issues/679>.

agners, 2023-12-27:
Ah yeah right the health checks cause this. So yes, this is regular operation and unlikely the culprit of the issue you are seeing. The slowdown it is likely caused by something else.

MatteGary, 2023-12-27:
> Is there maybe more in the logs of Supervisor when the add-on shuts down? E.g. is it the Supervisor shutting down the add-on or is it the add-on itself which shuts itself down?
>
> I'd guess an [...]

Resource doesn’t seems to be an issue (right now RAM 15%, CPU 4%, disk 6%).
Regarding supervisor logs this is the current extract (issue is still happening)

`23-12-27 13:16:11 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-27 13:16:11 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-12-27 13:16:11 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-12-27 13:17:58 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:17:59 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:18:25 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-27 13:18:25 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-27 13:18:30 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-27 13:18:30 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-12-27 13:18:30 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-12-27 13:20:20 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:20:28 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:20:44 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-27 13:20:44 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-27 13:20:50 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-27 13:20:50 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-12-27 13:20:50 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-12-27 13:22:41 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:22:58 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:23:04 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-27 13:23:04 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-27 13:23:09 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-27 13:23:09 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-12-27 13:23:09 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-12-27 13:24:58 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:25:02 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:25:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-27 13:25:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-27 13:25:29 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-27 13:25:29 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-12-27 13:25:29 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-12-27 13:27:23 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:27:29 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:27:43 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-27 13:27:43 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-27 13:27:48 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-27 13:27:48 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-12-27 13:27:48 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
23-12-27 13:29:34 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:29:59 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:30:02 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-27 13:30:02 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-27 13:30:07 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-27 13:30:07 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
[... cut ...]
23-12-27 13:46:19 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
23-12-27 13:46:19 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
23-12-27 13:46:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
23-12-27 13:46:24 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
23-12-27 13:46:24 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running `

MatteGary, 2023-12-27:
> These `*.mount` messages are likely related to docker's health checks and very normal/common: [docker/for-linux#679](https://github.com/docker/for-linux/issues/679).

Only few users reported the “deactivated successfully” logs in the issue you referred. They make the logs useless. How can I be sure those are “good” logs, and how can I investigate more my issue?

agners, 2023-12-27:
> 23-12-27 13:46:19 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257

Sounds like NodeRED sends some invalid websocket message? :thinking: 

You filed this as a Home Assistant OS issue, what makes you belive this is a OS issue? I think a new version/configuration change of another component of the system made this error to start appear.

agners, 2023-12-27:
Actually:

```

23-12-27 13:46:01 ERROR (MainThread) [supervisor.homeassistant.api] Error on call https://172.30.32.1:8123/api/core/state: 
23-12-27 13:46:19 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
```

This are independent API calls from the Supervisor to the Core. It looks more like your Core is having troubles. I'd check it's logs. Also make sure to disable custom components to see if any of these cause the issue.

MatteGary, 2023-12-27:
> > 23-12-27 13:46:19 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
> 
> Sounds like NodeRED sends some invalid websocket message? 🤔
> 
> You filed this as a Home Assistant OS issue, what makes you belive this is a OS issue? I think a new version/configuration change of another component of the system made this error to start appear.

How can you say this logs are related to NodeRED?
These are the logs from NodeRED

```
27 Dec 15:59:55 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:02:09 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:02:14 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:02:14 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:04:29 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:04:34 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:04:34 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:06:48 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:06:53 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:06:53 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:09:08 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:09:13 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:09:13 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:11:27 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:11:32 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:11:32 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:13:47 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:13:52 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:13:52 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:16:06 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:16:11 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:16:11 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:18:25 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:18:30 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:18:30 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:20:45 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:20:50 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:20:50 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:23:04 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:23:09 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:23:09 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:25:23 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:25:28 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:25:28 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:27:43 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:27:48 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:27:48 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:30:03 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:30:08 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:30:08 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:32:22 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:32:27 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:32:27 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:34:42 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:34:47 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:34:47 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:37:01 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:37:06 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:37:06 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:39:21 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:39:26 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:39:26 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:41:40 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
27 Dec 16:41:45 - [info] [server:Home Assistant] Connecting to http://supervisor/core
27 Dec 16:41:45 - [info] [server:Home Assistant] Connected to http://supervisor/core
27 Dec 16:43:59 - [info] [server:Home Assistant] Connection closed to http://supervisor/core
[... cut ...]
[INFO] 127.0.0.1:44356 - 49101 "PTR IN 137.136.99.36.in-addr.arpa. udp 55 true 2048" SERVFAIL qr,rd,ra 44 2.084173749s
```
I don't know if this is meant to be like this, or are a sign of some sort of failure.
I haven't added any new custom component in a while.

I think my point is: my HA OS installation is not working properly, there are evidence of issue in the logs but to me (I do this for a living) the logs are not enough to understand where is the issue and what caused it.

Do you know where to point me to get more information on the issue I'm experiencing?

Thanks

MatteGary, 2023-12-28:
In the end it may be related to a custom component... I added through HACS a component to connect to my ABB Solar Inverter, and it is not working (configuration error). After I removed it from the configuration, everything started to work fine again. I tried to add the configuration again, and the screen froze with a spinning wheel, the configuration was never added, and the usual logs reappeared. 

This is the error thrown by the component
```
Logger: aiohttp.server
Source: /usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py:421
First occurred: 13:19:58 (1 occurrences)
Last logged: 13:19:58

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 233, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 149, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 63, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 177, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 293, in async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 389, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/config_flow.py", line 90, in async_step_user
    uid = await self.test_connection(name, host, port, slave_id, base_addr, scan_interval)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/config_flow.py", line 65, in test_connection
    self.hub_data = await self.hub.async_get_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 185, in async_get_data
    self.read_sunspec_modbus_model_1()
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 221, in read_sunspec_modbus_model_1
    read_model_1_data.registers, byteorder=Endian.BIG
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 11 lines cut ...]
Last logged: 13:19:58

[140110664856768] Supervisor from 172.30.32.2 (HomeAssistantSupervisor/2023.12.0 aiohttp/3.9.1 Python/3.11): Unexpected error inside websocket API
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/http.py", line 371, in async_handle
    msg = await wsock.receive()
          ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_ws.py", line 472, in receive
    await self.pong(msg.data)
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_ws.py", line 329, in pong
    await self._writer.pong(message)
  File "/usr/local/lib/python3.11/site-packages/aiohttp/http_websocket.py", line 709, in pong
    await self._send_frame(message, WSMsgType.PONG)
  File "/usr/local/lib/python3.11/site-packages/aiohttp/http_websocket.py", line 682, in _send_frame
    self._write(header + message)
[... cut ...]
The entire UI froze for around 1 minute then everything started working again. Since it wasn't added correctly, the issue was momentarily. Before I had the configuration (it was working at some point) and I guess HA core kept retrying to setup the component, creating all the mess. 

How can it be that the fail of an integration cause so much issue to the entire system?
I already opened a issue to the integration creator, but he claim that since the entire HA system froze, the issue is not inside its component but somewhere else.

mnorrsken, 2023-12-28:
Try delete the custom component completely from the file system, restart, and try to re add it.

MatteGary, 2024-01-02:
> Try delete the custom component completely from the file system, restart, and try to re add it.

Tried but it didn't helped, the issue is still there every time I re-enable the custom component. I guess I can close this issue and see if the owner of the component do something about it.
Thanks to everyone.

MatteGary, 2024-01-02:
Close since it was an issue related to a custom component, even if not sure on how to solve it.

closed 2024-01-02 by MatteGary