home-assistant/operating-system #3022: HAOS 10.5 to 11.2 update failing (Intel Nuc)

cesa030, 2023-12-27:
### Describe the issue you are experiencing

I can't upgrade HAOS since 10.5. After hitting the install butto I get a notification to check supervisor.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Go to updates
2. Update Home Assistant Operating System
3. Get a notification to check supervisor

### Anything in the Supervisor logs that might be useful for us?

```txt
ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot rootfs.1: Syncing content to disk failed: Input/output error
```

### Anything in the Host logs that might be useful for us?

```txt
[140655392152256] Error updating Home Assistant Operating System: Unknown error, see supervisor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 212, in async_install
    await async_update_os(self.hass, version)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 53, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Unknown error, see supervisor

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 238, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2067, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2104, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 272, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 878, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 432, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 214, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating Home Assistant Operating System: Unknown error, see supervisor
```

### System information

Version | core-2023.12.3
Installationstyp | Home Assistant OS
Entwicklung | false
Supervisor | true
Docker | true
Benutzer | root
Virtuelle Umgebung | false
Python-Version | 3.11.6
Betriebssystemfamilie | Linux
Betriebssystem-Version | 6.1.45
CPU-Architektur | x86_64
Zeitzone | Europe/Berlin
Konfigurationsverzeichnis | /config

agners, 2023-12-27:
Can you also share the Supervisor logs? (Settings > System > Logs > choose Supervisor on the top right).

cesa030, 2023-12-27:
Sure,

23-12-27 14:30:21 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
23-12-27 14:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
23-12-27 14:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
23-12-27 14:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
23-12-27 14:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
23-12-27 14:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
23-12-27 14:31:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
23-12-27 14:32:53 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
23-12-27 14:32:53 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
23-12-27 14:32:53 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
23-12-27 14:32:53 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
23-12-27 14:32:53 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-12-27 14:32:53 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
23-12-27 14:32:54 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-12-27 14:32:54 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
23-12-27 14:32:54 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-12-27 14:40:08 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-12-27 14:42:35 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-12-27 14:44:49 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/11.2/haos_generic-x86-64-11.2.raucb
23-12-27 14:45:12 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-11.2.raucb
23-12-27 14:45:17 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot rootfs.1: Syncing content to disk failed: Input/output error

agners, 2023-12-27:
```
23-12-27 14:45:17 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot rootfs.1: Syncing content to disk failed: Input/output error
```

That sounds like a disk error, possible a hardware problem :cold_face: Maybe the host logs contains more information about the error.

closed 2024-04-03 by github-actions[bot]