home-assistant/operating-system #3023: Wifi drivers missing for Realtek 8822CE

vinculis, 2023-12-27:
### Describe the issue you are experiencing

I'm using mini PC and I'm missing wifi drivers for Realtek 8822CE (https://cateee.net/lkddb/web-lkddb/RTW88_8822CE.html).

No wifi card detected.
I think there is a driver missing for this chip.
Could you include this driver in the next image?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Install generic image on device with Intel Centrino Advanced-N 6205.
2. login
3. nmcli device

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### Additional information

Realtek 8822CE

agners, 2023-12-27:
Those got just very recently added with #2951. Can you try HAOS 11.3.rc1 (available on the Beta channel).

vinculis, 2023-12-28:
@agners that worked, thanks!

closed 2023-12-28 by vinculis