home-assistant/operating-system #3025: HaOS for Odroid M1

KarelHaruda, 2023-12-29:
### Describe the issue you are experiencing

HaOs for Odroid M1 not working.
I followed your instructions to install HAOS on odroid m1 (https://www.home-assistant.io/installation/odroid)
Everything went well, I uploaded the IMG to a 32GB SD card, but Odroid is unable to boot the OS from this card. Even Odroid won't even enter the boot loader when the card is inserted. I still updated the boot loader according to the procedure for eMMC, but the result is the same

### What operating system image do you use?

odroid-m1 (Hardkernel ODROID-M1)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Follow instruction here https://www.home-assistant.io/installation/odroid

Target SD card is not bootable. Odroid cannot start with inserted card.

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing to attache
```

### Anything in the Host logs that might be useful for us?

```txt
nothing
```

agners, 2023-12-29:
> Even Odroid won't even enter the boot loader

How do you know? Do you have access to the serial console?

Note that the image won't work for ODROID-M1S. The image is working fine on my ODROID-M1 here.

KarelHaruda, 2023-12-29:
I have ODROID M1 4GB Ram  (not M1S)

I have a monitor connected to it via HDMI. Without the SD card it shows the UBOOT botloader and if I put the SD card in and turn on Odroid then just a black screen and nothing happens. I tried waiting for a long time (hours) but still just a black screen. As soon as I remove the SD card and turn the power off and on, the boot loader immediately starts again. I currently have Debian 10 installed on my Odroid on an M.2 SSD and I have supervised HA installed there. Everything works perfectly, including the fact that the entire system boots directly from the M.2 disk and I don't need any SD card.
I wanted to try the option of having HaOS directly instead of Debian, but there is still the problem that HaOS cannot boot directly from M.2

agners, 2023-12-29:
The fact that the screen remains black tells me that the SPL on the internal SPI tries to boot the HAOS provided U-Boot. However, it seems that the HAOS U-Boot gets stuck for some reason. You'd need to hook up a serial console to check the boot logs of U-Boot.

The U-Boot shipped with HAOS 11.2 definitely works here on two ODROID-M1 boards. So this must be either a corrupted SD card or some new board revision which behaves differently.

KarelHaruda, 2023-12-29:
It won't be a broken sd card as I uploaded an IMG to the same card to upgrade the boot loader and that went well and the update went through fine.
I don't know if it is not related to the fact that a partition table is always created on that card which is GPT and not MBR. I messed around with it a bit and tried to upload the IMG via Linux using xzcat, but the result was the same. In addition, Linux fdisk reported some discrepancies in the partition table after uploading the IMG to the sd card. I first tried leaving it as is and then had fdisk fix it, but still the same result, black screen.
I'll try to connect there via serial but probably won't be right away as I need HA running now.

lennartschulte, 2024-01-05:
Any Update on this issue? Got the same problem no on my RasPi 4 after the update today.

KarelHaruda, 2024-01-06:
I'm currently waiting for a UART - USB converter so I can capture the odroid boot log. I'll let you know then

K.

karel-haruda-abra, 2024-01-24:
Sorry, it was a false alarm. Tried installing Debian 12 from netoboot on Odroid M1 and same result. Black screen. I searched and found that it might be an incompatible monitor. I was using an older LCD that only had VGA and I used a VGA to HDMI converter and that was really it. When I connected it to a new monitor with a direct connection via HDMI, both the installation of Debian 12 and HAOS works, it can be seen that it boots and starts up correctly.
You can tell by the way that the blue LED on the odroid board flashes but the screen is black = incompatible display (almost certainly)

karel-haruda-abra, 2024-01-24:
Can be closed .. :)

agners, 2024-01-24:
Thanks for the update!

closed 2024-01-24 by agners