home-assistant/operating-system #3036: update.home_assistant_operating_system_update ENTITY NOT AVAILABLE

spanzetta, 2024-01-04:
### Describe the issue you are experiencing

Starting from the latest OS builds I often see that the entity update.home_assistant_operating_system_update is not available..
Sometimes after a restart the entity returns available.. but not all the time.. 

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

11.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. simply start HA and see if that entity is available
2. sometimes it is not available
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing relevant
```

### Anything in the Host logs that might be useful for us?

```txt
nothing relevant
```

### System information

## System Information

version | core-2024.1.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.58-haos-raspi
arch | armv7l
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4999
Installed Version | 1.33.0
Stage | waiting
Available Repositories | 1378
Downloaded Repositories | 19
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.2
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 28.7 GB
disk_used | 13.4 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.7.0), SQLite Web (4.0.0), Duck DNS (1.15.0), Home Assistant Google Drive Backup (0.112.1), Glances (0.20.0), Advanced SSH & Web Terminal (17.0.1), Firefox (1.0.7), Samba share (12.2.0), Network UPS Tools (0.12.2), CEC Scanner (3.0), Firefox (Edge) (1.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 11
-- | --
resources | 10
views | 11
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 25, 2023 at 09:42
-- | --
current_recorder_run | January 4, 2024 at 16:55
estimated_db_size | 397.49 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.3 (a8c6d45)
-- | --
cloud_online | 1 / 1
local_online | 1 / 1

</details>

closed 2024-01-05 by github-actions[bot]

IIIdefconIII, 2024-03-04:
same here as of today, anyone knows a fix?

[image]

IIIdefconIII, 2024-03-14:
dunno what happen, i guess breaking changes with the new version?