home-assistant/operating-system #3045: Lost Node-Red Server after OS 11.3 update (aarch64 /odroid-n2 Blue)

radionerd1, 2024-01-05:
### Describe the issue you are experiencing

Updated OS 11.3, after Nod-Red stopped working.
All flows are stopped and in red  (Add-on  Ver 16.0.2), unknown, missing:
Lost server
api-call-service
server-state-changed
ha-fire-event

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

11.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. All Packages were up to date prior to OS update
2. Have full backup available 7 days ago 
3.updated OS 11.3
4. Nod-red stopped working
5. restated node-red and HA
6. restored last Core, no fix

...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-01-05 01:26:17.025 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration thermal_comfort which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.026 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration nodered which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.027 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration nws_alerts which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.032 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration weatheralerts which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.033 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration weatherflow_forecast which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.034 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration openhasp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.035 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration dahua which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.036 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration ble_monitor which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.037 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration keymaster which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:17.038 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-05 01:26:19.156 WARNING (Recorder) [homeassistant.components.recorder.util] The system could not validate that the sqlite3 database at //config/home-assistant_v2.db was shutdown cleanly
2024-01-05 01:26:19.178 WARNING (Recorder) [homeassistant.components.recorder.util] Ended unfinished session (id=621 from 2023-12-28 08:58:23.287510)
2024-01-05 01:26:26.573 ERROR (MainThread) [homeassistant.components.automation] Automation with alias 'Clothes Washer Running' failed to setup triggers and has been disabled: Unknown device '1a1eaec8ace86510537a6a83d9455126'
2024-01-05 01:26:26.575 ERROR (MainThread) [homeassistant.components.automation] Automation with alias 'Clothes Washer Idle' failed to setup triggers and has been disabled: Unknown device '1a1eaec8ace86510537a6a83d9455126'
2024-01-05 01:26:27.562 ERROR (MainThread) [homeassistant.components.automation] Blueprint 'HASP p[x].b[y] displays the current temperature with icon and colors' generated invalid automation with inputs {'suffix_select': '°', 'haspdevice': 'ace826f6c65cf43e0a15611ae7b713cf'}: Entity IDs cannot be None for dictionary value @ data['entity_id']. Got None
2024-01-05 01:26:27.584 ERROR (MainThread) [homeassistant.components.automation] Blueprint 'HASP p[x].b[y] displays the current temperature with icon and colors' generated invalid automation with inputs {'suffix_select': '°', 'haspdevice': 'ace826f6c65cf43e0a15611ae7b713cf', 'haspbutton': 5}: Entity IDs cannot be None for dictionary value @ data['entity_id']. Got None
2024-01-05 01:26:44.265 ERROR (SyncWorker_4) [pyicloud.base] Authentication required for Account. (421)
2024-01-05 01:26:46.323 ERROR (SyncWorker_4) [homeassistant.components.icloud.account] Your password for 'maryot@comcast.net' is no longer working; Go to the Integrations menu and click on Configure on the discovered Apple iCloud card to login again
2024-01-05 01:26:46.994 ERROR (MainThread) [homeassistant.components.automation] Automation with alias 'Front Door Unlocked ' failed to setup actions and has been disabled: Unknown device '1e4b3f7856d6d6b95df8820870c1c7f3'
2024-01-05 01:26:47.890 ERROR (Amcrest Bo-Cam) [homeassistant.components.amcrest] Bo-Cam camera offline: Login error:
2024-01-05 01:26:47.891 WARNING (Amcrest Bo-Cam) [homeassistant.components.amcrest] Error while processing events from Bo-Cam camera: LoginError()
2024-01-05 01:26:50.833 ERROR (MainThread) [homeassistant.components.camera] Error while setting up amcrest platform for camera
Traceback (most recent call last):
File "/usr/local/lib/python3.11/site-packages/amcrest/http.py", line 188, in _async_generate_token
resp = (await self._async_command(cmd)).content.decode()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/amcrest/http.py", line 388, in _async_command
raise LoginError()
amcrest.exceptions.LoginError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 361, in _async_setup_platform
await asyncio.shield(task)
File "/usr/src/homeassistant/homeassistant/components/amcrest/camera.py", line 146, in async_setup_platform
serial_number = await device.api.async_serial_number
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/amcrest/system.py", line 134, in async_serial_number
return pretty(await self._async_magic_box("getSerialNo"))
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/amcrest/http.py", line 511, in _async_magic_box
ret = await self.async_command(f"magicBox.cgi?action={action}")
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/amcrest/__init__.py", line 205, in async_command
ret = await super().async_command(*args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/amcrest/http.py", line 261, in async_command
await self._async_generate_token()
File "/usr/local/lib/python3.11/site-packages/amcrest/http.py", line 194, in _async_generate_token
resp = (await self._async_command(cmd)).content.decode()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/amcrest/http.py", line 388, in _async_command
raise LoginError()
amcrest.exceptions.LoginError
```

### Anything in the Host logs that might be useful for us?

```txt
Had problem with Nod-Red on prior Node-Red update (16.0.1), repaired prior problem by changing addon configs folder.
```

agners, 2024-01-06:
Can you try downgrading to HAOS 11.2 to see if that indeed fixes the problem? Use `ha os update --version 11.2`.

radionerd1, 2024-01-06:
Thanks,

There are several folks reporting same problem on Node-Red git after OS update. 
There is a work around posted below
I can try later, Wife happy now ;-)
https://github.com/hassio-addons/addon-node-red/issues/1786#issuecomment-1879650671

radionerd1, 2024-01-07:
@frenck Updated Node-Red 
This fixed the problem
Thanks!!!

closed 2024-01-07 by radionerd1