home-assistant/operating-system #3048: No Communication with Silicon Labs Radio on Home Assistant Yellow

ChristianForstner, 2024-01-06:
### Describe the issue you are experiencing

Hi there,

I recieved a brand new HA Yellow and installed HA OS using the installer created with RPi Imager.
Using the Silicon Labs Multiprotocol add-on to enable multiprotocol support fails as the universal_silabs_flasher cannot communicate with the radio.

Trying to install the firmware manually following the instructions at https://github.com/NabuCasa/silabs-firmware/wiki/Flash-Silicon-Labs-radio-firmware-manually results in the same error.

The issue is probably related to https://github.com/home-assistant/operating-system/issues/2639. At some point, I was able to see the radio at `/dev/ttyAMA4` when I (presumably wiped the `emmc` and) booted from an external USB stick with HA OS 9.5 flashed on it using the RPi Imager. However, the system booted with OS version 11.3.

Comparing the commits 
https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/board/raspberrypi/yellow/patches/linux/0002-ARM-dts-bcm2711-yellow-Mux-UART4-for-SiLabs-radio-mo.patch and 
https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/board/raspberrypi/yellow/patches/linux/0013-ARM-dts-bcm2711-yellow-use-serial-alias-to-enumerate.patch
shows that the radio was moved from `/dev/ttyAMA4` to `/dev/ttyAMA1` .

And
```
# ls -l /dev/tty[AS]*
crw-rw----    1 root     dialout   204,  64 Apr  4  2023 /dev/ttyAMA0
crw-rw----    1 root     dialout   204,  65 Jan  6 17:48 /dev/ttyAMA1
crw--w----    1 root     tty       204,  66 Jan  6 18:08 /dev/ttyAMA2
````
confirms that indeed there is no `/dev/ttyAMA4`.
Nevertheless, the communication via `/dev/ttyAMA1` is also not working. Manually starting the add-on with the configuration correctly set to `/dev/ttyAMA1` and a baudrate of `460800` results in the same output log when the firmware is flashed.

If there is `/dev/serial1`
```
# ls -l /dev/serial*
lrwxrwxrwx    1 root     root             7 Apr  4  2023 /dev/serial0 -> ttyAMA0
```
it is not on my system.

Some additional information:
````
# cat /etc/os-release 
NAME="Home Assistant OS"
VERSION="11.3 (Yellow)"
ID=haos
VERSION_ID=11.3
PRETTY_NAME="Home Assistant OS 11.3"
CPE_NAME=cpe:2.3:o:home-assistant:haos:11.3:*:production:*:*:*:yellow:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS Yellow"
VARIANT_ID=yellow
SUPERVISOR_MACHINE=yellow
SUPERVISOR_ARCH=aarch64
````

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

11.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Clean installation of HA OS on HA Yellow
2. Enable IEEE 802.15.4. radio multiprotocol support in `System -> Hardware -> Configure`
3. Log file of Silicon Labs Multiprotocol show output with error message

### Anything in the Supervisor logs that might be useful for us?

```txt
[... 390 lines cut ...]
await asyncio.sleep(ADDON_STATE_POLL_INTERVAL)
File "/usr/local/lib/python3.11/asyncio/tasks.py", line 649, in sleep
return await future
^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 320, in async_configure
result = await self._async_handle_step(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 416, in _async_handle_step
result: FlowResult = await getattr(flow, method)(user_input)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/silabs_multiprotocol_addon.py", line 535, in async_step_start_addon
await self.start_task
File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/silabs_multiprotocol_addon.py", line 344, in _resume_flow_when_done
await awaitable
File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/silabs_multiprotocol_addon.py", line 93, in async_start_addon_waiting
await self.async_wait_until_addon_state(AddonState.RUNNING)
File "/usr/src/homeassistant/homeassistant/components/homeassistant_hardware/silabs_multiprotocol_addon.py", line 76, in async_wait_until_addon_state
async with asyncio.timeout(ADDON_INFO_POLL_TIMEOUT):
File "/usr/local/lib/python3.11/asyncio/timeouts.py", line 111, in __aexit__
raise TimeoutError from exc_val
TimeoutError
```

ChristianForstner, 2024-01-07:
I was able to reproduce the configuration with booting from an external USB stick with HA OS 9.5 installed although the system version shows as HA OS 11.3.

Listing the serial ports
```
# ls -l /dev/tty[AS]*
crw-rw----    1 root     audio       204,  64 Jan   7  10:25 /dev/ttyAMA0
crw-rw----    1 root     audio       204,  68 Jan   7  10:53 /dev/ttyAMA4
crw--w----    1 root     audio       204,  69 Jan   7  10:24 /dev/ttyAMA5
crw--w----    1 root     audio         4,  64 Apr   4   2023 /dev/ttyS0
```
confirms the existence of `/dev/ttyAMA4`.

The radio can now be seen by the Silicon Labs Multiprotocol add-on as I suppose by the attached log. Nevertheless, flashing the firmware on the radio fails anyway, as now the same error as in https://github.com/home-assistant/addons/issues/2848 is shown.

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service mdns successfully started
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/check-cpcd-shm.sh
[09:56:49] INFO: Starting mDNS Responder...
Default: mDNSResponder (Engineering Build) (Jan  5 2024 22:17:54) starting
cont-init: info: /etc/cont-init.d/check-cpcd-shm.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting

-----------------------------------------------------------
 Add-on: Silicon Labs Multiprotocol
 Zigbee and OpenThread multiprotocol add-on
-----------------------------------------------------------
 Add-on version: 2.4.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.3  (aarch64 / yellow)
 Home Assistant Core: 2024.1.2
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[09:56:51] INFO: The selected serial port is not a USB device.
[09:56:51] WARNING: No firmware found for the selected device, assuming firmware is installed.
s6-rc: info: service universal-silabs-flasher successfully started
s6-rc: info: service cpcd-config: starting
[09:56:52] INFO: Generating cpcd configuration.
s6-rc: info: service cpcd-config successfully started
s6-rc: info: service cpcd: starting
[09:56:53] INFO: Starting cpcd...
WARNING in function 'main' in file /usr/src/cpc-daemon/main.c at line #186 : Running CPCd as 'root' is not recommended. Proceed at your own risk.
[09:56:53:320946] Info : [CPCd v4.3.1.0] [Library API v3] [RCP Protocol v4]
[09:56:53:321096] Info : Git commit: 133b29678b3d0bc7578e098d2f46b4d5bcd2ebb4 / branch: 
[09:56:53:321100] Info : Sources hash: ff8300587e7e4ab1def7a89a272c0baef32f9eb3bff9b0ba06b94e655d652367
[09:56:53:321121] WARNING : In function 'main' in file /usr/src/cpc-daemon/main.c at line #186 : Running CPCd as 'root' is not recommended. Proceed at your own risk.
[09:56:53:321165] Info : Reading cli arguments
[09:56:53:321174] Info : /usr/local/bin/cpcd 
[09:56:53:322096] Info : Reading configuration
[09:56:53:322114] Info :   file_path = /usr/local/etc/cpcd.conf
[09:56:53:322116] Info :   instance_name = cpcd_0
[09:56:53:322119] Info :   socket_folder = /dev/shm
[09:56:53:322121] Info :   operation_mode = MODE_NORMAL
[09:56:53:322124] Info :   use_encryption = false
[09:56:53:322126] Info :   binding_key_file = /etc/binding-key.key
[09:56:53:322128] Info :   stdout_tracing = false
[09:56:53:322131] Info :   file_tracing = false
[09:56:53:322133] Info :   lttng_tracing = false
[09:56:53:322135] Info :   enable_frame_trace = false
[09:56:53:322137] Info :   traces_folder = /dev/shm/cpcd-traces
[... 90 lines cut ...]
otbr-agent[181]: 49d.18:33:25.632 [W] P-RadioSpinel-: Wait for response timeout
otbr-agent[181]: 49d.18:33:25.632 [C] P-RadioSpinel-: Failed to communicate with RCP - no response from RCP during initialization
otbr-agent[181]: 49d.18:33:25.634 [C] P-RadioSpinel-: This is not a bug and typically due a config error (wrong URL parameters) or bad RCP image:
otbr-agent[181]: 49d.18:33:25.635 [C] P-RadioSpinel-: - Make sure RCP is running the correct firmware
otbr-agent[181]: 49d.18:33:25.635 [C] P-RadioSpinel-: - Double check the config parameters passed as `RadioURL` input
otbr-agent[181]: 49d.18:33:25.636 [C] Platform------: HandleRcpTimeout() at radio_spinel.cpp:2066: RadioSpinelNoResponse
[11:24:46] WARNING: otbr-agent exited with code 6 (by signal 0).
OTBR_FORWARD_INGRESS  all opt    in * out wpan0  ::/0  -> ::/0  
Chain OTBR_FORWARD_INGRESS (0 references)
target     prot opt source               destination         
DROP       all      anywhere             anywhere             PKTTYPE = unicast
DROP       all      anywhere             anywhere             match-set otbr-ingress-deny-src src
ACCEPT     all      anywhere             anywhere             match-set otbr-ingress-allow-dst dst
DROP       all      anywhere             anywhere             PKTTYPE = unicast
ACCEPT     all      anywhere             anywhere            
otbr-ingress-deny-src
otbr-ingress-deny-src-swap
otbr-ingress-allow-dst
otbr-ingress-allow-dst-swap
OTBR_FORWARD_EGRESS  all opt    in wpan0 out *  ::/0  -> ::/0  
Chain OTBR_FORWARD_EGRESS (0 references)
target     prot opt source               destination         
ACCEPT     all      anywhere             anywhere            
[11:24:46] INFO: OTBR firewall teardown completed.
```

agners, 2024-01-08:
The tty serial port for the on-board radio is and was `ttyAMA1`, as listed in the FAQ (see https://yellow.home-assistant.io/faq/#what-is-the-serial-port-of-the-zigbee-radio).

Not sure why/what happens in your USB boot case. Is there maybe a custom device tree, device tree overlay or config.txt active? Maybe the firmare/bootloader mixes device tree from the on-board eMMC with the kernel from the USB drive?

In any case, I highly recommend to try to work with the eMMC only installation, and the latest version of HAOS and try to find what is wrong there. My production instance works fine with HAOS 11.3 and Silicon Labs Multiprotocol here. A few things to check on your side:

1. Is there a ZHA/Zigbee Home Automation integration active? If so, what URL is it using? It should not point to the serial port directly

[image]

2. Do you have add-on installed which might access the serial port? E.g. check if "OpenThread Border Router" or Zigbee2MQTT is installed. If so, check that they are either shutdown or not use the serial port in question.

ChristianForstner, 2024-01-08:
Thanks for the fast response.

The USB boot case was more of a random discovery I made and is something that I would not use in production anyway. And, what you are suggesting, the mixing of the device tree of the eMMC installation and the kernel from the USB drive, could indeed be the case.

I just did again a clean installation of HAOS 11.3 (without recovering from any backup or so), so no addons are installed and only the default integrations (HA Supervisor, Met.no, Radio Browser, Shopping List, Sun) are configured as shipped with the installation by default.

The only addon I now install is the Silicon Labs Multiprotocol from the store. The configuration is 

[image]

The log shows the identical error I posted above:

```
[19:47:54] INFO: The otbr-web is disabled.
s6-rc: info: service mdns: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service mdns successfully started
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/check-cpcd-shm.sh
[19:47:54] INFO: Starting mDNS Responder...
Default: mDNSResponder (Engineering Build) (Jan  5 2024 22:17:54) starting
cont-init: info: /etc/cont-init.d/check-cpcd-shm.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting

-----------------------------------------------------------
 Add-on: Silicon Labs Multiprotocol
 Zigbee and OpenThread multiprotocol add-on
-----------------------------------------------------------
 Add-on version: 2.4.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.3  (aarch64 / yellow)
 Home Assistant Core: 2024.1.2
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[19:47:56] INFO: Detected Home Assistant Yellow
[19:47:56] INFO: Starting universal-silabs-flasher with /dev/ttyAMA1
2024-01-08 19:47:57.978 homeassistant universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.3.1', ezsp_version=None, ot_rcp_version=None, fw_type=<FirmwareImageType.RCP_UART_802154: 'rcp-uart-802154'>, baudrate=460800)
2024-01-08 19:47:57.979 homeassistant universal_silabs_flasher.flash INFO The '--yellow-gpio-reset' flag is deprecated. Use '--bootloader-reset' instead, see --help for details.
2024-01-08 19:47:57.980 homeassistant universal_silabs_flasher.flasher INFO Triggering yellow bootloader
2024-01-08 19:47:58.286 homeassistant universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-01-08 19:48:00.291 homeassistant universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2024-01-08 19:48:04.612 homeassistant universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2024-01-08 19:48:08.933 homeassistant universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 230400 baud
2024-01-08 19:48:13.255 homeassistant universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-01-08 19:48:18.268 homeassistant universal_silabs_flasher.flasher INFO Probing ApplicationType.SPINEL at 460800 baud
Error: Failed to probe running application type
s6-rc: warning: unable to start service universal-silabs-flasher: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service mdns: stopping
s6-rc: info: service banner: stopping
Default: mDNSResponder (Engineering Build) (Jan  5 2024 22:17:54) stopping
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
[19:48:22] INFO: mDNS ended with exit code 4 (signal 0)...
s6-rc: info: service mdns successfully stopped
[... 61 lines cut ...]
  File "/usr/local/lib/python3.11/site-packages/bellows/ezsp/__init__.py", line 151, in reset
    await self._gw.reset()
  File "/usr/local/lib/python3.11/site-packages/bellows/uart.py", line 261, in reset
    async with asyncio_timeout(RESET_TIMEOUT):
  File "/usr/local/lib/python3.11/asyncio/timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

and

```
Logger: py.warnings
Source: /usr/local/lib/python3.11/warnings.py:109
First occurred: 7:51:27 PM (5 occurrences)
Last logged: 7:51:27 PM

/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:288: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi' warnings.warn(
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:288: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio' warnings.warn(
[... cut ...]
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:285: NativePinFactoryFallback: Falling back to the experimental pin factory NativeFactory because no other pin factory could be loaded. For best results, install RPi.GPIO or pigpio. See https://gpiozero.readthedocs.io/en/stable/api_pins.html for more information. warnings.warn(NativePinFactoryFallback(native_fallback_message))
```

I'm not sure, if the missing RPi.GPIO package is causing all this trouble. But if it's really required, shouldn't it be shipped by default?

ChristianForstner, 2024-02-25:
I got a replacement and now it's working. Therefore, I will close the issue as resolved.

closed 2024-02-25 by ChristianForstner

pierrecdn, 2024-05-25:
Hi, new HA Yellow user here. Getting the exact same behavior with the zigbee stack.

Any advice / contact that can help to troubleshoot? If that's the only solution, what did you replace to have it working?