home-assistant/operating-system #3054: Cannot start SSH on haos_rpi5-64-11.3.img.xz

drorya1, 2024-01-08:
### Describe the issue you are experiencing

While trying to start the SSH service I get the following message 
2024/01/08 18:57:42:2759] N: ttyd 1.7.3-a8cae75 (libwebsockets 4.3.2-v4.3.2)
[2024/01/08 18:57:42:2760] N: tty configuration:
[2024/01/08 18:57:42:2761] N:   start command: tmux -u new -A -s homeassistant bash -l
[2024/01/08 18:57:42:2761] N:   close signal: SIGHUP (1)
[2024/01/08 18:57:42:2761] N:   terminal type: xterm-256color
[2024/01/08 18:57:42:2775] N:    /usr/lib/libwebsockets-evlib_uv.so
[2024/01/08 18:57:42:2776] N: lws_create_context: LWS: 4.3.2-v4.3.2, NET CLI SRV H1 H2 WS ConMon IPV6-off
[2024/01/08 18:57:42:2777] W: lws_create_context: unreasonable ulimit -n workaround
[2024/01/08 18:57:42:2778] N: elops_init_pt_uv:  Using foreign event loop...
[2024/01/08 18:57:42:2779] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
[2024/01/08 18:57:42:2780] N: __lws_lc_tag:  ++ [vh|0|netlink] (1)
[2024/01/08 18:57:42:2781] N: __lws_lc_tag:  ++ [vh|1|default||8099] (2)
[2024/01/08 18:57:42:2782] N: [vh|1|default||8099]: lws_socket_bind: source ads 0.0.0.0
[2024/01/08 18:57:42:2783] N: __lws_lc_tag:  ++ [wsi|1|listen|default||8099] (2)
[2024/01/08 18:57:42:2783] N:  Listening on port: 8099
s6-rc: info: service legacy-services successfully started
[18:57:42] WARNING: SSH port is disabled. Prevent start of SSH server.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.1.63-haos-raspi

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.boot up the system 
2.try to start the Advanced SSH & Web Terminal or Terminal & SSH addons
3. Addons cannot be started 
...

### Anything in the Supervisor logs that might be useful for us?

```txt
4-01-08 19:26:11 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Advanced SSH & Web Terminal to start, took more than 120 seconds
24-01-08 19:26:59 WARNING (MainThread) [supervisor.addons.addon] core_ssh is already running!
24-01-08 19:27:26 WARNING (MainThread) [supervisor.docker.addon] Advanced SSH & Web Terminal running with disabled protected mode!
24-01-08 19:27:26 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_a0d7b954_ssh application
24-01-08 19:27:26 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/ssh/aarch64 with version 17.0.2
24-01-08 19:27:32 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Terminal & SSH to start, took more than 120 seconds
24-01-08 19:28:31 WARNING (MainThread) [supervisor.addons.addon] a0d7b954_sonweb is already running!
24-01-08 19:28:59 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Terminal & SSH to start, took more than 120 seconds
24-01-08 19:29:26 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Advanced SSH & Web Terminal to start, took more than 120 seconds
24-01-08 19:29:51 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
24-01-08 19:29:51 WARNING (MainThread) [supervisor.addons.options] Option 'dark_mode' does not exist in the schema for Node-RED (a0d7b954_nodered)
24-01-08 19:29:51 WARNING (MainThread) [supervisor.addons.options] Option 'require_ssl' does not exist in the schema for Node-RED (a0d7b954_nodered)
24-01-08 19:30:31 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon TasmoAdmin to start, took more than 120 seconds
24-01-08 19:32:12 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
24-01-08 19:32:51 WARNING (MainThread) [supervisor.addons.addon] core_ssh is already running!
24-01-08 19:33:07 WARNING (MainThread) [supervisor.addons.addon] core_ssh is already running!
24-01-08 19:34:07 WARNING (MainThread) [supervisor.addons.addon] core_ssh is already running!
24-01-08 19:34:43 WARNING (MainThread) [supervisor.docker.addon] Advanced SSH & Web Terminal running with disabled protected mode!
24-01-08 19:34:43 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning addon_a0d7b954_ssh application
24-01-08 19:34:44 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/ssh/aarch64 with version 17.0.2
24-01-08 19:34:51 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
24-01-08 19:34:51 WARNING (MainThread) [supervisor.addons.options] Option 'dark_mode' does not exist in the schema for Node-RED (a0d7b954_nodered)
24-01-08 19:34:51 WARNING (MainThread) [supervisor.addons.options] Option 'require_ssl' does not exist in the schema for Node-RED (a0d7b954_nodered)
24-01-08 19:34:51 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Terminal & SSH to start, took more than 120 seconds
24-01-08 19:35:07 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Terminal & SSH to start, took more than 120 seconds
24-01-08 19:35:19 WARNING (MainThread) [supervisor.addons.addon] core_ssh is already running!
24-01-08 19:36:07 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Terminal & SSH to start, took more than 120 seconds
24-01-08 19:36:44 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Advanced SSH & Web Terminal to start, took more than 120 seconds
24-01-08 19:36:51 WARNING (MainThread) [supervisor.addons.addon] core_ssh is already running!
24-01-08 19:37:19 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Terminal & SSH to start, took more than 120 seconds
24-01-08 19:38:51 WARNING (MainThread) [supervisor.addons.addon] Timeout while waiting for addon Terminal & SSH to start, took more than 120 seconds
24-01-08 19:39:51 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
24-01-08 19:39:51 WARNING (MainThread) [supervisor.addons.options] Option 'dark_mode' does not exist in the schema for Node-RED (a0d7b954_nodered)
24-01-08 19:39:51 WARNING (MainThread) [supervisor.addons.options] Option 'require_ssl' does not exist in the schema for Node-RED (a0d7b954_nodered)
```

### Anything in the Host logs that might be useful for us?

```txt
2024/01/08 18:57:42:2759] N: ttyd 1.7.3-a8cae75 (libwebsockets 4.3.2-v4.3.2)
[2024/01/08 18:57:42:2760] N: tty configuration:
[2024/01/08 18:57:42:2761] N:   start command: tmux -u new -A -s homeassistant bash -l
[2024/01/08 18:57:42:2761] N:   close signal: SIGHUP (1)
[2024/01/08 18:57:42:2761] N:   terminal type: xterm-256color
[2024/01/08 18:57:42:2775] N:    /usr/lib/libwebsockets-evlib_uv.so
[2024/01/08 18:57:42:2776] N: lws_create_context: LWS: 4.3.2-v4.3.2, NET CLI SRV H1 H2 WS ConMon IPV6-off
[2024/01/08 18:57:42:2777] W: lws_create_context: unreasonable ulimit -n workaround
[2024/01/08 18:57:42:2778] N: elops_init_pt_uv:  Using foreign event loop...
[2024/01/08 18:57:42:2779] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
[2024/01/08 18:57:42:2780] N: __lws_lc_tag:  ++ [vh|0|netlink] (1)
[2024/01/08 18:57:42:2781] N: __lws_lc_tag:  ++ [vh|1|default||8099] (2)
[2024/01/08 18:57:42:2782] N: [vh|1|default||8099]: lws_socket_bind: source ads 0.0.0.0
[2024/01/08 18:57:42:2783] N: __lws_lc_tag:  ++ [wsi|1|listen|default||8099] (2)
[2024/01/08 18:57:42:2783] N:  Listening on port: 8099
s6-rc: info: service legacy-services successfully started
[18:57:42] WARNING: SSH port is disabled. Prevent start of SSH server.
```

### System information

<html>
<body>

Version | core-2024.1.2
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.11.6
Operating System Family | Linux
Operating System Version | 6.1.63-haos-raspi
CPU Architecture | aarch64
Timezone | Asia/Jerusalem
Configuration Directory | /config

Host Operating System | Home Assistant OS 11.3
-- | --
Update Channel | stable
Supervisor Version | supervisor-2023.12.0
Agent Version | 1.6.0
Docker Version | 24.0.7
Disk Total | 28.5 GB
Disk Used | 9.6 GB
Healthy | true
Supported | true
Board | rpi5-64
Supervisor API | ok
Version API | ok
Installed Add-ons | File editor (5.7.0), Advanced SSH & Web Terminal (17.0.2), Samba share (12.2.0), Node-RED (17.0.0), Check Home Assistant configuration (3.11.0), TasmoAdmin (0.29.0), Mosquitto broker (6.4.0), InfluxDB (5.0.0), FTP (5.0.1), Cloudflared (5.0.0), Ring-MQTT with Video Streaming (5.6.3), RTSPtoWeb - WebRTC (1.4.0), Home Assistant Google Drive Backup (0.112.1), Network UPS Tools (0.12.2), Terminal & SSH (9.8.1)

sairon, 2024-01-09:
It's likely not an OS issue per se, but the message makes it quite clear here:
```
 [18:57:42] WARNING: SSH port is disabled. Prevent start of SSH server.
 ```
You must set port the SSH will listen on in the Configuration tab. Also make sure you don't try to run both add-ons with the same port, either use only one of the add-ons or use different SSH port for each.

sairon, 2024-01-09:
> I dont have any ability to enable those ports on the GUI  unless I will manipulate the network part in the OS .

I don't get it. What makes you think so?

closed 2024-01-09 by sairon