home-assistant/operating-system #3061: Remote UI Connection lost

mweiss1965, 2024-01-10:
### Describe the issue you are experiencing

After Upgrading to OS 11.4 Remote UI could not connect.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.4 -> back to 11.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Updating OS from 11.3 to 11.4
2. Rebooting
3. Remote UI not Connected
4. Rollback 11.3
5. Remote UI Connected

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

RutgerDiehard, 2024-01-10:
I too experienced lack of connectivity after installing Home Assistant Operating System 11.4 on a Raspberry Pi 4. I had to power down/up twice before Home Assistant was accessible and shows HA OS 11.4 installed.

sairon, 2024-01-10:
@mweiss1965 Please try reproducing the issue again and providing logs as suggested in [Discussions](https://github.com/home-assistant/operating-system/discussions/3057#discussioncomment-8080343).

sairon, 2024-01-11:
Closed as reporter working after another update here: https://github.com/home-assistant/operating-system/discussions/3057#discussioncomment-8083458