home-assistant/operating-system #3067: deferred probe pending

dbrb2, 2024-01-12:
### Describe the issue you are experiencing

When rebooting a pi5 running HA OS from within HA, the boot hangs at "deferred probe pending" 
A power cycle works fine if I remove and replace the USB C connector, as does a restart of HA (not the system) from within HA

When running, everything seems perfectly stable. However, if I get into the state above all I can do is reflash and restore from backups. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 11.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Flash new HA OS onto an SSD, then boot from SSD using a pi5
2. Restore config and plugins from previously created backup 
3. A hard power cycle (removing and replacing power connector) works fine 
4. A soft restart from within HA works fine - "Restart Home Assistant"
5. A "Reboot System" from within HA hangs at boot, getting as far as the error: deferred probe pending
6. At this stage, the only resolution I have found is to reflash and restore from backup again 
7. If i avoid using "Reboot System" then the setup seems rocks solid with no problems at all

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing I can see
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing I can see
```

### System information

## System Information

version | core-2024.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.63-haos-raspi
arch | aarch64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1442
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 916.2 GB
disk_used | 18.0 GB
healthy | true
supported | true
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.15.0), Advanced SSH & Web Terminal (17.0.3), NGINX Home Assistant SSL proxy (3.7.0), TasmoAdmin (0.29.0), MariaDB (2.6.1), phpMyAdmin (0.9.1), Mosquitto broker (6.4.0), Home Assistant Google Drive Backup (0.112.1), GivTCP (2.4.3), Network UPS Tools (0.12.2), File editor (5.7.0), appdaemon-predbat (1.0.5)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 6
views | 9
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 1 January 2024 at 15:05
-- | --
current_recorder_run | 9 January 2024 at 23:01
estimated_db_size | 2230.67 MiB
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
used_requests | 6
rooftop_site_count | 2

</details>

iacavd, 2024-01-12:
same issue/message after updating to 11.4 on Rasberry pi 5 8G > USB boot 
rendered perfectly running HA useless cant even take backup

sandervg, 2024-01-12:
exact same issue here. running 11.4 on a pi 5 4G using a USB3 SSD adapter.

murray-m17, 2024-01-13:
just had this happen to me too, I'm running 11.4 on raspberry pi 5 4gb and I'm using an ssd connected through usb3 to sata adapter.

AgainstGreaterOdds, 2024-01-14:
Same here. Was working in the dev release, but upon update to 11.4 from within HA I get the same hang at boot.

The line says:
 [10.981171] platform kgdboc: deferred probe pending.

Any help?

joss044, 2024-01-16:
Same here. Same error message. I have a screen printout I can post if needed.

Hardware:
Rpi-5 with 4GB, set to USB boot priority
Samsung 250GB 860 EVO SSD on USB - no SD-card

martin131, 2024-01-17:
Same issue with my NVME to USB adapter

josephhungkk, 2024-01-18:
same issue with my HAT Drive Bottom

agners, 2024-01-19:
> When rebooting a pi5 running HA OS from within HA, the boot hangs at "deferred probe pending"

So if this is reboot related, then it is a duplicate of #3065.

If you have booting problems using an external drive from the start, it could be related to #3072 (e.g. probably @martin131's case)

@josephhungkk is your case on reboot only? If so, then this is most likely #3065 too. If a new installation doesn't boot, please open a new issue along with all information.

I am closing this as duplicate, since that is what the original post was about.

martin131, 2024-01-19:
@agners 
In my case, I booted into HAOS first time, restored from backup, and then rebooted. Now I cannot boot due to `deferred probe pending` error

agners, 2024-01-19:
@martin131 did you run a OS update? Or just a regular reboot?

martin131, 2024-01-19:
@agners  Only supervisor update, I think.

dbrb2, 2024-01-19:
> So if this is reboot related, then it is a duplicate of #3065.
>
> If you have booting problems using an external drive from the start, it could be related to #3072 (e.g. probably @martin131's case) [...]

It could be a duplicate, however the point at which the hang occurs is different, so it may not be the same underlying problem

agners, 2024-01-19:
> > When rebooting a pi5 running HA OS from within HA, the boot hangs at "deferred probe pending"
> 
> So if this is reboot related, then it is a duplicate of #3065.

Actually, this is not related to a missing/non-working `cmdline.txt`, as is #3065 about. So reopening.

The rootwait option causes the boot to hang.

It seems to me that the kernel is not able to find the USB device anymore :thinking: 

Thanks @sairon and @dbrb2 for bringing this up!

sairon, 2024-01-19:
> Actually, this is not related to a missing/non-working cmdline.txt, as is https://github.com/home-assistant/operating-system/issues/3065 about. So reopening.

Well - turns out it is. I tried installing HAOS on a USB drive and corrupted the `cmdline` parameter in `config.txt`. The behavior is exactly the same as describe in this thread, after mounting the `sda` device, boot hangs and only writes `platform kgdboc: deferred probe pending` after a few seconds. But the cause is indeed waiting for the rootfs. We can close this again, it is already fixed for upcoming updates.

For the broken systems, remedy is exactly the same as in the other ticket: 
https://github.com/home-assistant/operating-system/issues/3065#issuecomment-1898093065

closed 2024-01-19 by sairon

iacavd, 2024-01-20:
Hi does #3065 fixes this as well , both seem to related/same issue

iacavd, 2024-01-20:
Will test

zukunft-robots, 2024-02-17:
@agners @martin131 I have the same issue of "deferred probe pending " I bought new pi 5 and I am trying to install ubuntu version. I flashed ubuntu using pi imager, but while it's on booting i got an error sazing that "deferred probe pending" . If anyone knows about this issue please let me know.

agners, 2024-02-17:
@zukunft-robots this issue tracker is about Home Assist OS. I am not familiar with Ubuntu RPi 5 support.

RLB7777, 2024-10-22:
Same for me, with RP5 and a USB SSD. Rebooting the system (not Hassio, but rebooting the Pi) via the web interface causes it to hang and I have to pull the power from the RP5 to get it to restart.

kelddamsbo, 2024-10-30:
I also have exact the same problem, reboot unit not possible without hanging

kelddamsbo, 2024-10-30:
Yes agree

berlund, 2024-10-31:
I'm having the same issue. Restored a backup from an old installation on a Raspi 3B onto a Raspi 5. Now rebooting from the web ui will not work until I power-cycle the pi.

Core 2024.10.4
Supervisor 2024.10.3
Operating System 13.2
Frontend 20241002.4

https://github.com/home-assistant/core/issues/129560

user86000, 2024-11-01:
same problem
pi5 with ssd
sometimes reboot from ui works sometimes it hangs

k2dp2k, 2024-11-16:
same here... pi5 8gb with nvme using GeeekPi P31 M.2 NVME M-Key PoE+ HAT.... sometimes it restarts, most of the time it not... interesting fakt, the fan speeds up to it´s limit when stuck in reboot loop...

gioassio89, 2024-11-24:
Same here with web UI, RPi5 on Nvme heat+PoE.. i also tried to restore a full backup but the problem persist.

k2dp2k, 2024-11-24:
I give up fiddeling around with that Problem, i use a  mini pc with an n95 16gb of ram and an nvme powered by an poe Adapter… After small adjustments in the bios i get the same power consumption like the pi5… and nearly the same price…