home-assistant/operating-system #3070: HAOS upgrade fails to load on reboot

Altair31, 2024-01-13:
### Describe the issue you are experiencing

I'm not able anymore to upgrade HAOS to 11.4 version (and previous ones).
When I start the upgrade, everything works as expected until the reboot.
Here the system is stuck just after discovering the partitions (external SSD) and nothing else happens even if I wait for hours.
The only things that make the system react is to plug/unplug USB devices.

I have to reboot 3 times the system to get it work again on the 11.1 version of HAOS.
Any other module upgrades work fine.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start an upgrade of HAOS

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2024.1.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4877
Installed Version | 1.33.0
Stage | running
Available Repositories | 1371
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.0
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 54.3 GB
disk_used | 9.2 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | DHCP server (1.3.1), ESPHome (2023.12.5), File editor (5.7.0), NGINX Home Assistant SSL proxy (3.7.0), Network UPS Tools (0.13.0), Studio Code Server (5.15.0), Terminal & SSH (9.8.1), Dnsmasq (1.6.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 5
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 10 janvier 2024 à 11:05
-- | --
current_recorder_run | 13 janvier 2024 à 14:42
estimated_db_size | 167.51 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

user86000, 2024-01-14:
same too with raspberry pi 5
upgrade to has 11.4 fails

tested with boot from sd and from new pcie hat with nmve
teste with new system without addonsor extensions

first reeboot i.o, 2nd reboot fails

dwgtx, 2024-01-14:
I have the same issue, although I’m on an x86 NUC. Early research points to a cifs_mount failed w/return code = -101 error, which means the OS starts before the network connection is established.

dwgtx, 2024-01-15:
I'm running HAOS on an Intel NUC. Everything was running well (11.3) until the upgrade to 11.4, after which I was unable to access the homeassistant.local port. Viewing the NUC locally revealed a cifs_mount failed w/return code = -101 error, which means the OS starts before the network connection is established. Tried several 11.4 reinstalls to no avail - same issue after each reboot. Note that HAOS 11.4 appears to boot correctly but without network access. I did some research and saw suggestions about disabling IPV6, or modifying config files to add a delay to wait for an established network, but I've never had to worry about that prior to 11.4, and don't need want to worry about it now. I'll wait for the HA team 11.4.1 release which fixes their mistake.
I reinstalled 11.3 and restored my last daily backup. It's nice to be back in Home Automation Heaven...

Altair31, 2024-01-20:
Hi @dwgtx ,
It seams that your issue is not the same than mine since your HAOS boots (mine does not).
You should better open your own ticket to get a proper answer to your issue.

sairon, 2024-01-24:
From your description it seems the kernel is waiting for the root filesystem to be mounted, but for some reason it never happens. Would it be possible to check f it behaves the same with a fresh install on the same HW?

Altair31, 2024-01-24:
BTW, I have no boot issue if I reboot without HAOS upgrade

j3sch, 2024-02-01:
I run HA on a proxmox server. After upgrading to 11.4 I couldn't access HA anymore. I tried rebooting the virtual machine multiply times and it didn't work. Then I stopped the virtual machine and started it again and now everythings works fine.

Altair31, 2024-02-09:
Issue is still there on 11.5

closed 2024-05-18 by github-actions[bot]