home-assistant/operating-system #3072: Raspberry PI5 boot from USB3.0 port not possible

ragman11, 2024-01-14:
### Describe the issue you are experiencing

Boot from USB with an SSK NVME Adapter. Works fine when using the USB2.0 ports on the PI5, When I plug it into the USB3.0 port no boot possible. Get stick after a few seconds.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Boot from USB SSK NVME adapter on usb2.0 port - HA is running fine.
2. trying the same plugged into the USB3.0 port - no boot possible

### Anything in the Supervisor logs that might be useful for us?

```txt
na
```

### Anything in the Host logs that might be useful for us?

```txt
na
```

### System information

## System Information

version | core-2024.1.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.63-haos-raspi
arch | aarch64
timezone | Europe/Vienna
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4998
Installed Version | 1.33.0
Stage | running
Available Repositories | 1378
Downloaded Repositories | 34

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 228.5 GB
disk_used | 9.0 GB
healthy | true
supported | true
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.4.0), HoyMiles Solar Data Gateway Add-on (0.24.1d), Terminal & SSH (9.8.1), File editor (5.7.0), HoyMiles Solar Gateway Addon (1.0.9), evcc (0.123.8), Studio Code Server (5.15.0), HoyMiles Solar Gateway Edge Addon (1.2.2), Home Assistant Google Drive Backup (0.112.1), OneDrive Backup (2.2.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 34
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 4, 2024 at 13:19
-- | --
current_recorder_run | January 14, 2024 at 09:47
estimated_db_size | 1068.65 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

ragman11, 2024-01-14:
it was a faulty cable, working fine now

ragman11, 2024-01-17:
I tested it again with USB 3.0 cables, 3 different nvme Adapter, all are working and booting on PI4 USB 3.0 port

ragman11, 2024-01-19:
[image]
With an USB3 Cable it does not boot, USB 2 cable runs fine

manmei80, 2024-01-19:
I have no problem with booting from usb3. Ugreen usb3.2 case with kingston a400 ssd.

ragman11, 2024-01-19:
> I have no problem with booting from usb3. Ugreen usb3.2 case with kingston a400 ssd.

are you using a usb 3.2 cable?  blue inside usb connector. With a usb2.0 cable I can boot too.
I tried the same ugreen 3.2 with 2 different nvme

manmei80, 2024-01-19:
I use the cable that comes with the case. It is a 3.2 cable.

ragman11, 2024-01-19:
> I use the cable that comes with the case. It is a 3.2 cable.

ok, still strange, I‘m using a 
Intenso 250GB M.2 SSD PCIe Premium, 2100 MB/s, (PCI Express Gen.3x4 NVMe 1.3, Solid State Drive)

second one is
fanxiang S500 Pro 256GB NVMe SSD M.2 PCIe Gen3x4 2280 SLC Cache 3D NAND TLC, Bis zu 2800 MB/s

Raspian is working fine with this seTup on USB3

ryansyah, 2024-01-20:
I am seeing similar behavior. 

Booting haos_rpi5-64-11.5.dev20240117 on a nvme ssd in ugreen USB enclosure. Using enclosure provided USB 3.2 cable in USB3 port (both/either).

Drive: [TEAMGROUP MP44 1TB SLC Cache Gen 4x4 M.2 2280 PCIe 4.0](https://a.co/d/aVKjAOL)

Enclosure: [UGREEN SSD Enclosure](https://a.co/d/fvEfetG)

Booting from sdcard works fine. Just found out that booting using the USB2 ports also work fine.

closed 2024-04-30 by github-actions[bot]

slapin95, 2024-08-11:
Same on pi4
I have changed in cmdline.txt (from a PC)
dwc_otg.lpm_enable=0 console=tty1 usb-storage.quirks=174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u,174c:225c:u,7825:a2a4:u
to
dwc_otg.lpm_enable=0 console=tty1 usb-storage.quirks=174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u,174c:225c:u,7825:a2a4:u,152d:0562:u