home-assistant/operating-system #3075: USB ports lost on Intel Nuc at random intervals

djansen1987, 2024-01-15:
### Describe the issue you are experiencing

My zigbee2mqqt was failing once in a while. Rebooting the system resolved it. Today it happened again so i started digging through the log files, there i found in de debug log of zigbee2mqtt that the USB Dongle was suddenly disconnected. Looking into the HA core logfiles i could see at that time not only my zigbee dongle became unavailable, but also my Bluetooth chip and DSMR P1 USB dongle.

If i go into the terminal and do a less /dev/tty… there is no USB device available.

I Also have a Intel Nuc, NUC5i3RYH with HA:
Home Assistant 2024.1.3
Supervisor 2023.12.0
Operating System 11.4
Frontend 20240104.0

The USB dongle is also extended with a sort usb extend cable (20 cm)

If i go the all hardware under system i see a couple of usb ports listed as:
```

**usb1**
Subsystem:
usb
Device path:
/dev/bus/usb/001/001
Attributes:
BUSNUM: '001'
DEVNAME: /dev/bus/usb/001/001
DEVNUM: '001'
DEVPATH: /devices/pci0000:00/0000:00:1d.0/usb1
DEVTYPE: usb_device
DRIVER: usb
ID_BUS: usb
ID_FOR_SEAT: usb-pci-0000_00_1d_0
ID_MODEL: EHCI_Host_Controller
ID_MODEL_ENC: EHCI\x20Host\x20Controller
ID_MODEL_ID: '0002'
ID_PATH: pci-0000:00:1d.0
ID_PATH_TAG: pci-0000_00_1d_0
ID_REVISION: '0601'
ID_SERIAL: Linux_6.1.71-haos_ehci_hcd_EHCI_Host_Controller_0000:00:1d.0
ID_SERIAL_SHORT: 0000:00:1d.0
ID_USB_INTERFACES: ':090000:'
ID_USB_MODEL: EHCI_Host_Controller
ID_USB_MODEL_ENC: EHCI\x20Host\x20Controller
ID_USB_MODEL_ID: '0002'
ID_USB_REVISION: '0601'
ID_USB_SERIAL: Linux_6.1.71-haos_ehci_hcd_EHCI_Host_Controller_0000:00:1d.0
ID_USB_SERIAL_SHORT: 0000:00:1d.0
ID_USB_VENDOR: Linux_6.1.71-haos_ehci_hcd
ID_USB_VENDOR_ENC: Linux\x206.1.71-haos\x20ehci_hcd
ID_USB_VENDOR_ID: 1d6b
ID_VENDOR: Linux_6.1.71-haos_ehci_hcd
ID_VENDOR_ENC: Linux\x206.1.71-haos\x20ehci_hcd
ID_VENDOR_ID: 1d6b
MAJOR: '189'
MINOR: '0'
PRODUCT: 1d6b/2/601
SUBSYSTEM: usb
TAGS: ':seat:'
TYPE: 9/0/0
USEC_INITIALIZED: '1594405'

**usb2**
Subsystem:
usb
Device path:
/dev/bus/usb/002/001
Attributes:
BUSNUM: '002'
DEVNAME: /dev/bus/usb/002/001
DEVNUM: '001'
DEVPATH: /devices/pci0000:00/0000:00:14.0/usb2
[... 392 lines cut ...]
dashboards | 2
-- | --
resources | 6
views | 31
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 5, 2024 at 5:38 PM
-- | --
current_recorder_run | January 15, 2024 at 10:22 AM
estimated_db_size | 2373.16 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

djansen1987, 2024-01-16:
Hi @agners, thanks for you quick reply. Unfortunally when i created [my post ](https://community.home-assistant.io/t/usb-ports-lost/446289/9?u=djansen1987) 2 months back i was running version 11.1 and it might even already been in an OS version before that. Strange part is it could be running days, sometimes weeks without an issue. But sometimes also twice a day. Nothing noteworting happening that can relate the events for me. I will maintain looking in the logs when it occurse again, but this makes the wife very unhappy...

agners, 2024-01-16:
@djansen1987 you can also go further back then 11.1, e.g. 10.5. You are certain that the same hardware combination used to work reliably at one point?

From the host logs you've provided (which include the kernel logs) it seems they are cut-off or really doesn't have any issue reported (no log entries from January 16th).

It does sound like it is OS related indeed since the device doesn't come back even after unplugging/replugging, but I'd expect kernel errors in this case.

Can you double check the kernel logs using `ha host logs --identifier kernel -n 10000`? Especially for that timeframe the Core and Z2M reported issues?

Somapolice, 2024-01-16:
I am having the same issue as you with a Dell Wyse 5070 running 11.4

First the bluetooth times out then it cascades to my USB ports timing out (I have a zwave controller, a zigbee controller, and a coral tpu connected via bluetooth).

khayamgondal, 2024-01-17:
Same problem here for last 6 months. Hardware is Intel NUC 5i5RYK. I have tried several Linux kernel, Ubuntu and Home Assistant versions but the problem persists.  I have a Zwave and zigbee USB sticks connected. I guess it is either a problem with the USB hub (Bluetooth is internally connected to the USB HUB) or a power supply issue. I am using a generic 65W PSU.
Note: I am using HA docker not HAOS

djansen1987, 2024-01-25:
Hi @agners, did you find anything in the logs ? It occurred multiple times in the meanwhile. If need i can collect more logs

agners, 2024-02-02:
The logs show various USB related issues, e.g. this:

```
Jan 15 08:53:06 homeassistant kernel: usb 2-7: reset full-speed USB device number 5 using xhci_hcd
Jan 15 08:53:21 homeassistant kernel: usb 2-7: device descriptor read/64, error -110
Jan 15 08:53:37 homeassistant kernel: usb 2-7: device descriptor read/64, error -110
Jan 15 08:53:37 homeassistant kernel: usb 2-7: reset full-speed USB device number 5 using xhci_hcd
Jan 15 08:53:53 homeassistant kernel: usb 2-7: device descriptor read/64, error -110
Jan 15 08:54:09 homeassistant kernel: usb 2-7: device descriptor read/64, error -110
Jan 15 08:54:09 homeassistant kernel: usb 2-7: reset full-speed USB device number 5 using xhci_hcd
Jan 15 08:54:14 homeassistant kernel: xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Jan 15 08:54:20 homeassistant kernel: xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Jan 15 08:54:20 homeassistant kernel: usb 2-7: device not accepting address 5, error -62
Jan 15 08:54:20 homeassistant kernel: usb 2-7: reset full-speed USB device number 5 using xhci_hcd
Jan 15 08:54:31 homeassistant kernel: xhci_hcd 0000:00:14.0: Abort failed to stop command ring: -110
Jan 15 08:54:31 homeassistant kernel: xhci_hcd 0000:00:14.0: xHCI host controller not responding, assume dead
Jan 15 08:54:31 homeassistant kernel: xhci_hcd 0000:00:14.0: HC died; cleaning up
Jan 15 08:54:31 homeassistant kernel: xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
Jan 15 08:54:31 homeassistant kernel: usb 2-1: USB disconnect, device number 2
Jan 15 08:54:31 homeassistant kernel: usb 2-1: failed to send control message: -19
Jan 15 08:54:31 homeassistant kernel: ch341-uart ttyUSB1: ch341-uart converter now disconnected from ttyUSB1
Jan 15 08:54:31 homeassistant kernel: ch341 2-1:1.0: device disconnected
Jan 15 08:54:31 homeassistant kernel: usb 2-3: USB disconnect, device number 3
Jan 15 08:54:31 homeassistant kernel: ftdi_sio ttyUSB0: error from flowcontrol urb
Jan 15 08:54:31 homeassistant kernel: ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
Jan 15 08:54:31 homeassistant kernel: ftdi_sio 2-3:1.0: device disconnected
Jan 15 08:54:31 homeassistant kernel: usb 2-4: USB disconnect, device number 4
Jan 15 08:54:31 homeassistant kernel: usb 2-7: USB disconnect, device number 0
```

This is most likely a Linux kernel issue. Given that USB seems to work rather reliable for must users, it seems to be some interaction with newer stable kernel releases and your particular system.

We are soon going to update to Linux 6.6, initial development version on the dev channel will soon follow once https://github.com/home-assistant/operating-system/pull/3063 is merged. Maybe this new major release helps? :thinking: :crossed_fingers:

djansen1987, 2024-02-02:
Okay, goed to hear it is still moving forward. Will wait for the new kernal update to be production ready

henriklund, 2024-02-25:
@djansen1987 what Zigbee dongle are you using?

sairon, 2024-02-26:
@djansen1987 OS version 12 with the updated kernel is in the stable channel, please let us know if it fixed your issues.

djansen1987, 2024-02-27:
@henriklund i have a "zzh! CC2652R Multiprotocol RF Stick" with the latest firmware. I do have to say i has been a while for it to break. What i also noticed but it might be a coincidence. That when it happend the stick was in a continues "permit join" state as i forgot to turn it of a couple of time. Did not yet had the time to see if i could reproduce. 

PS:
Maybe unrelated but, last night my homeassistant web interface was not responding so was the app. I was able to run turn off/on devices via google -> nabu casa. When i when to the console screen on my NUC i saw the following:

[image]
(ignore the ```````` on the command line as that was me pluggin in a keyboard)

henriklund, 2024-02-28:
My observations are somewhat similar and then again...  running with a ConBee II and Aeotec ZWave Stick for a few years. Until recently I (also) ran Home Assistant from a NUC. Although all devices worked like a charm after a reboot, I would frequently (within days or 1-2 weeks) experience the USB sticks getting disconnected. On a few occassions they reconnected, but the predominant experience was that they remained disconnected. This was a bit of a nuisance for which I had to create a few automations to monitor the state and implement responses (restart integration or even reboot the host). Two months ago I decided to run HA virtualized (on Proxmox) as means to better utilize the NUC capacity. All hardware remained the same. I noted an improved handling of the hardware (less frequent disconnects of USB), Only problem now was that the Zigbee network (through ConBee II) started becoming unstable (I saw errors in the logs that indicated some capacity constrain / device fault). I opted for replacing with ConBee III (less hassle on migration) and behold, no more USB disconnects. To say I was pleased is an understatement.

That is until I started rebooting the NUC. Now the ZWave chimes in with a connect, only to disconnect 20sec later:
[image]

Ejecting and reinserting the USB (from Proxmox by removing / adding the device) works without problems and afterwards everything is stable.
[image]

(this was also the case before the latest update).

sairon, 2024-02-28:
@henriklund So do I understand it correctly that with bare-metal install, you had more disconnects, and with ConBee III in a VM, it does not happen anymore?

About the first log capture, if you don't filter for USB, is there anything around that USB disconnect event that would justify it?

henriklund, 2024-02-28:
@sairon you are correct. My infrastructure seems far better off now (running virtualized and with ConBee III) than two months ago when it ran bare-metal. As soon as all is up and running everything appears stable; nothing in logs, no sudden lights failing to turn on / off. The only hiccup left if the odd disconnect of the ZWave USB during boot. And even here I yesterday saw the odd case (when trying it out) where everything started up and worked; it happened once during 5 tries.

This is what I see relates to discovering USBs:
[image]
Then a lot of logging related to apparmour, docker, network, CIFS before this:
[image]

henriklund, 2024-03-08:
@sairon, I do not know if my error qualifies as a new issue (I do not want to hijack this one). If so, please let me know.

However, I had a strange observation today. I was trying to experiment with some simple monitoring of my ZWave. While doing so I restarted ZWave JS UI (to simulate loss of ZWave). What intrigued me was that the first restart went fine while the second caused the USB to become disconnected. Time between restarts did not seem to influence the outcome (whether it was 2 minutes or 5); the USB was no longer visible in ` /dev/serial/by-id/` and last entry was a USB disconnect. A quick remove and add in Proxmox (ie. a virtual inject insert) and USB was back.

henriklund, 2024-03-08:
Worth mentioning; this appears to a consistent behaviour. Second restart makes the USB disconnect.

sairon, 2024-03-11:
@henriklund Problems with the ZWave sticks are tracked in #2995, and there is a fix/workaround in 12.1.rc1. Let's keep this issue only for problems not related to ZWave sticks, and continue in discussion in the other one if the ZWave problems persist for you after upgrading to 12.1.rc1.

closed 2024-06-17 by github-actions[bot]

acvigue, 2024-07-22:
Seeing the same issue with a 5th gen i5 nuc

henriklund, 2024-07-23:
@acvigue, this issue is closed. I think it would be better to raise a new request to track. I decided to do something in an attempt to ‘fix’ this since I ran HA inside a ProxMox on my NUC. So far it has not reoccurred since. The rough outline was that I:
- installed an Alpine LXC
- Installed ZwaveJS on Alpine
- Made a complete copy of ZWaveJS configuration from HA docker to Alpine
- Changed USB pass through to Alpine
- Stopped HA ZwaveJS and pointed ZWave to Alpine
- started ZwaveJS on Alpine and checked everything was OK (ie. all devices found)
This is not for the faint hearted, however, so far everything behaves perfectly and no more drops of the Zwave stick

Martinvdm, 2024-08-25:
Although this is an old topic, it seems that i have the same combination. Conbee II, Aeontec Zwave. After enabling Bluetooth integration, it occasionally disconnects USBs.