home-assistant/operating-system #3078: IPv6 functionality is broken

Vectrobe, 2024-01-16:
### Describe the issue you are experiencing

Any kind of network operation with any ipv6 address is completely broken, most commands such as ping return "bad address" or "Address not available"
This is despite ipv6 being available and the device being assigned correct local and global ipv6 addresses, and having direct access to a working ipv6 network, it is also still possible to ping the homeassistant device externally using mDNS, and it will both return the correct ipv6 local fe80 address, and successfully ping the device consistently. The problem is internal to the OS and its networking functionality.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. install the OS to a device (whether or not its specific to ARM64 is unclear)
2. ensure its connected to a working ipv6 network
3. run 'ping -6 homeassistant' or 'ping ::1', it will fail

### Anything in the Supervisor logs that might be useful for us?

```txt
WARNING (MainThread) [supervisor.api.ingress] No valid ingress session
```

### Anything in the Host logs that might be useful for us?

```txt
unknown
```

### System information

## System Information

version | core-2024.1.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.63-haos-raspi
arch | aarch64
timezone | Australia/Sydney
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 57.8 GB
disk_used | 5.9 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | CEC Scanner (3.0), VLC (0.3.0), File editor (5.7.0), Terminal & SSH (9.8.1), Matter Server (5.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 15, 2024 at 4:43 AM
-- | --
current_recorder_run | January 15, 2024 at 4:02 PM
estimated_db_size | 90.59 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Additional information

moved from https://github.com/home-assistant/core/issues/108071#issuecomment-1891523208

documentation failed to mention this for OS related faults

agners, 2024-01-16:
> run 'ping -6 homeassistant' or 'ping ::1', it will fail

In which context do you run this command? I assume using the SSH/Web terminal?

The SSH/Web terminal runs as an add-on, and for add-ons we indeed do not have IPv6 support currently. This is a Docker/Container/Supervisor limitations. There have been initial works being done on OS side, and attempts in implementing IPv6 support for add-ons on the Supervisor side (see https://github.com/home-assistant/supervisor/issues/2133). But currently support is not completed yet.

In general, HAOS itself has IPv6 support. Home Assistant Core, which runs the main web server to deliver the front-end and connects to devices through it's integration runs on host network, and therefor supports IPv6 too. Add-on such as Matter and the OpenThread add-on, both technologies rely on IPv6, run in host network context too, and have IPv6 support.

> This is despite ipv6 being available and the device being assigned correct local and global ipv6 addresses, and having direct access to a working ipv6 network, it is also still possible to ping the [...]

mDNS and pinging link local IPv6 addresses should work fine from the host network context. Did you try to just use the mDNS name in Home Assistant Core?

Vectrobe, 2024-01-16:
So if its specifically the web side terminal at fault I would have to use the physical display output and keyboard to verify?

agners, 2024-01-16:
Yes, or you can use the "Advanced SSH & Web Terminal" from the community store. It runs on host network too (as indicated by the icon in the add-on store).

mytering, 2024-02-10:
id like to confirm i experience the same thing after updating to 2024.2.1
because of that my matter server also shows error 

2024-02-10 12:39:34 core-matter-server zeroconf[127] WARNING Error with socket 66 (('::1', 5353, 0, 0))): [Errno 101] Network is unreachable
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 1200, in sendto
    self._sock.sendto(data, addr)
OSError: [Errno 101] Network is unreachable

latargere, 2024-02-16:
Same here ipv6 seems broken since the last update

agners, 2024-02-16:
@mytering @latargere the issue you are seeing in the Matter add-on is not an issue per see https://github.com/home-assistant/core/issues/109926#issuecomment-1934379702. This is a library bug trying to use the IPv6 localhost (`::1`) for multicast, which is not supported. This is not problematic for Matter operation. Please ignore the message, we are working on getting the localhost interface ignored.

agners, 2024-02-16:
I am closing this issue as this is not a operating system issue per-se (see comment above). The missing IPv6 functionality for add-ons is handled by the linked issue above.

closed 2024-02-16 by agners