home-assistant/operating-system #3082: [supervisor.homeassistant.core] Error on Home Assistant installation.

meyxiam, 2024-01-16:
### Describe the issue you are experiencing

Bonjour,

Je suis sur RPI5 et avec la version 11.3. (ou 11.4) sur SDCard Sandisk de 64Go classe 10 J'ai voulu installer le module complémentaire Piper et il m'a demandé d'installer la version suivante du supervisor. J'ai redémarré et là kernel error sur le point de montage. Plus rien à faire ... Et ça me fait plusieurs fois que ça arrive.
Donc je décide de de me résintaller une image de la 11.4 sur une vieille clé USB de 4 Go. Je la branche sur le RPI5, ça boot et l'installation commence. Pour au bout de 20 min ne pas arriver à lancer mon HA.

J'ai une sauvegarde, j'espère que je pourrai l'utiliser en ayant changé la sdCard par une clé USB.
Voici les logs [supervisor.homeassistant.core] Error on Home Assistant installation.

[logs (1).txt](https://github.com/home-assistant/operating-system/files/13956417/logs.1.txt)

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.Mettre l'image 11.4 ou 11.3 (j'ai essayé les 2) sur clé USB 4Go branchée en USB 3
2.Démarrer le RPI
3.Attendre 20 min l'install et accéder à http://homeassistant.local:8123/ et constater le pb.

### Anything in the Supervisor logs that might be useful for us?

```txt
non
```

### Anything in the Host logs that might be useful for us?

```txt
non
```

AlexanderJHi, 2024-01-17:
same issue here with 11.4 VMDX COre is broken CLI wont' start. Configuration.yaml missing, i managed to add it but this then created more errors.

sairon, 2024-01-24:
The original issues looks like a duplicate of #3065, where `config.txt` is not correctly updated after RPi 5 OS update. Should be fixed by the next OS upgrade, or follow the steps described in the issue.

Also, only please use English for issue reports. While we usually can use a translator to get the gist of the problem, it makes it hard or impossible for others to search for issues they might be having.

closed 2024-01-24 by sairon