home-assistant/operating-system #3089: no rockchip npu module in odroid-m1 kernel...

mcarbonneaux, 2024-01-17:
### Describe the issue you are experiencing

to use npu to do inference with tensorlite we need to have the npu module... and it's seem to be absent on the haos 11.4 on odroid-m1...

### What operating system image do you use?

odroid-m1 (Hardkernel ODROID-M1)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

on the host i can see the load of the npu (link to CONFIG_ROCKCHIP_RKNPU_DEBUG_FS):
```
# cat /sys/kernel/debug/rknpu/load
cat: can't open '/sys/kernel/debug/rknpu/load': No such file or directory
```

none of the config related to Rockchip NPU are in config of the kernel.
```
# zcat /proc/config.gz | grep RKNPU
```

the config suposed to be in the kernel...
```
#
# RKNPU
#
CONFIG_ROCKCHIP_RKNPU=y
CONFIG_ROCKCHIP_RKNPU_DEBUG_FS=y
# CONFIG_ROCKCHIP_RKNPU_PROC_FS is not set
# CONFIG_ROCKCHIP_RKNPU_FENCE is not set
CONFIG_ROCKCHIP_RKNPU_DRM_GEM=y
```

### Anything in the Supervisor logs that might be useful for us?

```txt
na
```

### Anything in the Host logs that might be useful for us?

```txt
na
```

### System information

## System Information

version | core-2024.1.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.71-haos
arch | aarch64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1378
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 458.4 GB
disk_used | 8.5 GB
healthy | true
supported | true
board | odroid-m1
supervisor_api | ok
version_api | ok
installed_addons | Asterisk (4.1.2), Frigate Rockchip Beta (0.13.0) (0.13.0-rc1-rk), File editor (5.7.0), Advanced SSH & Web Terminal (17.0.4), RPC Shutdown (2.4), Mosquitto broker (6.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 17, 2024 at 18:18
-- | --
current_recorder_run | January 17, 2024 at 21:15
estimated_db_size | 0.50 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

sairon, 2024-01-18:
The RKNPU driver exist only in some downstream Linux kernel forks. In HAOS we are using upstream Linux kernel (with the only painful exception of Raspberry Pi) and porting the driver to it will require nonnegligible effort that could become a rabbit hole in the future on every attempt for Linux kernel update.

The only windmill fight you can try here is to convince Rockchip folks to upstream the driver to mainline Linux. Before that happens, or before another open source driver is available, I'm sorry to disappoint, but the NPU driver won't make its way to HAOS.

closed 2024-01-18 by sairon

MarcA711, 2024-01-23:
Hey @sairon,

I am not sure, if this is the right place to ask. The Rockchip NPU toolkit requires the device-tree at `/proc/device-tree/`, but HomeAssistant allows to mount the device-tree to `/device-tree` only. If this could be changed, we could develop an addon that makes use of the NPU on HomeAssistant supervised installations (that use the downstream kernel). However, it wouldn't work on HAOS due to driver issues (as you pointed out).

Is it possible to add an option to the addon `config.yaml` to allow addon developers choose where to mount the device-tree? There already is the option `devicetree: true`, I suggest adding an option like `devicetree_path: "/devicetree"` (with default value `/devicetree` for compatibility) that allow passing another path.

iBobik, 2024-12-28:
Mention about it has been published to the mainline:
https://blog.tomeuvizoso.net/2024/06/rockchip-npu-update-4-kernel-driver-for.html?m=1