home-assistant/operating-system #3092: Unable to access NVMe SSD on VIM3

ljbeal, 2024-01-21:
### Describe the issue you are experiencing

Hi! 

I've been running home assistant for a while now installed on a VIM3 Pro. It's been running smoothly for almost a year now.

I am, however noticing that the dashboard is taking longer and longer to load data into the graphs. I read that the most probable cause of that is slow storage. This makes sense, since HAOS is installed onto the eMMC, which from my understanding is not _much_ better than an SD card. I've also experienced a few occasions where I will wake up to things not loading properly (or at all, requiring a restart). I get the feeling that the eMMC might be acting up.

To alleviate this, I bought an m.2 NVMe SSD, and tried (and failed) to move the install over to it. (oowow isn't able to flash to NVMe, but that's not particularly relevant here).

My backup plan was to reinstall to eMMC, and then use the SSD as the data dir. However I'm not able to access the drive.

I know that the drive functions, since during my attempts to install HAOS on the SSD, I tested an Ubuntu (eMMC) install, and formatted the drive there. So the hardware seems to function as expected.

The drive is not present in terminal with a `ha os datadisk list`, returning nothing:

```[core-ssh ~]$ ha os datadisk list
devices: []
disks: []
```

It's also not present in `hardware -> All Hardware`. In Ubuntu `lsblk` would show it under something similar to `nvme1` (don't recall the exact path but the key point is that it's not there in HA).

Not exactly sure what can be done about this (perhaps this is one to put on the khadas forums?) but any advice is welcome.

### What operating system image do you use?

khadas-vim3 (Khadas VIM3)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

VIM3 Pro

no presence of `nvme*` in All Hardware, nor is anything shown in `ha os datadisk list`

### Anything in the Supervisor logs that might be useful for us?

```txt
Only messages regarding my own automations
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing relevant. Here's the sum total of the logs while verifying info for this issue:

Jan 21 13:26:25 homeassistant kernel: audit: type=1334 audit(1705843585.756:363): prog-id=120 op=LOAD
Jan 21 13:26:25 homeassistant systemd-timesyncd[523]: Network configuration changed, trying to establish connection.
Jan 21 13:26:25 homeassistant systemd[1]: Started Journal Gateway Service.
Jan 21 13:26:25 homeassistant systemd-timesyncd[523]: Contacted time server 162.159.200.123:123 (time.cloudflare.com).
Jan 21 13:26:25 homeassistant systemd-journal-gatewayd[54245]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.

Nothing relevant in the logsf for the Terminal application, either.
```

### System information

## System Information

version | core-2024.1.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.71-haos
arch | aarch64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.33.0
Stage | running
Available Repositories | 1383
Downloaded Repositories | 4
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 27.9 GB
disk_used | 7.7 GB
healthy | true
supported | true
board | khadas-vim3
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.4.0), Studio Code Server (5.15.0), Terminal & SSH (9.8.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 1
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 12 January 2024 at 10:47
-- | --
current_recorder_run | 21 January 2024 at 09:00
estimated_db_size | 1445.04 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

agners, 2024-01-23:
It seems that the PCIe port is disabled by default. We do use upstream U-Boot, which does not include the `kbi` tool required by this guide: https://docs.khadas.com/products/sbc/vim3/configurations/usb-pcie-switch

However, the command does a simple I2C write, which can be done from the U-Boot prompt with the following commands:

```
HAOS> i2c dev 0
Setting bus to 0
HAOS> i2c mw 0x18 0x33 1
```

Unfortunately it seems that we still miss a driver to make PCIe (and NVMe) support fully work. I've enabled it with PR #3100. So it should be possible to use the NVMe with the next release.

ljbeal, 2024-01-23:
This is excellent news, thank you for the update. I'll be sure to test it out as soon as it's available

sairon, 2024-02-02:
The driver should be available since 11.5.rc1, so I'm closing this. Let us know if you encounter any issues.

closed 2024-02-02 by sairon

ljbeal, 2024-02-05:
Can confirm that everything is working on my device now. Updated and moved the data disk with no issues