home-assistant/operating-system #3097: map not visible on pi 5

invisible-ladka, 2024-01-22:
### Describe the issue you are experiencing

the map view is not visible nomatter what devicce i open it on phone , pc  etc. im usinng pi5

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

core-2024.1.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.i dont know 
2. im new to this stuff i just  installed  ha 2 times map was not showing both times
3.thats all
...

### Anything in the Supervisor logs that might be useful for us?

```txt
everything looks good and green!!
```

### Anything in the Host logs that might be useful for us?

```txt
looks good and i dont know what to looks for
```

invisible-ladka, 2024-01-22:
[image]

Pammetje, 2024-01-23:
I have a similar issue. I do not see the map in de Andoid mobile app. I do see the map in a browser (also in a browser on the android mobile device) My HA is connected to the internet and using a reverse proxy.

closed 2024-01-23 by github-actions[bot]

Pammetje, 2024-01-23:
I resolved my issue by completly remove de HomeAssitant app from my android device and reinstall the app from scratch  (removing/empty the app cache did not solve the issue)

invisible-ladka, 2024-01-23:
> I resolved my issue by completly remove de HomeAssitant app from my android device and reinstall the app from scratch (removing/empty the app cache did not solve the issue)

i have tried that  and i even installed ha twice on rp5  i face  this issue on every devide i use

Pammetje, 2024-01-23:
@invisible-ladka 
Can you try to put the map on a new dashboard with no theme enabled and with a Masonry (default), or Panel view ?

invisible-ladka, 2024-01-23:
> @invisible-ladka Can you try to put the map on a new dashboard with no theme enabled and with a Masonry (default), or Panel view ?
 i just  tried,  its same.. black

Pammetje, 2024-01-24:
De mapdata is coming from the following domain/url : https://basemaps.cartocdn.com/ 
Can you connect to/reach that URL from you PI ? 
Do you maybe have an CSP policy (Content-Security-Policy) header (is your HA connected to the internet via a Reverse Proxy ?)

invisible-ladka, 2024-01-25:
> De mapdata is coming from the following domain/url : https://basemaps.cartocdn.com/ Can you connect to/reach that URL from you PI ? Do you maybe have an CSP policy (Content-Security-Policy) header (is your HA connected to the internet via a Reverse Proxy ?)

no im not able to connect , no..

Pammetje, 2024-01-25:
Glad it's been resolved