home-assistant/operating-system #3103: Can't randomly connect to Home Assistant local IP or external Nabu Casa URL. Only host reboot fixes it.

jjvelar, 2024-01-24:
### Describe the issue you are experiencing

Hi, I’m looking for some help on an issue that I’m crazy about.
I have Home Assistant OS installed on a NUC with Nabu Casa to be able to connect from outside.
Since a couple of weeks ago, and randomly once or twice a day, I am unable to access Home Assistant.
When I access through the IP inside the house I have a “connection error” whether I access from the browser or with the app on the mobile. And if it happens when I’m out of home, I can’t connect against the URL provided by Nabu Home either.
The only way to solve it is to physically reboot the NUC.
That's the info available about my installation:

- Version | core-2024.1.5
- Installation Type | Home Assistant OS
- Development | false
- Supervisor | true
- Docker | true
- User | root
- Virtual Environment | false
- Python Version | 3.11.6
- Operating System Family | Linux
- Operating System Version | 6.1.71-haos
- CPU Architecture | x86_64
- Timezone | Europe/Madrid
- Configuration Directory | /config

Thanks in advance for your help!

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.71-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. It happens randomly.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Not really.
```

### Anything in the Host logs that might be useful for us?

```txt
Not really.
```

### System information

Nothing to be repaired.

AndyKirby, 2024-01-24:
BTW the ip info for the wifi interface shows that there is no IP config and the actual wifi config seems to have set itself to disabled.

Restarting HA makes no difference you have to power cycle the whole thing to get it back. WHere upon it works till ti randomly fails a gain in a random (single number) of hours. 

If doing a bunch of work that uses the interface heavily (ESPHome builds and updates) it fails a lot quicker.

joshrose54, 2024-01-24:
I am having a very similar issue and no logs to show for it.

Every 15/20 mins home assistant loses connection with my router (It's connected via Ethernet) and for a few seconds I can't connect to the UI. 99% of the time it reconnects and then all is fine, but sometimes it needs a full power cycle.

I am able to verify this is happening by logging into my router after an occurrence of the above and seeing that the Pi running home assistant has only been connected for a few seconds.

No idea what is going on but it's causing me a lot of issues and definitely only started happening with the latest update.

jjvelar, 2024-01-24:
> I am having a very similar issue and no logs to show for it.
>
> Every 15/20 mins home assistant loses connection with my router (It's connected via Ethernet) and for a few seconds I can't connect to [...]

"only started happening with the lastest update" is exactly what happened to me!

joshrose54, 2024-01-24:
The only possible log I can find that seems to match-up with the disconnection and reconncetion is below:

```
Jan 24 23:15:43 homeassistant systemd-timesyncd[1154]: Network configuration changed, trying to establish connection.
Jan 24 23:15:43 homeassistant systemd-timesyncd[1154]: Contacted time server 128.138.140.44:123 (128.138.140.44).
Jan 24 23:15:44 homeassistant systemd-journal-gatewayd[3782]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.
```

jjvelar, 2024-01-25:
Hi,

This night connection was lost again so I have verified that:

Neither local IP, nor .local address or external Nabu Casa URL couldn’t be reached through ethernet or wifi.
I can ping Home Assistant local IP address
Home Assistant Observer webpage shows:
|Supervisor:|Connected|
|Supported:|Supported|
|Healthy:|Healthy|
I had to reboot host to fix the issue.

Thanks in advance for your help.

jjvelar, 2024-01-26:
Any idea or suggestion? It is really annoying! Thanks!

agners, 2024-01-31:
> I can ping Home Assistant local IP address
> Home Assistant Observer webpage shows:
> |Supervisor:|Connected|
> |Supported:|Supported|
> |Healthy:|Healthy|

The fact that you can ping the system and those metrics are healthy points to an issue with Home Assistant Core. Home Assistant Core provides the web server which is running on port 8123, so it seems that Core is no longer serving anything.

Can you check and share the Supervisor and Core logs? Especially after an outage. Also you might try restarting Home Assistant into safe mode to see if the issue goes away.

The HAOS issue tracker is the wrong place to further debug it. Probably the best location is the [community forum Installation section](https://community.home-assistant.io/c/installation/17). Feel free to tag me or leave a link here.

I'll mark this as Core issue and close it since at this point it is very unlikely that the problem is caused by Home Assistatn OS itself.

@AndyKirby @joshrose54 you seem to have lower layer network connectivity issue, which is different from what @jjvelar reports. Also on different platforms. Can you create separate issues along with all information for the problems you are seeing? Thanks!

closed 2024-01-31 by github-actions[bot]

jjvelar, 2024-02-04:
> The fact that you can ping the system and those metrics are healthy points to an issue with Home Assistant Core. Home Assistant Core provides the web server which is running on port 8123, so it [...]

Done!

https://community.home-assistant.io/t/cant-randomly-connect-to-home-assistant-local-ip-or-external-nabu-casa-url-only-host-reboot-fixes-it/684622

jjvelar, 2024-02-09:
Just for you all to know, it was the Bambu Lab integration.
Once I disabled it, had no more connection issues.