home-assistant/operating-system #3109: Boot Issue with Raspberry Pi 5 After Fresh Install Using RPi Imager 1.8.5

fbnksr, 2024-01-25:
### Describe the issue you are experiencing

I'm encountering a problem with my Raspberry Pi 5 where it fails to boot properly after I performed a fresh install of Home Assistant OS 11.4 using the Raspberry Pi Imager version 1.8.5. When powered on, the green light on the Pi flickers, and my monitor intermittently tries to display something but ends up showing "no signal."

Attempting to fix this, I connected the SD card to my MacBook to edit the config.txt file in the boot partition. However, the partition is not mounting, showing as "Volume failed to mount." I tried mounting it in read-only mode using diskutil mount readOnly, but this also failed.

I'm seeking assistance on how to resolve this boot issue, particularly with mounting the boot partition on macOS to edit the config.txt file or any other potential solutions.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Setup: Used Raspberry Pi Imager version 1.8.5 to install Home Assistant OS 11.4 on a Raspberry Pi 5.

2. SD Card Formatting: Formatted the SD card using the Raspberry Pi Imager and selected the Home Assistant OS 11.4 image for the Raspberry Pi 5.

3. Observing Raspberry Pi Behavior: After installation, observed that the Raspberry Pi's green light flickered intermittently and the connected monitor displayed "no signal."

4. Accessing the Boot Partition: Attempted to access the config.txt file on the SD card's boot partition using a MacBook. Encountered errors when trying to mount the partition, both in standard and read-only modes, with the system failing to recognize the partition.

5. Current State: Raspberry Pi remains unbootable, and the SD card’s boot partition is inaccessible on macOS.

### Anything in the Supervisor logs that might be useful for us?

```txt
No access to any supervisor logs
```

### Anything in the Host logs that might be useful for us?

```txt
No access to any Host logs
```

### System information

N/A

### Additional information

N/A

agners, 2024-01-25:
Did you try a second monitor? Maybe some type of monitor issue.

You don't need a monitor, your Home Assistant installation should become available after a few minutes at `http://homeassistant.local:8123`.

fbnksr, 2024-01-25:
Thank you for your suggestion. I tried using a different monitor but unfortunately, it didn't resolve the issue. My intention for using a monitor was to simply to confirm that the boot process was working. Regarding the Home Assistant access, I initially tried to connect using the hostname I set up in the RPi Imager. However, accessing it via http://homeassistant.local:8123 worked. Thank you for the help!!

closed 2024-01-25 by fbnksr