home-assistant/operating-system #3112: Can not boot HA in proxmox 8.1.4 after upgrade to 11.5.rc1

Rrapa, 2024-01-25:
### Describe the issue you are experiencing

System stuck on boot with message Slot A (OK=1 TRY=1)
Restoring to previous version helps. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

 11.5.rc1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to new RC1 version

### Anything in the Supervisor logs that might be useful for us?

```txt
Not possible to access logs
```

### Anything in the Host logs that might be useful for us?

```txt
I'm on Proxmox 8.1.4

16 x Intel(R) Core(TM) i7-10700T CPU @ 2.00GHz (1 Socket)
--
Kernel Version Linux 6.5.11-7-pve (2023-12-05T09:44Z)
Boot Mode EFI
Manager Version pve-manager/8.1.4
```

AllardKatan, 2024-01-25:
I also experienced this error, but I am on Virtualbox instead of Proxmox.

agners, 2024-01-25:
I was able to reproduce this, there seems to be a problem with the current OVA image.

You can either manually select the other boot slot, or reset the machine 4 times to let it fall back to the second boot slot. It then should boot again.

agners, 2024-01-25:
Actually it seems the system boots it just doesn't display this on the console.

The release is currently no longer on the beta channel for the `ova` image.

AllardKatan, 2024-01-26:
> I was able to reproduce this, there seems to be a problem with the current OVA image.
> 
> You can either manually select the other boot slot, or reset the machine 4 times to let it fall back to the second boot slot. It then should boot again.

Ah, thanks for that info. I had given up after three tries, plus a bunch of restart attempts from the snapshot taken after the first fail, and restored the backup of three days ago. Of course I should have made a snapshot before updating to a beta version but I was tired and not paying attention, I realized too late there is (of course) no automatic backup for OS updates like there is for regular updates.
I noticed that if you restart after manually selecting slot A, it will try slot A again, and fail. If I restart four times it will go to slot B, and restarting after that will automatically start from slot B.
If I can help to tackle this by supplying any log files let me know.

Rrapa, 2024-01-26:
I have skipped that update but I may restor it in that case.. looks like was not patient enough but it was really confisuing.

closed 2024-01-26 by sairon

Igloczek, 2024-02-06:
Not sure if this is realted, but while upgrading to 11.5 my Proxmox VM went into some weird limbo state, reporintg "io-error" on HA VM, so I wasn't able to even kill it. 
After the whole Proxmox restart HA was alive, but still on 11.4 🤔
Tried two times, exact same results.

Rrapa, 2024-02-06:
Works OK for me

agners, 2024-02-06:
@Igloczek if that is reproducible for you, probably make sense to open a new issue.

commit 3d234144 (2024-01-26): Add kernel options missing as a result of enabling CONFIG_EXPERT (#3114)

Enabling CONFIG_EXPERT, which was a dependency of some options we try
to set by our config fragments, had a side-effect of toggling some other
options, most importantly the framebuffer console support. Enable the
options found by diffing old and new kernel configs.

Fixes #3112
commit 01c8c1a5 (2024-02-05): Add kernel options missing as a result of enabling CONFIG_EXPERT (#3114)

Enabling CONFIG_EXPERT, which was a dependency of some options we try
to set by our config fragments, had a side-effect of toggling some other
options, most importantly the framebuffer console support. Enable the
options found by diffing old and new kernel configs.

Fixes #3112

(cherry picked from commit 3d234144a23993b69b569c8d2c666945e2c274a1)