home-assistant/operating-system #3119: System may report to google

RaspberryDiff, 2024-01-27:
### Describe the issue you are experiencing

Downloaded and imaged HAOS 11.4 for pi 4b.

Have not connected a blue cable to the pi, and journalctl -f, shows dockerd trying to resolve checkonline<dot>home-assistant<dot>io (markup added) from 8.8.{4.4,8.8}:53

Can HAOS not tell google i'm running it out-of-the-box? Or could this be because it does not have a dns from DHCP yet?

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Image sd card
2. Boot sd card
3. At console after, init, altF2 
4. Run journalctl -f
5. Observe logged entries trying to resolve via google.

### Anything in the Supervisor logs that might be useful for us?

```txt
None. The system is letterally just dd'd to the sd-card.
```

### Anything in the Host logs that might be useful for us?

```txt
None
```

agners, 2024-01-31:
> Can HAOS not tell google i'm running it out-of-the-box? Or could this be because it does not have a dns from DHCP yet?

Home Assistant should respect the DNS server provided by your DHCP. 

Can you check the DNS configuration reported by `nmcli`?

RaspberryDiff, 2024-02-01:
Hi Agners, thankyou for responding.
I can confirm, no address is available on end0.
I am yet to connect a network - I hope I mentioned this in the original ticket.

Do you see the dns lookup logs in journalctl when following the steps to reproduce the issue?

RaspberryDiff, 2024-02-01:
Could be non-private defaults as indicated in resolved.conf ...

freezurbern, 2024-02-05:
Perhaps its the default in Docker?
> If the container can't reach any of the IP addresses you specify, it uses Google's public DNS server at 8.8.8.8. This allows containers to resolve internet domains.
From https://docs.docker.com/network/#dns-services

RaspberryDiff, 2024-02-10:
Much as the industry has adopted principles like 'fail-safe' and 'fail secure', perhaps the product should fail-private?

dafunkydan, 2024-02-10:
> Home Assistant should respect the DNS server provided by your DHCP.

In Fact, it does only 50%, as it has a hard-coded DNS.

At least, there seems to be possibility to disable it:
https://community.home-assistant.io/t/improve-privacy-stop-using-hardcoded-dns/273496/86
https://www.home-assistant.io/more-info/unsupported/dns_server/

closed 2024-05-18 by github-actions[bot]