home-assistant/operating-system #3136: Correction/amendments needed to instructions on Wiki for Hardkernel Odroid XU boot partition troubleshooting/fixing

dgom, 2024-02-02:
There needs to be a correction made to this wiki page: https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/hardkernel/odroid-xu4.md

I can confirm that the instructions outlined in @mad-ady  post below are correct and working while the current XU4 instructions are wrong since the eMMC and SD devices are mixed up in the recommended commands to run to fix boot partition issues.

as originally posted: 
_Originally posted by @mad-ady in https://github.com/home-assistant/operating-system/issues/389#issuecomment-1473765242_

Start Quote:
Now, I've flashed HASSOS 9.5 to an 8GB orange EMMC that previously ran Hardkernel's Ubuntu. I also have a serial cable, so I did all these steps over serial, with network unplugged.

After flashing HASSOS to the EMMC, it fails to boot, because the older uboot that lives in the hidden partition can't parse/load the HASSOS kernel/config:
```
U-Boot 2017.05-00002-gd0752a9-dirty (Sep 08 2017 - 11:33:04 +0900) for ODROID-XU4

CPU:   Exynos5422 @ 800 MHz
Model: Odroid XU4 based on EXYNOS5422
Board: Odroid XU4 based on EXYNOS5422
Type:  xu4
DRAM:  2 GiB
MMC:   EXYNOS DWMMC: 0, EXYNOS DWMMC: 1
MMC Device 0 (eMMC): 7.3 GiB
Info eMMC rst_n_func status = enabled
MMC Device 1 ( SD ): 14.8 GiB

*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Press quickly 'Enter' twice to stop autoboot:  0
reading boot.ini
** Unable to read file boot.ini **
Failed to mount ext2 filesystem...
** Unrecognized filesystem type **
Failed to mount ext2 filesystem...
** Unrecognized filesystem type **
** Invalid partition 2 **
** Invalid partition 2 **
cfgload: no boot.ini or empty file
mmc block read, dev 0, addr 0x40008000, blk start 2047, blk cnt 16384
Bad Linux ARM zImage magic!
Exynos5422 # 
```

So, I flashed an SD card with the same HASSOS image, flipped the boot switch to SD and booted from SD:

```

U-Boot 2022.01 (Jan 24 2023 - 19:20:17 +0000) for ODROID-XU3/XU4/HC1/HC2

DRAM:  2 GiB
MMC:   EXYNOS DWMMC: 0, EXYNOS DWMMC: 2
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Odroid XU3/XU4/HC1/HC2 based on Exynos5422
Type:  xu4
Boot device: MMC(2)
Net:   No ethernet found.
switch to partitions #0, OK
mmc2 is current device
Scanning mmc 2:1...
Found U-Boot script /boot.scr
2619 bytes read in 16 ms (159.2 KiB/s)
## Executing script at 50000000
switch to partitions #0, OK
mmc2 is current device
loading env...

MMC read: dev # 2, block # 1222656, count 32 ... 32 blocks read: OK
36 bytes read in 14 ms (2 KiB/s)
64699 bytes read in 19 ms (3.2 MiB/s)
Trying to boot slot A, 1 attempts remaining. Loading kernel ...
10855112 bytes read in 2932 ms (3.5 MiB/s)
storing env...

MMC write: dev # 2, block # 1222656, count 32 ... 32 blocks written: OK
bootargs=zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=fdd22907379e4f71b9ae826e48900299 fsck.repair=yes root=PARTUUID=48617373-06 rootfstype=squashfs ro rootwait rauc.slot=A console=tty1 console=ttySAC2,115200
Starting kernel
Kernel image @ 0x42000000 [ 0x000000 - 0xa5a2c8 ]
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
   Loading Device Tree to 4ffed000, end 4ffffcba ... OK

Starting kernel ...

... output ommited ...

Welcome to Home Assistant
homeassistant login: root
Last login: Wed Jul 13 11:32:50 on ttySAC2
Welcome to Home Assistant OS.

Use `ha` to access the Home Assistant CLI.
# 
```
- note, that when booted from SD, the SD becomes /dev/mmcblk0, and the EMMC becomes /dev/mmcblk1! You can check with:
```
# ls -l /dev/mmcblk*
brw-rw----    1 root     disk      179,   0 Jul 13 11:31 /dev/mmcblk0
brw-rw----    1 root     disk      179,   1 Jul 13 11:31 /dev/mmcblk0p1
brw-rw----    1 root     disk      179,   2 Jul 13 11:31 /dev/mmcblk0p2
brw-rw----    1 root     disk      179,   3 Jul 13 11:31 /dev/mmcblk0p3
brw-rw----    1 root     disk      179,   4 Jul 13 11:31 /dev/mmcblk0p4
brw-rw----    1 root     disk      179,   5 Jul 13 11:31 /dev/mmcblk0p5
brw-rw----    1 root     disk      179,   6 Jul 13 11:31 /dev/mmcblk0p6
brw-rw----    1 root     disk      179,   7 Jul 13 11:31 /dev/mmcblk0p7
brw-rw----    1 root     disk      179,   8 Jul 13 11:31 /dev/mmcblk0p8
brw-rw----    1 root     disk      179,   9 Jul 13 11:31 /dev/mmcblk0p9
brw-rw----    1 root     disk      179,  16 Jul 13 11:31 /dev/mmcblk1
brw-rw----    1 root     disk      179,  32 Jul 13 11:31 /dev/mmcblk1boot0
brw-rw----    1 root     disk      179,  48 Jul 13 11:31 /dev/mmcblk1boot1
brw-rw----    1 root     disk      179,  17 Jul 13 11:31 /dev/mmcblk1p1
brw-rw----    1 root     disk      179,  18 Jul 13 11:31 /dev/mmcblk1p2
brw-rw----    1 root     disk      179,  19 Jul 13 11:31 /dev/mmcblk1p3
brw-rw----    1 root     disk      179,  20 Jul 13 11:31 /dev/mmcblk1p4
brw-rw----    1 root     disk      179,  21 Jul 13 11:31 /dev/mmcblk1p5
brw-rw----    1 root     disk      179,  22 Jul 13 11:31 /dev/mmcblk1p6
brw-rw----    1 root     disk      179,  23 Jul 13 11:31 /dev/mmcblk1p7
brw-rw----    1 root     disk      179,  24 Jul 13 11:31 /dev/mmcblk1p8
brw-rw----    1 root     disk      179,  25 Jul 13 11:36 /dev/mmcblk1p9
crw-------    1 root     root      244,   0 Jul 13 11:31 /dev/mmcblk1rpmb
```
Only the EMMC has mmcblkXboot0, mmcblkXboot1 (which are the hidden partitions).

- enable write for the hidden partition:
```
# echo 0 > /sys/block/mmcblk1boot0/force_ro
```
(the wiki page lists mmcblk0boot0, which doesn't exist!)
- copy HASSOS u-boot to the hidden partition:
```
# dd if=/dev/mmcblk1 of=/dev/mmcblk1boot0 bs=512 skip=63 seek=62 count=1440
1440+0 records in
1440+0 records out
737280 bytes (720.0KB) copied, 0.288219 seconds, 2.4MB/s

```
(again, the wiki uses mmcblk0 instead of mmcblk1!

- next, `poweroff`, remove the SD card and flip the switch to the EMMC position, and the board boots normally!

Cheers!

End Quote

_Originally posted by @mad-ady in https://github.com/home-assistant/operating-system/issues/389#issuecomment-1473765242_

dgom, 2024-02-03:
Fiddled more with this and it shows that sometimes the eMMC is /dev/mmcblk0 and sometimes /dev/mmcblk1.
As @mad-ady points out:
"Only the EMMC has mmcblkXboot0, mmcblkXboot1" (which are the hidden partitions)."
So checking for theese will give the info regarding which mmcblkX to use in with the dd-commands.

darkxst, 2024-02-17:
Can you try this on 12.0.rc1.

There is now an alias in the device tree that should fix the mmc index to 0

```
        aliases {
                mmc0 = &mmc_0;
        };

```

dgom, 2024-03-19:
Good that the mmc index is now fixed to 0! Seems as a viable solution.
Unfortunately I don´t have the possibility to test this any more....

closed 2024-06-25 by github-actions[bot]