home-assistant/operating-system #3140: Option for addons to set the device-tree mount path

MarcA711, 2024-02-04:
### Describe the issue you are experiencing

I am not sure if this is the right place to ask. If not, please tell me where I should put this.
Is it possible to add an option to the addon configuration `config.yaml` to allow addon developers to choose where to mount the device-tree? There already is the option `devicetree: true`, I suggest adding an option like `devicetree_path: "/devicetree"` (with default value `/devicetree` for compatibility) that allow passing another path. This would allow me to create a frigate addon that uses hardware acceleration using the NPU on modern rockchip boards (see https://github.com/blakeblackshear/frigate-hass-addons/issues/145).

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

11.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No.
```

### Anything in the Host logs that might be useful for us?

```txt
No.
```

### System information

Rockchip boards supporting rknn-toolkit2

Bascht74, 2024-02-04:
As Home Assistant supports the "Home Assistant Blue" with an ODROID-N2+ chip, 
I really hope that your wish will be granted.

I am using the same device (with the blue cover) and looking forward to use frigate with hardware acceleration. 
That would be a dream.

Bascht74, 2024-02-04:
@MarcA711 Could this be the right spot: 
https://github.com/home-assistant/supervisor/issues/4863

I think that supervisor has the logic for the `config.yaml`

agners, 2024-02-05:
> @MarcA711 Could this be the right spot:
> [home-assistant/supervisor#4863](https://github.com/home-assistant/supervisor/issues/4863)
> agners responded to the request already.

Yes, let's collaborate/discuss the issue in https://github.com/home-assistant/supervisor/issues/4863.

closed 2024-02-05 by agners