home-assistant/operating-system #3145: I can not boot at 11.5 OS

nazeg, 2024-02-05:
### Describe the issue you are experiencing

I have a Raspbeyy Pi 4. Everything was working without problem with 1.4. Last week, i installed 1.5 rc1 and i couldn't start HA anymore. Then i reinstalled the system, leaved from Beta channel and stayed at 1.4. Today i saw 1.5 was relased so i thought this issue has fixed but i got the same problem. It never boots, i also can not see Raspberry's IP address on router.

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

1.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Installed 1.5 rc1 and the system never booted
2. Installed a fresh system with 1.4, leaved beta channel
3. Today i saw 1.5 was relased, i installed it and got the same issue. It never boots, never gets an IP
...

### Anything in the Supervisor logs that might be useful for us?

```txt
.
```

### Anything in the Host logs that might be useful for us?

```txt
.
```

### System information

.

### Additional information

.

kastmgnru, 2024-02-05:
Same for me. Generic x86

Joo01, 2024-02-05:
Don't come up with me anymore after the update!

Joo01, 2024-02-05:
Don't come up with me anymore after the update to 11.5 OS! 
HA run on Windows - VirtualBOX .
But after it couldn't start, it apparently did an automatic downgrade to 11.4 OS and started up normally again?! Can this be?
So now it runs with 11.4 OS and I don't dare to upgrade again up with 11.5.

wschaeffer, 2024-02-05:
Same for me, can't connect to the pi after updating to 11.5. 
Raspberry Pi 4, using Home Assistant Operating System.

agners, 2024-02-05:
In general, we have all tested all boards extensively with HAOS 11.5. E.g. I have tested Raspberry Pi 4 with SD card as well as SSD boot. This doesn not mean that there aren't any bugs, but it means that whatever bugs you encounter, it must be environment specific, as in, it depends on the exact device type and the hardware you have connected to your device etc. So please always state as much information as possible.

:information_source:  This issue is about Raspberry Pi 4. If you have a different device, please open a separate issue!

@nazeg Do you see anything when connecting an HDMI screen? What is your setup exactly (SD card, type, make, USB devices connected etc.)

> After updating to 11.5 OS it no longer works for me either!
HA run on Windows - VirtualBOX .
But after it couldn't start, it apparently did an automatic downgrade to 11.4 OS and started up normally again?! Can this be?
So now it runs with 11.4 OS and I don't dare to upgrade again up with 11.5.

Home Assistant OS has a fallback feature, where after 3 unsuccessful tries it falls back to the previously installed version. It seems that go triggered in your case. In any case, logs/screenshots from the boot attempt when it failed would be helpful. Please open a separate issue, as this is about Raspberry Pi 4.

Raspberry Pi 4 users:  Maybe disconnecting power 3 times and reapplying power for ~20-30s might help to trigger the fallback in your case too.

nazeg, 2024-02-05:
@agners Unfortunately i can't see anything when i connect it via HDMI, only black screen. Also as i said the device can't get an ip address. I use SSD and a zigbee adapter, that's all.

agners, 2024-02-05:
@nazeg are you booting from the SD card? What type and make is your SSD? What type and make is your USB to SSD adapter?

wschaeffer, 2024-02-05:
Looks like it is stuck in a boot loop. Both red and green LED are active and within a minute they are dimmed for a second. In my router I see constantly see reconnections happening.

pwnpanda, 2024-02-05:
I am in the same boat, also on a RPI 4.
Also using a zigbee adapter: https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/

I had to reset the device manually after running the update, as the pi never came back. After reset, it was still not working. (Presenting on the network)

I connected it via the LAN-port, with the zigbee-adapter connected and gained access to the device, but a lot of functionality is not working and the WiFi does not connect.

Which logs are the relevant ones here? The system is clearly not working as intended, indicated by lots of errors in integrations, backups not working in UI, WiFi connection not working etc.

It also appears that something failed when making a backup, as no pre-update backup was created, even though it was enabled, stopping me from just rolling back to the previous version.

Screen when accessing backup via UI:
[image]

Login screen over SSH showing no wifi connected:
[image]

Adding the logs I found, but does not seem like there is anything of interest there:
[home-assistant_2024-02-05T19-38-24.548Z.log](https://github.com/home-assistant/operating-system/files/14171379/home-assistant_2024-02-05T19-38-24.548Z.log)
[host_2024-02-05T19-01-51.207Z.log](https://github.com/home-assistant/operating-system/files/14171380/host_2024-02-05T19-01-51.207Z.log)
[supervisor_2024-02-05T19-40-59.215Z.log](https://github.com/home-assistant/operating-system/files/14171377/supervisor_2024-02-05T19-40-59.215Z.log)

Will report back after trying to boot without zigbee adapter and after resetting repeatedly in hopes of rolling back automatically.

nazeg, 2024-02-06:
Btw, it's rollbacked by using that way but i see that Operating System is 11.5 now. How can it be?

C0MTRAYA, 2024-02-06:
Same here, tried on a test card because it could not boot since 11.4

-Using a PI4 4Go
-Im currently on OS 11.2
-Booting from an SD card
-I have a sonoff zigbee 3.0 usb dongle plus v2 and a Zooz ZST10 700 Series Z-Wave Plus S2 USB ZST10 700 connected
-Connected through ethernet cable

Heres what I have from the HDMI output:
[image]

noizefloor, 2024-02-07:
I had the same problem on a Raspberry PI 3B. No output on HDMI, so I suspected it could be the boot loader. I managed to fix my problem by overwriting the boot partition with an earlier version. I would assume it was the u–boot update that came with 11.5.

pdeloos, 2024-02-08:
I have the same problem. The curious thing is that when I connect the Raspberry PI to my laptop via USB-C, I do get output on my monitor and everything works fine. When I then try again via the power supply that used to work, again no output on my monitor.

erkr, 2024-02-08:
Related to issue https://github.com/home-assistant/operating-system/issues/3059?

pdeloos, 2024-02-08:
No, I just figured out that the cause is most likely that the power adapter I'm using does not provide the required levels anymore. I'm now using my iPhone charger and it's working fine. Sorry for the confusion.

martymarty004, 2024-02-08:
I also have been experiencing these issues on my RPi 4B + HAOS on USB SSD (connected to USB2.0).
The system randomly hangs, and after I reboot it there is a hole in the history and there are no traces of errors on `journalctl`.

It's like the SSD gets disconnected for some reason.
Also, the recordings stop way before I am able to notice the issue (the system will still "work" for some time without a disk attached)

For example, I updated to 11.5 this morning and everything was fine until I noticed that DNS was dead (AdGuard on HA) and the RPi refused connections over SSH (though it was still pinging).

C0MTRAYA, 2024-02-10:
if it might help someone, I've deleted the boot partition and formatted my test card, installed a fresh install on 11.5 and it booted (which wasn't on 11.4), restaured from a full backup... and:
[image]
I've shutdown the host (not just close ha) and unplug the pi for good mesures, plugged it back in and it boot like normal! gonna run it on the test card for a few days and if everything's fine, I'll updates my main cards!

closed 2024-05-18 by github-actions[bot]